esphome/issues #3464: CO2 sensor can't send data to Home Assistant.

kevinblo, 2022-07-24:
### The problem

Every few days, the HA stops receiving data from the sensor. Prior to this, the sensor worked for more than a year without failures. At the same time, these data are visible in the asphome logs. It helps to disconnect the sensor from the power supply and connect it back. Or delete and add this sensor to HA again.
```
[12:49:06][C][api:138]: API Server:
[12:49:06][C][api:139]:   Address: 192.168.1.214:6053
[12:49:06][C][api:143]:   Using noise encryption: NO
[12:49:32][D][mhz19:057]: MHZ19 Received CO₂=829ppm Temperature=37°C Status=0x00
[12:49:32][D][sensor:124]: 'MH-Z19 CO2 Value': Sending state 829.00000 ppm with 0 decimals of accuracy
[12:49:32][D][sensor:124]: 'MH-Z19 Temperature': Sending state 37.00000 °C with 0 decimals of accuracy
[12:50:32][D][mhz19:057]: MHZ19 Received CO₂=832ppm Temperature=37°C Status=0x00
[12:50:32][D][sensor:124]: 'MH-Z19 CO2 Value': Sending state 832.00000 ppm with 0 decimals of accuracy
[12:50:32][D][sensor:124]: 'MH-Z19 Temperature': Sending state 37.00000 °C with 0 decimals of accuracy
[12:51:32][D][mhz19:057]: MHZ19 Received CO₂=831ppm Temperature=37°C Status=0x00
[12:51:32][D][sensor:124]: 'MH-Z19 CO2 Value': Sending state 831.00000 ppm with 0 decimals of accuracy
[12:51:32][D][sensor:124]: 'MH-Z19 Temperature': Sending state 37.00000 °C with 0 decimals of accuracy
```

[image]

[image]

### Which version of ESPHome has the issue?

2022.6.2

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2022.7.4

### What platform are you using?

ESP8266

### Component causing the issue

MH-Z19

### Example YAML snippet

```yaml
uart:
  rx_pin: GPIO12
  tx_pin: GPIO14
  baud_rate: 9600

sensor:
  - platform: mhz19
    co2:
      name: "MH-Z19 CO2 Value"
    temperature:
      name: "MH-Z19 Temperature"
    update_interval: 60s
    automatic_baseline_calibration: true
```

### Anything in the logs that might be useful for us?

```txt
[12:49:06][I][app:102]: ESPHome version 2022.6.2 compiled on Jul  4 2022, 17:18:49
[12:49:06][C][status_led:019]: Status LED:
[12:49:06][C][status_led:020]:   Pin: GPIO2
[12:49:06][C][wifi:491]: WiFi:
[12:49:06][C][wifi:353]:   Local MAC: E0:98:06:84:A1:DF
[12:49:06][C][wifi:354]:   SSID: [redacted]
[12:49:06][C][wifi:355]:   IP Address: 192.168.1.214
[12:49:06][C][wifi:356]:   BSSID: [redacted]
[12:49:06][C][wifi:358]:   Hostname: 'co2-sensor'
[12:49:06][C][wifi:360]:   Signal strength: -70 dB ▂▄▆█
[12:49:06][C][wifi:364]:   Channel: 10
[12:49:06][C][wifi:365]:   Subnet: 255.255.255.0
[12:49:06][C][wifi:366]:   Gateway: 192.168.1.1
[12:49:06][C][wifi:367]:   DNS1: 0.0.0.0
[12:49:06][C][wifi:368]:   DNS2: 0.0.0.0
[12:49:06][C][logger:275]: Logger:
[12:49:06][C][logger:276]:   Level: DEBUG
[12:49:06][C][logger:277]:   Log Baud Rate: 115200
[12:49:06][C][logger:278]:   Hardware UART: UART0
[12:49:06][C][uart.arduino_esp8266:102]: UART Bus:
[12:49:06][C][uart.arduino_esp8266:103]:   TX Pin: GPIO14
[12:49:06][C][uart.arduino_esp8266:104]:   RX Pin: GPIO12
[12:49:06][C][uart.arduino_esp8266:106]:   RX Buffer Size: 256
[12:49:06][C][uart.arduino_esp8266:108]:   Baud Rate: 9600 baud
[12:49:06][C][uart.arduino_esp8266:109]:   Data Bits: 8
[12:49:06][C][uart.arduino_esp8266:110]:   Parity: NONE
[12:49:06][C][uart.arduino_esp8266:111]:   Stop bits: 1
[12:49:06][C][uart.arduino_esp8266:115]:   Using software serial
[12:49:06][C][mhz19:094]: MH-Z19:
[12:49:06][C][mhz19:095]:   CO2 'MH-Z19 CO2 Value'
[12:49:06][C][mhz19:095]:     Device Class: 'carbon_dioxide'
[12:49:06][C][mhz19:095]:     State Class: 'measurement'
[12:49:06][C][mhz19:095]:     Unit of Measurement: 'ppm'
[12:49:06][C][mhz19:095]:     Accuracy Decimals: 0
[12:49:06][C][mhz19:095]:     Icon: 'mdi:molecule-co2'
[12:49:06][C][mhz19:096]:   Temperature 'MH-Z19 Temperature'
[12:49:06][C][mhz19:096]:     Device Class: 'temperature'
[12:49:06][C][mhz19:096]:     State Class: 'measurement'
[12:49:06][C][mhz19:096]:     Unit of Measurement: '°C'
[12:49:06][C][mhz19:096]:     Accuracy Decimals: 0
[12:49:06][C][mhz19:100]:   Automatic baseline calibration enabled on boot
[12:49:06][C][captive_portal:088]: Captive Portal:
[12:49:06][C][mdns:084]: mDNS:
[12:49:06][C][mdns:085]:   Hostname: co2-sensor
[12:49:06][C][ota:085]: Over-The-Air Updates:
[12:49:06][C][ota:086]:   Address: 192.168.1.214:8266
[12:49:06][C][ota:089]:   Using Password.
[12:49:06][C][api:138]: API Server:
[12:49:06][C][api:139]:   Address: 192.168.1.214:6053
[12:49:06][C][api:143]:   Using noise encryption: NO
[12:49:32][D][mhz19:057]: MHZ19 Received CO₂=829ppm Temperature=37°C Status=0x00
[12:49:32][D][sensor:124]: 'MH-Z19 CO2 Value': Sending state 829.00000 ppm with 0 decimals of accuracy
[12:49:32][D][sensor:124]: 'MH-Z19 Temperature': Sending state 37.00000 °C with 0 decimals of accuracy
[12:50:32][D][mhz19:057]: MHZ19 Received CO₂=832ppm Temperature=37°C Status=0x00
[12:50:32][D][sensor:124]: 'MH-Z19 CO2 Value': Sending state 832.00000 ppm with 0 decimals of accuracy
[12:50:32][D][sensor:124]: 'MH-Z19 Temperature': Sending state 37.00000 °C with 0 decimals of accuracy
[12:51:32][D][mhz19:057]: MHZ19 Received CO₂=831ppm Temperature=37°C Status=0x00
[12:51:32][D][sensor:124]: 'MH-Z19 CO2 Value': Sending state 831.00000 ppm with 0 decimals of accuracy
[12:51:32][D][sensor:124]: 'MH-Z19 Temperature': Sending state 37.00000 °C with 0 decimals of accuracy
[12:52:32][D][mhz19:057]: MHZ19 Received CO₂=829ppm Temperature=37°C Status=0x00
[12:52:32][D][sensor:124]: 'MH-Z19 CO2 Value': Sending state 829.00000 ppm with 0 decimals of accuracy
[12:52:32][D][sensor:124]: 'MH-Z19 Temperature': Sending state 37.00000 °C with 0 decimals of accuracy
[12:53:32][D][mhz19:057]: MHZ19 Received CO₂=828ppm Temperature=37°C Status=0x00
[12:53:32][D][sensor:124]: 'MH-Z19 CO2 Value': Sending state 828.00000 ppm with 0 decimals of accuracy
[12:53:32][D][sensor:124]: 'MH-Z19 Temperature': Sending state 37.00000 °C with 0 decimals of accuracy
[12:54:32][D][mhz19:057]: MHZ19 Received CO₂=826ppm Temperature=37°C Status=0x00
[12:54:32][D][sensor:124]: 'MH-Z19 CO2 Value': Sending state 826.00000 ppm with 0 decimals of accuracy
[12:54:32][D][sensor:124]: 'MH-Z19 Temperature': Sending state 37.00000 °C with 0 decimals of accuracy
[12:55:32][D][mhz19:057]: MHZ19 Received CO₂=825ppm Temperature=37°C Status=0x00
[12:55:32][D][sensor:124]: 'MH-Z19 CO2 Value': Sending state 825.00000 ppm with 0 decimals of accuracy
[12:55:32][D][sensor:124]: 'MH-Z19 Temperature': Sending state 37.00000 °C with 0 decimals of accuracy
[12:56:32][D][mhz19:057]: MHZ19 Received CO₂=824ppm Temperature=37°C Status=0x00
[12:56:32][D][sensor:124]: 'MH-Z19 CO2 Value': Sending state 824.00000 ppm with 0 decimals of accuracy
[12:56:32][D][sensor:124]: 'MH-Z19 Temperature': Sending state 37.00000 °C with 0 decimals of accuracy
```

ssieb, 2022-08-02:
Check the HA logs for why it's not able to connect to the device.

kevinblo, 2022-08-02:
I haven't errors about co2 sensor in HA

closed 2022-12-16 by github-actions[bot]