esphome/issues #3465: Missing Arduino.h for for esp32doit-espduino board

putz612, 2022-07-24:
### The problem

While trying to build for a this board I got fatal errors due to missing Arduino.h.

### Which version of ESPHome has the issue?

2002.6.2

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

Home Assistant 2022.7.5

### What platform are you using?

ESP32

### Board

Wemos D1 R32 ESP32

### Example YAML snippet

```yaml
esphome:
  name: ${node_name}
  platform: ESP32
  board: esp32doit-espduino
```

### Anything in the logs that might be useful for us?

```txt
compilation terminated.
*** [.pioenvs/hvac32/src/esphome/components/captive_portal/captive_portal.cpp.o] Error 1
In file included from /piolibs/ArduinoJson/src/ArduinoJson/Configuration.hpp:156:0,
                 from /piolibs/ArduinoJson/src/ArduinoJson.hpp:7,
                 from /piolibs/ArduinoJson/src/ArduinoJson.h:9,
                 from src/esphome/components/json/json_util.h:11,
                 from src/esphome/components/json/json_util.cpp:1:
/config/.esphome/platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:185:26: fatal error: pins_arduino.h: No such file or directory
```

### Additional information

https://docs.platformio.org/en/latest/boards/espressif32/esp32doit-espduino.html
https://www.cnx-software.com/2017/09/04/espduino-32-wemos-d1-r32-esp32-boards-support-some-arduino-uno-shields/

putz612, 2022-07-24:
Related to #2766

ssieb, 2022-08-01:
```yaml
esp32:
  board: esp32doit-espduino
  framework:
    version: 2.0.3
    platform_version: 4.4.0
```
works with dev esphome.

closed 2022-12-15 by github-actions[bot]