esphome/issues #3466: Installation on Fedora 35 via pip fails

Mebus, 2022-07-24:
### The problem

Hallo, 

unfortunately I cannot install ESPHome on Fedora 35 via:

The result is:

```
 pip3 install --user esphome
Collecting esphome
  Using cached esphome-2022.6.2-py2.py3-none-any.whl (2.3 MB)
Requirement already satisfied: PyYAML==6.0 in ./.local/lib/python3.10/site-packages (from esphome) (6.0)
Requirement already satisfied: click==8.1.3 in ./.local/lib/python3.10/site-packages (from esphome) (8.1.3)
Requirement already satisfied: zeroconf==0.38.4 in ./.local/lib/python3.10/site-packages (from esphome) (0.38.4)
Requirement already satisfied: tornado==6.1 in /usr/lib64/python3.10/site-packages (from esphome) (6.1)
Collecting paho-mqtt==1.6.1
  Using cached paho-mqtt-1.6.1.tar.gz (99 kB)
Collecting esptool==3.3.1
  Using cached esptool-3.3.1.tar.gz (213 kB)
Collecting aioesphomeapi==10.8.2
  Using cached aioesphomeapi-10.8.2-py2.py3-none-any.whl (49 kB)
Collecting kconfiglib==13.7.1
  Using cached kconfiglib-13.7.1-py2.py3-none-any.whl (145 kB)
Collecting voluptuous==0.13.1
  Using cached voluptuous-0.13.1-py3-none-any.whl (29 kB)
Requirement already satisfied: tzdata>=2021.1 in ./.local/lib/python3.10/site-packages (from esphome) (2022.1)
Collecting colorama==0.4.4
  Using cached colorama-0.4.4-py2.py3-none-any.whl (16 kB)
Collecting pyserial==3.5
  Using cached pyserial-3.5-py2.py3-none-any.whl (90 kB)
Collecting esphome-dashboard==20220508.0
  Using cached esphome_dashboard-20220508.0-py3-none-any.whl (601 kB)
Collecting platformio==5.2.5
  Using cached platformio-5.2.5.tar.gz (218 kB)
Collecting tzlocal==4.2
  Using cached tzlocal-4.2-py3-none-any.whl (19 kB)
Requirement already satisfied: protobuf<4.0,>=3.12.2 in /usr/lib/python3.10/site-packages (from aioesphomeapi==10.8.2->esphome) (3.14.0)
Collecting noiseprotocol<1.0,>=0.3.1
  Using cached noiseprotocol-0.3.1-py3-none-any.whl (20 kB)
Collecting bitstring>=3.1.6
  Using cached bitstring-3.1.9-py3-none-any.whl (38 kB)
Requirement already satisfied: cryptography>=2.1.4 in /usr/lib64/python3.10/site-packages (from esptool==3.3.1->esphome) (3.4.7)
Requirement already satisfied: ecdsa>=0.16.0 in /usr/lib/python3.10/site-packages (from esptool==3.3.1->esphome) (0.17.0)
Collecting reedsolo<=1.5.4,>=1.5.3
  Using cached reedsolo-1.5.4.tar.gz (271 kB)
Requirement already satisfied: bottle==0.12.* in /usr/lib/python3.10/site-packages (from platformio==5.2.5->esphome) (0.12.21)
Collecting marshmallow<4,>=2
  Using cached marshmallow-3.17.0-py3-none-any.whl (48 kB)
Collecting pyelftools<1,>=0.27
  Using cached pyelftools-0.28-py2.py3-none-any.whl (155 kB)
Requirement already satisfied: requests==2.* in /usr/lib/python3.10/site-packages (from platformio==5.2.5->esphome) (2.27.1)
Requirement already satisfied: semantic_version==2.9.* in ./.local/lib/python3.10/site-packages (from platformio==5.2.5->esphome) (2.9.0)
Requirement already satisfied: tabulate==0.8.* in /usr/lib/python3.10/site-packages (from platformio==5.2.5->esphome) (0.8.9)
Collecting aiofiles==0.8.*
  Using cached aiofiles-0.8.0-py3-none-any.whl (13 kB)
Collecting ajsonrpc==1.*
  Using cached ajsonrpc-1.2.0-py3-none-any.whl (22 kB)
Requirement already satisfied: starlette==0.18.* in ./.local/lib/python3.10/site-packages (from platformio==5.2.5->esphome) (0.18.0)
Requirement already satisfied: uvicorn==0.17.* in ./.local/lib/python3.10/site-packages (from platformio==5.2.5->esphome) (0.17.6)
Requirement already satisfied: wsproto==1.0.* in ./.local/lib/python3.10/site-packages (from platformio==5.2.5->esphome) (1.0.0)
Requirement already satisfied: urllib3<1.27,>=1.21.1 in /usr/lib/python3.10/site-packages (from requests==2.*->platformio==5.2.5->esphome) (1.26.7)
Requirement already satisfied: idna<4,>=2.5 in /usr/lib/python3.10/site-packages (from requests==2.*->platformio==5.2.5->esphome) (3.2)
Requirement already satisfied: charset-normalizer~=2.0.0 in /usr/lib/python3.10/site-packages (from requests==2.*->platformio==5.2.5->esphome) (2.0.4)
Requirement already satisfied: anyio<4,>=3.0.0 in ./.local/lib/python3.10/site-packages (from starlette==0.18.*->platformio==5.2.5->esphome) (3.6.1)
Collecting pytz-deprecation-shim
  Using cached pytz_deprecation_shim-0.1.0.post0-py2.py3-none-any.whl (15 kB)
Requirement already satisfied: asgiref>=3.4.0 in ./.local/lib/python3.10/site-packages (from uvicorn==0.17.*->platformio==5.2.5->esphome) (3.5.2)
Requirement already satisfied: h11>=0.8 in ./.local/lib/python3.10/site-packages (from uvicorn==0.17.*->platformio==5.2.5->esphome) (0.13.0)
Requirement already satisfied: ifaddr>=0.1.7 in /usr/lib/python3.10/site-packages (from zeroconf==0.38.4->esphome) (0.1.7)
[... 172 lines cut ...]
                     von /usr/include/python3.10/Python.h:83,
                     von creedsolo.c:16:
    /usr/include/python3.10/cpython/unicodeobject.h:551:42: Anmerkung: hier deklariert
      551 | Py_DEPRECATED(3.3) PyAPI_FUNC(PyObject*) PyUnicode_FromUnicode(
          |                                          ^~~~~~~~~~~~~~~~~~~~~
    error: command '/usr/bin/gcc' failed with exit code 1
    ----------------------------------------
ERROR: Command errored out with exit status 1: /usr/bin/python3 -u -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-s6jfdqca/reedsolo_3fd0e47f2aa34203b1fdcf5fd0cb5320/setup.py'"'"'; __file__='"'"'/tmp/pip-install-s6jfdqca/reedsolo_3fd0e47f2aa34203b1fdcf5fd0cb5320/setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-lqu0l6jg/install-record.txt --single-version-externally-managed --user --prefix= --compile --install-headers /home/mebus/.local/include/python3.10/reedsolo Check the logs for full command output.
```

Is this a bug or just an issue with my machine?

Mebus

### Which version of ESPHome has the issue?

2022.6.2

### What type of installation are you using?

pip

### What platform are you using?

ESP32

ssieb, 2022-08-01:
It must be an issue with your machine somehow.  I just tried a clean F35 install and `pip install esphome` ran with no errors.

KiLLeRRaT, 2022-10-20:
I'm running into the same thing, 

Manjaro 22.0.0, 5.15.74-3-MANJARO
Python 3.10.7
pip 22.2.2

Same error as you.

KiLLeRRaT, 2022-10-20:
Hi,

Just found something that worked for me,

So it's the specific `reedsolo` package causing it, and I found that if you do the below, it worked:
```
pip install --only-binary :all: --extra-index-url https://dl.espressif.com/pypi/ reedsolo
pip install --user esphome
```

I found it here: https://pythontechworld.com/issue/espressif/esptool/711

Hope it helps you too.

Cheers,