esphome/issues #3469: Camera doesn't work with pulse_meter integration

hugokernel, 2022-07-26:
### The problem

It is not possible to use esp32_camera and pulse_meter integrations both together.

Camera setup always failed.

### Which version of ESPHome has the issue?

2022.6.2

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

2022.7.0

### What platform are you using?

ESP32

### Board

esp32cam

### Component causing the issue

pulse_meter, esp32_camera

### Example YAML snippet

```yaml
esphome:
  name: camppulse
  platform: ESP32
  board: esp32cam

<<: !include network.yaml

captive_portal:

# Enable Home Assistant API
api:
  password: !secret password

ota:
  password: !secret password

logger:
  level: INFO

sensor:
  - platform: pulse_meter
    name: Pulse
    pin: GPIO16
    internal_filter: 10us
    accuracy_decimals: 2
    timeout: 30s

esp32_camera:
  id: camera
  external_clock:
    pin: GPIO0
    frequency: 20MHz
  i2c_pins:
    sda: GPIO26
    scl: GPIO27
  data_pins: [GPIO5, GPIO18, GPIO19, GPIO21, GPIO36, GPIO39, GPIO34, GPIO35]
  vsync_pin: GPIO25
  href_pin: GPIO23
  pixel_clock_pin: GPIO22
  power_down_pin: GPIO32
  name: "${friendly_name} image"
  test_pattern: false
  vertical_flip: false
  horizontal_mirror: false
  jpeg_quality: 63
```

### Anything in the logs that might be useful for us?

```txt
INFO Waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from watermeter.local using esphome API
INFO Successfully connected to watermeter.local
[06:50:24][I][app:102]: ESPHome version 2022.6.2 compiled on Jul 26 2022, 06:49:32
[06:50:24][E][esp32_camera:129]:   Setup Failed: ERROR
```

### Additional information

I tried to find out where the problem was and it seems to be with the interrupt handling.

If I remove the `attachInterrupt` call in the pulse_meter `setup`, there is no problem and camera initialization is ok.

Note:
- I use the `ESP32 AI Thinker` board (schematic: https://i0.wp.com/randomnerdtutorials.com/wp-content/uploads/2020/03/ESP32-CAM-AI-Thinker-schematic-diagram.png?quality=100&strip=all&ssl=1)
- The `GPIO16` is not used by the camera module (and problem is the same with other free GPIO)

Related links:
- https://github.com/espressif/esp32-camera/issues/256#issuecomment-813520803
- https://github.com/esphome/issues/issues/2483

hugokernel, 2022-09-30:
I have just tested with an ESP32-WROVER-DEV module (ESP32-WROVER-E module) and the problem is the same: It is not possible to use the pulse integration with the camera module.

```yaml
esphome:
  name: campulse
  platform: ESP32
  board: esp32cam

<<: !include network.yaml

captive_portal:

api:
  password: !secret password

ota:
  password: !secret password

logger:
  level: INFO

sensor:
  - platform: pulse_meter
    name: Pulse
    pin: GPIO16
    internal_filter: 10us
    accuracy_decimals: 2
    timeout: 30s

esp32_camera:
  name: My Camera
  id: camera
  external_clock:
    pin: GPIO21
    frequency: 10MHz
  i2c_pins:
    sda: GPIO26
    scl: GPIO27
  data_pins: [GPIO4, GPIO5, GPIO18, GPIO19, GPIO36, GPIO39, GPIO34, GPIO35]
  vsync_pin: GPIO25
  href_pin: GPIO23
  pixel_clock_pin: GPIO22
```

The output:

```
INFO Waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from campulse.local using esphome API
INFO Successfully connected to campulse.local
[08:12:39][I][app:102]: ESPHome version 2022.8.3 compiled on Sep 30 2022, 08:12:01
[08:12:39][E][esp32_camera:095]:   Setup Failed: ESP_ERR_NOT_FOUND
```

If I remove the pulse integration, camera works well.

ssieb, 2022-09-30:
Yes, as described in the first comment, there's a problem with how interrupts work.  There's a conflict between what the camera framework does and everything else.

hugokernel, 2022-09-30:
> Yes, as described in the first comment, there's a problem with how interrupts work. There's a conflict between what the camera framework does and everything else.

I understood well, I was just bringing a test with a different device to rule out a hardware problem.

kaangoksal, 2022-10-03:
I have the exact same problem.... Can't use interrupts with esp_camera.h, I'm using esp-idf and native esp functions to define interrupts. 

```
static void configureInterruptGPIO() {
    ESP_LOGI(TAG, "Configuring GPIO interrupt");

    gpio_config_t io_conf = {};

    //interrupt of rising edge
    io_conf.intr_type = GPIO_INTR_POSEDGE;
    //bit mask of the pins, use GPIO4/5 here
    io_conf.pin_bit_mask = GPIO_INPUT_PIN_SEL;
    //set as input mode
    io_conf.mode = GPIO_MODE_INPUT;
    //enable pull-up mode
    io_conf.pull_up_en = 1;
    gpio_config(&io_conf);

    //gpio_install_isr_service(ESP_INTR_FLAG_DEFAULT); // I guess because of camera init this is not needed? 

    gpio_isr_handler_add(GPIO_INPUT_IO_0, gpio_isr_handler, (void*) GPIO_INPUT_IO_0);
}
```
here is how they are called 
```
void app_main(void)
{
    gpio_evt_queue = xQueueCreate(10, sizeof(uint32_t));
    configureSDCard();
    init_camera(FRAMESIZE_UXGA);

    configureLedGPIO();

    configureInterruptGPIO();
}

```

when configureInterruptGPIO calls, I get 
`Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.`

AapoTahkola, 2022-10-03:
I am having issues with ai thinker esp32cam and servo control. After changing ledc to use different channel(channel 0 is used by the cam) I get no output. https://community.home-assistant.io/t/ai-thinker-esp32-cam-with-servo-control/466312
Should I file another bug report?

kaangoksal, 2022-10-04:
@AapoTahkola this document is pretty useful to find out what pins you can use [document](https://github.com/raphaelbs/esp32-cam-ai-thinker/blob/master/docs/esp32cam-pin-notes.md) 

if you disable sd card there are pins that you can use. such as 13

hugokernel, 2022-10-04:
I specify that in my 2 tests on the 2 cameras, the GPIO used for the `pulse_meter` integration was free and without conflict with the GPIOs used by the camera.

AapoTahkola, 2022-10-04:
> @AapoTahkola this document is pretty useful to find out what pins you can use [document](https://github.com/raphaelbs/esp32-cam-ai-thinker/blob/master/docs/esp32cam-pin-notes.md)
> 
> if you disable sd card there are pins that you can use. such as 13

I though naively as this esp32cam module does not have tf card functionality it would not be enabled. I have been trying with pins 12 and 14 mostly which are tf pins. How am I to accomplish this task of compiling version with that "SD_MMC.begin" hack enabled?

BTW, if you need a heavy handed tool for debugging hw stuff raspberry pi pico based logic analyzer is what I use. I took me quite a bit of banging head to wall to get https://github.com/pico-coder/sigrok-pico compiled and working on linux though. You should consider which rpi pico project you use based on what os(windows, linux) you have. 3.6v zener diodes and one resistor for each channel is the cheapest/easiest way of making it 5v tolerant as I see it.

AapoTahkola, 2022-10-05:
I get `Component esp32_camera_my cannot be loaded via YAML (no CONFIG_SCHEMA).` if I try to compile esp32_camera as external_component.

hugokernel, 2023-10-12:
I just retrying with others IO and the result is the same: There is no way to use the camera with the pulse_meter integration.