esphome/issues #3470: Installation Failure to ESP32-C3 (esp32-c3-devkitm-1) from Raspberry Pi 4 (64-bit)

Teleportist, 2022-07-26:
### The problem

I'm currently following this [guide](https://albert.nz/esp32-led-strip/)

Unfortunately the code fails to compile with the following log error;

INFO Reading configuration /config/esphome/wall-light-1.yaml...
WARNING GPIO8 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Generating C++ source...
INFO Compiling app...
Processing wall-light-1 (board: esp32-c3-devkitm-1; framework: espidf; platform: platformio/espressif32 @ 3.5.0)
--------------------------------------------------------------------------------
Tool Manager: Installing espressif/toolchain-xtensa-esp32s2 @ 8.4.0+2021r2-patch2
Error: Could not find the package with 'espressif/toolchain-xtensa-esp32s2 @ 8.4.0+2021r2-patch2' requirements for your system 'linux_aarch64'

### Which version of ESPHome has the issue?

2022.3.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.7.0

### What platform are you using?

ESP32-IDF

### Board

ESP32-C3 (ElectroDragon ESP LED Strip WIFI Control Board)

### Example YAML snippet

```yaml
esphome:
  name: wall-light-1
  platformio_options:
    board_build.flash_mode: dio

esp32:
  board: esp32-c3-devkitm-1
  variant: esp32c3
  framework:
    type: esp-idf
    version: latest

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: 'Wall Light Fallback Hotspot'
    password: !secret wifi_ap_password

logger:

api:

ota:
  password: !secret ota_password

light:
  - platform: cwww
    name: 'Wall light 1'
    cold_white: greenpin
    warm_white: redpin
    cold_white_color_temperature: 6000 K
    warm_white_color_temperature: 2700 K
    constant_brightness: true

output:
  - platform: ledc
    pin: GPIO5
    id: greenpin
    frequency: 19531Hz
  - platform: ledc
    pin: GPIO8
    id: redpin
    frequency: 19531Hz
  # - platform: ledc
  #   pin: GPIO12
  #   id: bluepin
  #   frequency: 19531Hz
  # - platform: ledc
  #   pin: GPIO14
  #   id: whitepin
  #   frequency: 19531Hz
```

### Anything in the logs that might be useful for us?

```txt
Tool Manager: Installing espressif/toolchain-xtensa-esp32s2 @ 8.4.0+2021r2-patch2
Error: Could not find the package with 'espressif/toolchain-xtensa-esp32s2 @ 8.4.0+2021r2-patch2' requirements for your system 'linux_aarch64'
```

### Additional information

I have been able to compile and install the code using another instance of ESPHome (Windows). This then shows the device as 'ONLINE' in the ESPHome Add-on, with it also being available in Home Assistant. But OTA updating is still not possible as the error persists.

Thanks

ssieb, 2022-08-01:
Espressif didn't provide a toolchain for 64-bit ARM for that version.  There's nothing we can do about it.  I think it might be resolved if you use the dev version of esphome because it looks like that's pulling `patch3` instead which I believe has an AARCH64 version.

closed 2022-08-01 by ssieb

Teleportist, 2022-08-02:
Thanks ssleb.

I've since given up on trying to employ esphome here, as getting MQTT to work reliably was also also proving troublesome.
But I do now have a working solution using Tasmota.

I'll certainly consider trying the dev version if need be in future.