esphome/issues #3471: mcp23x17 compile error

jjk0giap, 2022-07-27:
### The problem

linking error while compile esphome with MCP23017 i2c expander.

### Which version of ESPHome has the issue?

2022.6.2, 2022.7.0-dev

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.7.3

### What platform are you using?

ESP32

### Board

esp32-poe

### Component causing the issue

mcp23017

### Example YAML snippet

```yaml
i2c:
  scan: false
  frequency: 400kHz

mcp23017:
  - id: 'mcp23017_0'
    address: 0x20
  - id: 'mcp23017_1'
    address: 0x21
  - id: 'mcp23017_2'
    address: 0x22
  - id: 'mcp23017_3'
    address: 0x23
output:
  - platform: gpio
    id: ssr1_1
    pin:
      mcp23xxx: mcp23017_0
      number: 0
      mode: OUTPUT
      inverted: True
```

### Anything in the logs that might be useful for us?

```txt
Linking .pioenvs/main-lights-esp32/firmware.elf
.pioenvs/main-lights-esp32/src/esphome/components/mcp23x17_base/mcp23x17_base.cpp.o:(.rodata._ZTVN7esphome13mcp23x17_base12MCP23X17BaseE[vtable for esphome::mcp23x17_base::MCP23X17Base]+0x48): undefined reference to `esphome::mcp23xxx_base::MCP23XXXBase::read_reg(unsigned char, unsigned char*)'
.pioenvs/main-lights-esp32/src/esphome/components/mcp23x17_base/mcp23x17_base.cpp.o:(.rodata._ZTVN7esphome13mcp23x17_base12MCP23X17BaseE[vtable for esphome::mcp23x17_base::MCP23X17Base]+0x4c): undefined reference to `esphome::mcp23xxx_base::MCP23XXXBase::write_reg(unsigned char, unsigned char)'
collect2: error: ld returned 1 exit status
*** [.pioenvs/main-lights-esp32/firmware.elf] Error 1
```

### Additional information

worked fine on 2021.10.2 pip install

jjk0giap, 2022-07-29:
hint
I modiied mcp23xxx_base.h like below and code compile and work with no error:
` protected:
  // read a given register
  virtual bool read_reg(uint8_t reg, uint8_t *value) = 0;
  // write a value to a given register
  virtual bool write_reg(uint8_t reg, uint8_t value) = 0;
  // update registers with given pin value.
  virtual void update_reg(uint8_t pin, bool pin_value, uint8_t reg_a) = 0;`

ssieb, 2022-08-01:
It compiles fine for me.  Try doing a clean build.

jjk0giap, 2022-08-01:
i've tried clean build on three different platforms: HassOS on VM, docker with dev and pip install on different machines - always with compile error.

ssieb, 2022-08-02:
I just tried a clean install of Fedora 35, pip install esphome, and no compile error.

ssieb, 2022-08-02:
Can you paste your entire yaml?  Maybe there's something else.

jjk0giap, 2022-08-02:
pip install on windows machine
`esphome:
  name: main-lights-esp32
  platform: ESP32
  board: esp32-poe

external_components:
  - source: my_components
  
ethernet:
  type: LAN8720
  mdc_pin: GPIO23
  mdio_pin: GPIO18
  clk_mode: GPIO17_OUT
  phy_addr: 0
  power_pin: GPIO12

  manual_ip:
    static_ip: 192.168.1.18
    gateway: 192.168.1.1
    subnet: 255.255.255.0

debug:

logger:
  level: DEBUG

api:
 reboot_timeout: 0s

ota:

i2c:
  scan: false
  frequency: 400kHz

http_request:
  useragent: esphome/device
  timeout: 5s

uart:
  id: uart_bus
  baud_rate: 250000
  tx_pin: 4
  stop_bits: 2
  
dmx512:
  id: dmx
  uart_id: uart_bus
  tx_pin: 4
  uart_num: 1

media_player:
  - platform: i2s_audio
    name: Głośnik korytarz
    dac_type: external
    i2s_lrclk_pin: GPIO14
    i2s_bclk_pin: GPIO15
    i2s_dout_pin: GPIO32
    mode: stereo

globals:
  - id: light_mode_jadalnia
    type: int
    restore_value: yes
    initial_value: '3'
  - id: light_mode_kanapa
    type: int
    restore_value: yes
    initial_value: '3'
[... 1717 lines cut ...]
        - ON for at most 1s
        - OFF for at least 0.2s
        then:
          - http_request.get: http://192.168.1.21/win&PL=1
      - timing:
        - ON for at least 1s
        then:
          - http_request.get: http://192.168.1.21/win&PL=~
      - timing:
        - ON for at most 1s
        - OFF for at least 0.2s
        then:
          - http_request.get: http://192.168.1.21/win&T=2

  - platform: gpio
    name: "wiatrołap (korytarz)"
    pin:
      mcp23xxx: mcp23017_0
      number: 11
      mode: INPUT_PULLUP
      inverted: True
    on_press:
      then:
        - light.toggle: light_23
`

ssieb, 2022-08-02:
You didn't restrict the components from your external components directory.  Do you maybe have an mcp directory in there as well?

jjk0giap, 2022-08-02:
No, only dmx component there. As I said, after some tweaks in mcp23xxx_base.h this code compile with no error.

ssieb, 2022-08-02:
It compiles fine for me everywhere I test it and no one else has mentioned this problem.  I guess you can keep your tweaks.

jjk0giap, 2022-08-02:
Full log from HA addon
[logs_main-lights-esp32_compile.txt](https://github.com/esphome/issues/files/9240604/logs_main-lights-esp32_compile.txt)

closed 2022-12-16 by github-actions[bot]