esphome/issues #3472: Compilation IR rc6 error

jmigege, 2022-07-29:
### The problem

Hello,

When compiling a project with an rc6 binary, there are errors.

### Which version of ESPHome has the issue?

2022.6.2

### What type of installation are you using?

pip

### What platform are you using?

ESP32

### Board

mhetesp32minikit

### Component causing the issue

remote_base

### Example YAML snippet

```yaml
substitutions:
  devicename: telecommande
  upper_devicename: Telecommande
  
        
esphome:
  name: $devicename
  platform: ESP32
  board: mhetesp32minikit

logger:

button:
  - platform: restart
    name: "ESP${upper_devicename} Reboot"
      
binary_sensor:
  - platform: remote_receiver
    name: "Power"
    rc6:
      address: 0x00
      command: 0x0C

       
remote_receiver:
  pin:
    number: GPIO16
    inverted: True
    mode: INPUT_PULLUP  
  dump: rc6
```

### Anything in the logs that might be useful for us?

```txt
Log:
INFO Reading configuration EspTelecommande.yaml...
INFO Generating C++ source...
Traceback (most recent call last):
  File "/home/maison/.local/bin/esphome", line 8, in <module>
    sys.exit(main())
  File "/home/maison/.local/lib/python3.9/site-packages/esphome/__main__.py", line 931, in main
    return run_esphome(sys.argv)
  File "/home/maison/.local/lib/python3.9/site-packages/esphome/__main__.py", line 918, in run_esphome
    rc = POST_CONFIG_ACTIONS[args.command](args, config)
  File "/home/maison/.local/lib/python3.9/site-packages/esphome/__main__.py", line 330, in command_compile
    exit_code = write_cpp(config)
  File "/home/maison/.local/lib/python3.9/site-packages/esphome/__main__.py", line 163, in write_cpp
    generate_cpp_contents(config)
  File "/home/maison/.local/lib/python3.9/site-packages/esphome/__main__.py", line 175, in generate_cpp_contents
    CORE.flush_tasks()
  File "/home/maison/.local/lib/python3.9/site-packages/esphome/core/__init__.py", line 613, in flush_tasks
    self.event_loop.flush_tasks()
  File "/home/maison/.local/lib/python3.9/site-packages/esphome/coroutine.py", line 245, in flush_tasks
    next(task.iterator)
  File "/home/maison/.local/lib/python3.9/site-packages/esphome/__main__.py", line 155, in wrapped
    await coro(conf)
  File "/home/maison/.local/lib/python3.9/site-packages/esphome/components/remote_receiver/binary_sensor.py", line 9, in to_code
    var = await remote_base.build_binary_sensor(config)
  File "/home/maison/.local/lib/python3.9/site-packages/esphome/components/remote_base/__init__.py", line 221, in build_binary_sensor
    await builder(var, config)
  File "/home/maison/.local/lib/python3.9/site-packages/esphome/coroutine.py", line 87, in coro
    res = func(*args, **kwargs)
  File "/home/maison/.local/lib/python3.9/site-packages/esphome/components/remote_base/__init__.py", line 747, in rc6_binary_sensor
    ("device", config[CONF_DEVICE]),
KeyError: 'device'
--------------------------------
when removing line 747 from the __init__.py file of the remote_base component. This compiles a bit further but with other errors.
```

ssieb, 2022-08-01:
The binary sensor was clearly never tested...

closed 2022-08-02 by jesserockz

jmigege, 2022-08-02:
thanks for the quick resolution.