esphome/issues #3474: ws2812b led strip effect resets to none after led state changed

mrunmaydarpan, 2022-07-30:
### The problem

when ever i turn on the led strip the effect gets reset to NONE, its quite annoying that every time i have to change the effect. please give me any solution..

### Which version of ESPHome has the issue?

v2022.6.2

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

2022.7.7

### What platform are you using?

ESP8266

### Board

d1-mini

### Component causing the issue

light

### Example YAML snippet

```yaml
substitutions:
  deviceName: Ceiling Light
  led_1: GPIO14
  led_2: GPIO12
  led: GPIO02
  
esphome:
  name: ceiling-light
    
esp8266:
  board : d1_mini
  framework:
    version: 2.7.4

logger:
# api:
mqtt:
  broker: homeassistant.local
  username: hassio
  password: darkmaster
  id: mqtt_client
ota:
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap: {}
captive_portal:
web_server: 
  port: 80
  
light:
  - platform: fastled_clockless
    chipset: ws2812b
    pin: $led_1
    num_leds: 50
    rgb_order: RGB
    name: $deviceName
    effects:
      - addressable_rainbow:
      - addressable_rainbow:
          name: Rainbow Effect With Custom Values
          speed: 10
          width: 100
      - addressable_rainbow:
          name: Rainbow mega fast
          speed: 20
          width: 100
      - addressable_twinkle:
      - addressable_fireworks:
    on_turn_on:
      - logger.log: "Light Turned On!"
        
status_led:
  pin:
    number: $led
    inverted: false
```

jsuanet, 2022-08-03:
Add an `id: my_led`  to the light part and set the default effect in the `on_turn_on` action

```yaml
    on_turn_on:
      - logger.log: "Light Turned On!"
      - light.turn_on: 
          id: my_led
          effect: "Rainbow mega fast"
          brightness: 75%
```

mrunmaydarpan, 2022-08-04:
thank you very much,

it works exactly like i wanted...

jsuanet, 2022-08-04:
@mrunmaydarpan Fine that it works as you wanted. Please close this issue because this is not a bug/issue in ESPHome.

closed 2022-12-03 by ssieb