esphome/issues #3476: Sky Fan DC Tuya has alternative data types for some datapoints

gid204, 2022-07-31:
### The problem

The Skyfan DC Tuya integrated fan has some datapoints with differing data types to that of what ESPHome is expecting.

### Which version of ESPHome has the issue?

2022.6.2

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP8266

### Board

esp8265

### Component causing the issue

tuya

### Example YAML snippet

```yaml
My current config which has the fan, light switch and dimmer operating:

esphome:
  name: sam-fan
  platform: ESP8266
  board: esp8285
external_components:
  - source:
      type: git
      url: https://github.com/jeggleston1981/esphome
      ref: dev
    components: [ tuya ]

# Enable Home Assistant API
api:
# Logger not using the serial port as tuya needs it
logger:
  baud_rate: 0
  
ota:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  

uart:
  rx_pin: GPIO3
  tx_pin: GPIO1
  baud_rate: 9600
  
tuya:

fan:
  - platform: "tuya"
    name: "Sam Fan"
    switch_datapoint: 1
    speed_datapoint: 3
    direction_datapoint: 8
    speed_count: 6
    
light:
  - platform: "tuya"
    name: "Sam Light"
    switch_datapoint: 15
    dimmer_datapoint: 16
```

### Anything in the logs that might be useful for us?

```txt
datapoint list:

[16:12:51][C][tuya:033]: Tuya:
[16:12:51][C][tuya:050]:   Datapoint 16: int value (value: 0)
[16:12:51][C][tuya:048]:   Datapoint 15: switch (value: OFF)
[16:12:51][C][tuya:048]:   Datapoint 1: switch (value: OFF)
[16:12:51][C][tuya:054]:   Datapoint 2: enum (value: 0)
[16:12:51][C][tuya:050]:   Datapoint 3: int value (value: 0)
[16:12:51][C][tuya:054]:   Datapoint 8: enum (value: 0)
[16:12:51][C][tuya:054]:   Datapoint 19: enum (value: 2)
[16:12:51][C][tuya:054]:   Datapoint 22: enum (value: 0)
[16:12:51][C][tuya:068]:   Product: '{"p":"pktxz1vynowmavuc","v":"1.0.0","m":2}'
```
```

### Additional information

Currently the fan works by using the the external component: https://github.com/jeggleston1981/esphome
Which changes an enum to int: https://github.com/jeggleston1981/esphome/blob/dev/esphome/components/tuya/fan/tuya_fan.cpp#L83

Datapoint 19 is the color temperature which is expecting an enum (not an int).

It would be great to have the ability to configure these options.

ssieb, 2022-08-01:
The issue here is that the tuya fan component expects an enum for the speed, but the device has an integer.
And the tuya light component expects an integer for the color temperature, but the device has an enum.

gid204, 2022-11-11:
Hey @ssieb, have you had a chance to look at this?

memphisraynz, 2023-09-27:
@gid204 I cam across this as a way to add the light to possibly the same fan. Did you ever get to a solution?

gid204, 2023-09-28:
I did not unfortunately, I’ve just continued using the external component from [this repo](https://github.com/jeggleston1981/skyfandc) .

memphisraynz, 2023-09-28:
Thanks, I'm stil lgoing to buy one of those modules and see what I can do.
I see this person has a similar fan. Going to use it as a starting point.
https://community.home-assistant.io/t/arlec-grid-connect-ceiling-fans/261127

memphisraynz, 2023-10-14:
I got my module installed during the week.
It came with new config referencing https://github.com/jeggleston1981/skyfandc

`substitutions:
  name: easy-skyfandc-f17747
packages:
  jimmy.easy_skyfandc: github://jeggleston1981/skyfandc/micro_skyfan2.yaml
esphome:
  name: ${name}
  name_add_mac_suffix: false
api:
  encryption:
    key: Yth4pNq***********************mZjq7Y=

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password`

maddengithub, 2026-11-15:
[image]