esphome/issues #3477: Code Won't Compile ESP01

Morbidbystandard, 2022-08-01:
### The problem

Code won't install
Error: Could not find the package with 'platformio/tool-scons @ ~4.40200.0' requirements for your system 'linux_x86_64'

### Which version of ESPHome has the issue?

v2011.11.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.7.7

### What platform are you using?

ESP8266

### Board

ESP-01

### Component causing the issue

i2c

### Example YAML snippet

```yaml
api:

ota:

i2c:
  sda: 0
  scl: 2
  scan: True
  id: bus_a

sensor:
  - platform: ezo
    id: ec_ezo
    name: "EC"
    i2c_id: bus_a
    address: 100
    accuracy_decimals: 0
    unit_of_measurement: "uS"
    update_interval: 10s
  
#- platform: template 
#  name: "temperature compensation"
#  lambda: |-
#    return id(ec_ezo).set_tempcomp_value(id(rtd_ezo).state);

  - platform: ezo
    id: rtd_ezo
    name: "RTD Temperature"
    i2c_id: bus_a
    address: 102
    accuracy_decimals: 1
    unit_of_measurement: "°F"
    update_interval: 10s
    filters:
      offset: -0.35
    on_value:
      then:
      - lambda: id(ec_ezo).set_tempcomp_value(x);
  
#- platform: ezo
#  id: flow_ezo
#  name: "Flow Sensor"
#  address: 104
#  accuracy_decimals: 2
#  unit_of_measurement: "l/min"
#  update_interval: 1s
```

### Anything in the logs that might be useful for us?

```txt
Error: Could not find the package with 'platformio/tool-scons @ ~4.40200.0' requirements for your system 'linux_x86_64'
```

### Additional information

I'm running VM on a windows 10 host

ssieb, 2022-08-01:
You're using a very old version of esphome.  If you can reproduce the issue with the current version, then reopen it.

closed 2022-08-01 by ssieb

Morbidbystandard, 2022-08-02:
So, how is ESPhome this broken? Where are the updates?

ssieb, 2022-08-02:
I don't know how yours is so old.  I guess it depends on how you added it originally.
https://esphome.io/changelog/2022.4.0.html#home-assistant-addon-repo
You need to add the https://github.com/esphome/home-assistant-addon repo.  Delete your current addon and install the new one.

Ih8rain2, 2022-08-23:
I have the same issue with the latest release of ESPhome. previously no issues.