esphome/issues #3480: Midea IR has incorrect mode toggles when transmitting and using receiver

rdbahm, 2022-08-04:
### The problem

When using the Midea Climate IR component with a receiver, the receiver is not suppressed while transmitting. As a result, toggle values (such as "turbo/boost") are not reflected on the UI.

Transmitted from an actual remote:
A2.09.FF.FF.FF.74 (Turbo button on RG57A6 remote, as detected by an ESPHome device)

Results in the state:
Preset: BOOST
On an ESPHome device with the Midea_IR platform.

Sending "Boost" from Midea_IR:
A2.09.FF.FF.FF.74 (as detected by a second ESPHome device)
Results in two back-to-back states:
Preset: BOOST
Preset: NONE
This appears to be caused by the state being changed to "boost" because the state is being sent, then changing back to "none" as the receiver sees the "boost" command that we just transmitted and toggling back to "none."

Disabling the IR receiver (by unplugging it from the ESP) results in the expected behavior: The "preset" switches to "boost" and stays there.

I expect, though haven't tested, that this general issue may affect other IR Climate integrations which have a receive component and a toggle-type IR command.

### Which version of ESPHome has the issue?

2022.6.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.8

### What platform are you using?

ESP32

### Board

ESP32DEV

### Component causing the issue

midea_ir

### Example YAML snippet

```yaml
remote_transmitter:
  id: ir_tx
  pin:
    number: 26
  carrier_duty_percent: 50%

remote_receiver:
  pin:
    number: 27
    inverted: true
  dump: midea
  id: ir_rx

sensor:
  - platform: homeassistant
    id: room_temperature
    entity_id: sensor.living_room_temperature_aggregate
    internal: true
    unit_of_measurement: '°C'
    filters:
      - throttle: 10s
      - heartbeat: 2min
      - debounce: 1s
      - lambda: return (x - 32) / (9.0/5.0);

climate:
  - platform: midea_ir
    name: "Living Room Heat Pump"
    receiver_id: ir_rx
    use_fahrenheit: true
    sensor: room_temperature
    visual:
      min_temperature: 62 °F
      max_temperature: 86 °F
      temperature_step: 1
```

the-mentor, 2022-10-06:
Looks like I'm experiencing similar behavior on whirlpool ac.

When i try to power on /off the unit it behaves strangely and when I disable the receiver in the ir climate config everything works as expected except the receiver stuff. 

Seems like the remote doesn't ignore the command it just sent and confuses itself.

the-mentor, 2022-10-07:
By acting strangely it looks like it sends a power off command and then switches to heat right away.
If I had to guess it doesn't ignore its own outgoing blast.
Why it switches to heat is beyond me.

I wonder what we can do to fix if since this issue exists for over a 2 months.

the-mentor, 2023-02-25:
@jesserockz sounds like this can be a quick fix any idea where i should look in the code to try and narrow it down ?

TokarevSergey, 2023-07-02:
Добрый день, у меня есть кондиционер Galactic в корпусе Whirlpool. У меня такая же проблема, как у вас, подскажите, пожалуйста, как решить?