esphome/issues #3481: I2S Audio not available for IDF platform

ezpl, 2022-08-04:
### The problem

Using the IDF platform for an ESP32-S2 chip, ESPHome reports that I2S Audio / external DAC is not available 

``` yaml
# This feature is only available with frameworks ['arduino'].
  platform: i2s_audio
    id: mediaplayer_01_media_out
    name: MediaPlayer
    dac_type: external
    i2s_lrclk_pin: GPIO5
    i2s_dout_pin: GPIO18
    i2s_bclk_pin: GPIO4
    mode: stereo
```

Pinout is changed to what is suggested in Espressif documentation: 
https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/api-reference/peripherals/i2s.html

### Which version of ESPHome has the issue?

2022.6.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.8

### What platform are you using?

ESP32-IDF

### Board

Wemos Mini S2

### Component causing the issue

media_player / i2s_audio

### Example YAML snippet

```yaml
improv_serial:

media_player:
  - platform: i2s_audio
    id: "${id_prefix}_media_out"
    name: ${esp_name}
    dac_type: external
    i2s_lrclk_pin: GPIO5 # Word Select or Left/Right Clock
    i2s_dout_pin: GPIO18 # Data Out
    i2s_bclk_pin: GPIO4 # Bit Clock
```

JeeCee1, 2022-09-10:
I face the same problem. Combining
```
media_player: 
  - platform: i2s_audio
```
with
`bluetooth_proxy:`

and some other simple sensors gives a 99.9% full flash, which leads to crashes.
The `idf` framework is much more compact.

ssieb, 2022-09-11:
This is because the media player uses the Arduino player library.  There is very little esphome code involved.  You would have to port or rewrite the player library for idf.  This isn't really an issue, it's a feature request.

JeeCee1, 2022-09-13:
I wrote a [FR].(https://github.com/esphome/feature-requests/issues/1864) 

@ezpl I think we can close this issue?

closed 2022-09-13 by ezpl