esphome/issues #3485: BLE makes Shelly Plus 1 unresponsive

Adi146, 2022-08-06:
### The problem

When you enable `esp32_ble_tracker` on a single core ESP32 like the Shelly Plus 1, the device get very unresponsive. It is still working but it is very slow. You can see this when toggling the switch on the shelly that it takes a while until it is actually switched or when pinging the device you can see answers that take up to 2000ms. 

Additionally OTA crashes the device when you try to flash a firmware with `esp32_ble_trakcer` enabled (see logs)

### Which version of ESPHome has the issue?

2022.6.2

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

2022.7.7

### What platform are you using?

ESP32-IDF

### Board

Shelly plus 1

### Component causing the issue

esp32_ble_tracker

### Example YAML snippet

```yaml
esphome:
  name: shelly
  platformio_options:
    board_build.f_cpu: 160000000L

esp32:
  board: esp32doit-devkit-v1
  framework:
    type: esp-idf
    sdkconfig_options:
      CONFIG_FREERTOS_UNICORE: y
      CONFIG_ESP32_DEFAULT_CPU_FREQ_160: y
      CONFIG_ESP32_DEFAULT_CPU_FREQ_MHZ: "160"

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: ""

ota:
  password: ""

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

esp32_ble_tracker:
```

### Anything in the logs that might be useful for us?

```txt
[16:34:16][D][ota:139]: Starting OTA Update from ::FFFF:192.168.10.207...
[16:34:21]E (59417) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
[16:34:21]E (59417) task_wdt:  - loopTask (CPU 0)
[16:34:21]E (59417) task_wdt: Tasks currently running:
[16:34:21]E (59417) task_wdt: CPU 0: loopTask
[16:34:21]E (59417) task_wdt: Aborting.
[16:34:21]
[16:34:21]abort() was called at PC 0x400e81a4 on core 0
[16:34:22]
[16:34:22]Backtrace:0x40081952:0x3ffb07e0 0x400886fd:0x3ffb0800 0x4008e3fe:0x3ffb0820 0x400e81a4:0x3ffb0890 0x4008237e:0x3ffb08b0 0x4008180c:0x3ffbb180 0x40082c03:0x3ffbb1a0 0x40083975:0x3ffbb1c0 0x40083a7a:0x3ffbb1e0 0x400868a5:0x3ffbb200 0x400836ca:0x3ffbb220 0x4010184f:0x3ffbb270 0x400e42a6:0x3ffbb290 0x400d8109:0x3ffbb2c0 0x400d89b6:0x3ffbb2e0 0x400d8c06:0x3ffbb790 0x4014b081:0x3ffbb7b0 0x4014b141:0x3ffbb7d0 0x400dc54b:0x3ffbb7f0 0x400de4da:0x3ffbb820 0x400d6b2a:0x3ffbb840
WARNING Found stack trace! Trying to decode it
WARNING Decoded 0x40081952: esp_intr_noniram_enable at /home/adrian/.platformio/packages/framework-espidf/components/esp_system/intr_alloc.c:812
WARNING Decoded 0x400886fd: r_lld_evt_restart
WARNING Decoded 0x4008e3fe: _xt_panic at /home/adrian/.platformio/packages/framework-espidf/components/esp_system/port/arch/xtensa/panic_handler_asm.S:46
WARNING Decoded 0x400e81a4: btm_find_dev at /home/adrian/.platformio/packages/framework-espidf/components/bt/host/bluedroid/stack/btm/btm_dev.c:510
WARNING Decoded 0x4008237e: _xt_coproc_exc at /home/adrian/.platformio/packages/framework-espidf/components/freertos/port/xtensa/xtensa_vectors.S:887
WARNING Decoded 0x4008180c: esp_random at /home/adrian/.platformio/packages/framework-espidf/components/esp32/hw_random.c:55
WARNING Decoded 0x40082c03: esp_reent_init at /home/adrian/.platformio/packages/framework-espidf/components/newlib/reent_init.c:40
WARNING Decoded 0x40083975: esp_flash_read at /home/adrian/.platformio/packages/framework-espidf/components/spi_flash/esp_flash_api.c:650
WARNING Decoded 0x40083a7a: esp_flash_write at /home/adrian/.platformio/packages/framework-espidf/components/spi_flash/esp_flash_api.c:711
WARNING Decoded 0x400868a5: ld_inq_frm_isr_wrapper
WARNING Decoded 0x400836ca: esp_flash_init at /home/adrian/.platformio/packages/framework-espidf/components/spi_flash/esp_flash_api.c:237 (discriminator 1)
WARNING Decoded 0x4010184f: bta_sys_register at /home/adrian/.platformio/packages/framework-espidf/components/bt/host/bluedroid/bta/sys/bta_sys_main.c:523
WARNING Decoded 0x400e42a6: btm_ble_refresh_local_resolvable_private_addr at /home/adrian/.platformio/packages/framework-espidf/components/bt/host/bluedroid/stack/btm/btm_ble_addr.c:612
WARNING Decoded 0x400d8109: esphome::mdns::MDNSTXTRecord::~MDNSTXTRecord() at /home/adrian/sources/.esphome/build/shelly-bathroom-mirror/src/esphome/components/mdns/mdns_component.h:10
WARNING Decoded 0x400d89b6: esphome::ntc::NTC::setup() at /home/adrian/sources/.esphome/build/shelly-bathroom-mirror/src/esphome/components/ntc/ntc.cpp:11
WARNING Decoded 0x400d8c06: esphome::ota::OTAComponent::setup() at /home/adrian/sources/.esphome/build/shelly-bathroom-mirror/src/esphome/components/ota/ota_component.cpp:76
WARNING Decoded 0x4014b081: hci_vendor_ble_update_adv_report_flow_control_cmd_handler at dbg_task.c
WARNING Decoded 0x4014b141: hci_vendor_ble_update_duplicate_exceptional_list_cmd_handler at dbg_task.c
WARNING Decoded 0x400dc54b: esphome::optional<bool>::optional() at /home/adrian/sources/.esphome/build/shelly-bathroom-mirror/src/esphome/core/optional.h:43
 (inlined by) esphome::wifi::WiFiComponent::wifi_scan_start_() at /home/adrian/sources/.esphome/build/shelly-bathroom-mirror/src/esphome/components/wifi/wifi_component_esp_idf.cpp:725
WARNING Decoded 0x400de4da: esphome::Controller::setup_controller(bool) at /home/adrian/sources/.esphome/build/shelly-bathroom-mirror/src/esphome/core/controller.cpp:33 (discriminator 1)
WARNING Decoded 0x400d6b2a: esphome::api::ListEntitiesIterator::on_switch(esphome::switch_::Switch*) at /home/adrian/sources/.esphome/build/shelly-bathroom-mirror/src/esphome/components/api/list_entities.cpp:28
[16:34:22]
[16:34:22]
[16:34:22]ELF file SHA256: 92dfb64ce660cc48
[16:34:22]
[16:34:22]Rebooting...
[16:34:22]ets Jul 29 2019 12:21:46
[16:34:22]
[16:34:22]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[16:34:22]configsip: 188777542, SPIWP:0xee
[16:34:22]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[16:34:22]mode:DIO, clock div:2
[16:34:22]load:0x3fff0030,len:7568
[16:34:22]load:0x40078000,len:14912
[16:34:22]load:0x40080400,len:3688
[16:34:22]entry 0x4008067c
[16:34:22]I (28) boot: ESP-IDF 4.3.2 2nd stage bootloader
[16:34:22]I (28) boot: compile time 15:51:09
[16:34:22]I (28) boot: chip revision: 3
[16:34:22]I (31) boot_comm: chip revision: 3, min. bootloader chip revision: 0
[16:34:22]I (38) boot.esp32: SPI Speed      : 40MHz
[16:34:22]I (42) boot.esp32: SPI Mode       : DIO
[16:34:22]I (47) boot.esp32: SPI Flash Size : 4MB
[16:34:22]I (51) boot: Enabling RNG early entropy source...
[16:34:22]I (57) boot: Partition Table:
[16:34:22]I (60) boot: ## Label            Usage          Type ST Offset   Length
[16:34:22]I (68) boot:  0 nvs              WiFi data        01 02 00009000 00004000
[16:34:22]I (75) boot:  1 otadata          OTA data         01 00 0000d000 00002000
[16:34:22]I (83) boot:  2 phy_init         RF data          01 01 0000f000 00001000
[16:34:22]I (90) boot:  3 app0             OTA app          00 10 00010000 001c0000
[16:34:22]I (98) boot:  4 app1             OTA app          00 11 001d0000 001c0000
[16:34:22]I (105) boot: End of partition table
[16:34:22]I (109) boot_comm: chip revision: 3, min. application chip revision: 0
[16:34:22]I (116) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=19c04h (105476) map
[16:34:22]I (163) esp_image: segment 1: paddr=00029c2c vaddr=3ffb0000 size=035c8h ( 13768) load
[16:34:22]I (169) esp_image: segment 2: paddr=0002d1fc vaddr=40080000 size=02e1ch ( 11804) load
[16:34:22]I (174) esp_image: segment 3: paddr=00030020 vaddr=400d0020 size=7f6b8h (521912) map
[16:34:22]I (366) esp_image: segment 4: paddr=000af6e0 vaddr=40082e1c size=10d14h ( 68884) load
[16:34:22]I (395) esp_image: segment 5: paddr=000c03fc vaddr=50000000 size=00010h (    16) load
[16:34:22]I (405) boot: Loaded app from partition at offset 0x10000
[16:34:22]I (405) boot: Disabling RNG early entropy source...
[16:34:22]I (417) cpu_start: Pro cpu up.
[16:34:22]I (417) cpu_start: Single core mode
[16:34:22]I (426) cpu_start: Pro cpu start user code
[16:34:22]I (426) cpu_start: cpu freq: 160000000
```

michalmo, 2022-09-14:
I came across these problems too.

The OTA issue is because the firmware image (with ble included) is so large, that preparing the OTA partition takes longer than 5s and the task watch dog timer gets triggered. On my Shelly Plus 1PM it takes just over 5s, just longer than the watchdog timer. You can increase the timeout by adding `CONFIG_ESP_TASK_WDT_TIMEOUT_S: "10"` under `sdkconfig_options`.

In my case unresponsiveness was caused by a sensor that was set to update too frequently, and relieved by adjusting `update_interval`.  You might want to try adjusting the `scan_parameters` timings of the `esp32_ble_tracker` component to see if that helps your case.

Adi146, 2022-10-14:
Setting the `CONFIG_ESP_TASK_WDT_TIMEOUT_S: "10"`fixed the problem with the ota update.

I also tried to increase the scan interval with 
```
esp32_ble_tracker:
  scan_parameters:
    interval: 1min
```
this did help a bit so that the shelly doesn't crash immediately but when there are a bit more bt events it still seem to crash. Especially when enabling the new `bluetooth_proxy` the increased scan interval doesn't help at all and the shelly crashes instantly. I also think increasing the scan interval is no valid option as you might miss some events when doing this.

sanderlv, 2022-11-24:
Bump
```

esp32_ble_tracker:
#  scan_parameters:
#    interval: 30ms
    #active: false
bluetooth_proxy:

```
```
[15:40:15]Rebooting...
[15:40:15]ets Jul 29 2019 12:21:46
[15:40:15]
[15:40:15]rst:0xc (SW_CPU_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
[15:40:15]configsip: 188777542, SPIWP:0xee
[15:40:15]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[15:40:15]mode:DIO, clock div:2
[15:40:15]load:0x3fff0018,len:4
[15:40:15]load:0x3fff001c,len:1044
[15:40:15]load:0x40078000,len:8896
[15:40:15]load:0x40080400,len:5828
[15:40:15]entry 0x400806ac
[15:40:16]I (474) cpu_start: Pro cpu up.
[15:40:16]I (474) cpu_start: Single core mode
[15:40:16]I (482) cpu_start: Pro cpu start user code
[15:40:16]I (482) cpu_start: cpu freq: 160000000
[15:40:16]I (482) cpu_start: Application information:
[15:40:16]I (487) cpu_start: Project name:     shelly-plus-1pm-01
[15:40:16]I (492) cpu_start: App version:      2022.11.2
[15:40:16]I (498) cpu_start: Compile time:     Nov 24 2022 15:33:39
[15:40:16]I (504) cpu_start: ELF file SHA256:  47210795cec1bcfd...
[15:40:16]I (510) cpu_start: ESP-IDF:          4.3.2
[15:40:16]I (515) heap_init: Initializing. RAM available for dynamic allocation:
[15:40:16]I (522) heap_init: At 3FF80000 len 00002000 (8 KiB): RTCRAM
[15:40:16]I (528) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
[15:40:16]I (534) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
[15:40:16]I (540) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
[15:40:16]I (546) heap_init: At 3FFC9E70 len 00016190 (88 KiB): DRAM
[15:40:16]I (552) heap_init: At 3FFE0440 len 0001FBC0 (126 KiB): D/IRAM
[15:40:16]I (559) heap_init: At 40078000 len 00008000 (32 KiB): IRAM
[15:40:16]I (565) heap_init: At 4009C084 len 00003F7C (15 KiB): IRAM
[15:40:16]I (572) spi_flash: detected chip: generic
[15:40:16]I (576) spi_flash: flash io: dio
[15:40:16]I (581) cpu_start: Starting sche[I][logger:258]: Log initialized
[15:40:16][I][esp-idf:000]: I (606) gpio: GPIO[0]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
[15:40:16]
[15:40:16][I][app:029]: Running through setup()...
[15:40:16][I][esp-idf:000]: I (615) gpio: GPIO[26]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
[15:40:16]
[15:40:16][I][esp-idf:000]: I (619) gpio: GPIO[4]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
[15:40:16]
[15:40:16][I][esp-idf:000]: I (620) gpio: GPIO[25]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
[15:40:16]
[15:40:16][I][esp-idf:000]: I (638) gpio: GPIO[32]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
[15:40:16]
[15:40:16][I][esp-idf:000]: I (646) gpio: GPIO[33]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
[15:40:16]
[15:40:16][I][esp-idf:000]: I (656) gpio: GPIO[23]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
[15:40:16]
[15:40:16][I][esp-idf:000]: I (666) gpio: GPIO[5]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
[15:40:16]
[15:40:16][I][esp-idf:000]: I (677) gpio: GPIO[18]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
[15:40:16]
[15:40:16][I][esp-idf:000]: I (688) BTDM_INIT: BT controller compile version [6a07b06]
[15:40:16]
[15:40:16][I][esp-idf:000]: I (708) system_api: Base MAC address is not set
[15:40:16]
[15:40:16][I][esp-idf:000]: I (717) system_api: read default base MAC address from EFUSE
[15:40:16]
[... 66 lines cut ...]
[15:40:17][I][esp-idf:000]: I (1514) wifi_init: WiFi RX IRAM OP enabled
[15:40:17]
[15:40:17][I][esp-idf:000]: I (1518) wifi:
[15:40:17][I][esp-idf:000]: mode : sta (a8:03:2a:b1:9a:18)
[15:40:17][I][esp-idf:000]: 
[15:40:17]
[15:40:17][I][esp-idf:000]: I (1535) wifi:
[15:40:17][I][esp-idf:000]: enable tsf
[15:40:17][I][esp-idf:000]: 
[15:40:17]
[15:40:17][I][esp-idf:000]: I (1547) wifi:
[15:40:17][I][esp-idf:000]: Set ps type: 1
[15:40:17]
[15:40:17][I][esp-idf:000]: 
[15:40:17]
[15:40:17][I][esp-idf:000]: W (1556) wifi:
[15:40:17][I][esp-idf:000]: Error! Should use default active scan time parameter for WiFi scan when Bluetooth is enabled!!!!!!
[15:40:17]
[15:40:17][I][esp-idf:000]: 
[15:40:17]
[15:40:17][I][esp-idf:000]: I (1572) wifi:
[15:40:17][I][esp-idf:000]: Set ps type: 1
[15:40:17]
[15:40:17][I][esp-idf:000]: 
```

fluxsmith, 2023-01-03:
esp_ble_beacon also crashes the Shelly, as I discovered after I put it back in the wall behind the switches ;-(

evlo, 2023-01-24:
esp32-s3-devkitc-1 with eps32-s3-wroom-2 does provide very similar results