esphome/issues #3487: Example code for Shelly Dimmer 2

svh1985, 2022-08-08:
### The problem

I'm trying to understand the new Shell Dimmer feature within ESPHome, but I'm a bit lost. As a platform within the light Component, I wonder how dimming works with the 2 input buttons. On all my other Shelly 2.5 devices I had to set up a binary_sensor to capture a button input and map actions to an output GPIO relay.

Is this now all handled internally by the shelly_dimmer platform? Or is there anything I have to setup myself which is not in the [example code](https://esphome.io/components/light/shelly_dimmer.html)? 

I would like to ask for a more extensive example. Preferably so it works just as the one button setting in the native Shelly firmware.

@rnauber as one of the components contributors, can you maybe point me into the right direction?

Thank you for helping out! 🙏

### Which version of ESPHome has the issue?

2022.6.3

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP8266

### Board

Shelly Dimmer 2

### Component causing the issue

shelly_dimmer

rnauber, 2022-08-11:
Hi @svh1985 ,

> ### The problem
> I'm trying to understand the new Shell Dimmer feature within ESPHome, but I'm a bit lost. As a platform within the light Component, I wonder how dimming works with the 2 input [...]

Yes, you have to code it yourself. `shelly_dimmer` only provides you with all the things a [light component ](https://esphome.io/components/light/index.html) can do, e.g. set the brightness. The logic of determining that brightness is up to you. 

> 
> I would like to ask for a more extensive example. Preferably so it works just as the one button setting in the native Shelly firmware.
> 

Well, I can try try helping you create one for your use case. So how does the original firmware control the light with one button? Does a long press adjust the brightness and a short press toggles the light? Then you could try something like this:

```
binary_sensor:
  - platform: ...
    id: btn
    on_press: # increase brightness
      then:
        - delay: 1 s
        - if:
              condition:
                -  binary_sensor.is_on: btn
              then:
                - light.turn_on:
                    id: combined_light
                    brightness: 100%
        - while:
              condition:
                -  binary_sensor.is_on: btn
              then:
                - light.dim_relative:
                    id: combined_light
                    relative_brightness: -5%
                    transition_length: 0.1s
                - delay: 0.1s
                
    - timing: # Single Short Clicked
        - ON for at most 1s
        - OFF for at least 0.5s
      then:
        - logger.log: "Single Short Clicked"
        - lambda: |- 
            auto call2 = id(combined_light).toggle();
            call2.perform();

```

> @rnauber as one of the components contributors, can you maybe point me into the right direction?
> 
> Thank you for helping out! pray

Best,
Richard
> ### Which version of ESPHome has the issue?
> 2022.6.3
>
> ### What type of installation are you using?
> Home Assistant Add-on
>
> ### Which version of Home Assistant has the issue?
> _No response_
>
> ### What [...]

svh1985, 2022-08-16:
Hi @rnauber

Awesome, thanks for the example it's much clearer for me know.

I will use this example in my project and will post the results!

svh1985, 2022-09-11:
I would like to prevent the dimmer from turning completely off when dimming down. It would be best to use a light.brightness condition as requested here https://github.com/esphome/feature-requests/issues/1860

But because it does not yet exist I'm trying to create it using a work-around.

If I'm using `id(dimmer).remote_values.get_brightness()` I'm getting a scale that's not the expected 0-1000 value.
@rnauber do you know a way to get the brightness (0-1000) value?

rnauber, 2022-09-13:
> If I'm using id(dimmer).remote_values.get_brightness() I'm getting a scale that's not the expected 0-1000 value.
> @rnauber do you know a way to get the brightness (0-1000) value?

I think get_brightness() will give you a float between 0.0 and 1.0. So just try multiplying it with 1000?

freakshock88, 2022-10-12:
@svh1985 Do you have a working example for this? I would also like to be able to toggle the relays with the physical light switch.

svh1985, 2022-10-12:
@freakshock88 

This is the code I use now:

```
esphome:
  name: dimmer
  platform: ESP8266
  board: esp01_1m
  comment: Shelly Dimmer 2

substitutions:
  device_name: "Dimmer"
  max_power: "220.0" # watt
  max_temp: "70.0" # °C

# Enable logging
logger:
    baud_rate: 0

uart:
    tx_pin: 1
    rx_pin: 3
    baud_rate: 115200

# Enable Home Assistant API
api:
  password: !secret ha_api_pass

ota:
  password: !secret ota_pass

wifi:
  networks:
  - ssid: !secret wifi_ssid1
    password: !secret wifi_pass1

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "${device_name} Fallback Hotspot"
    password: !secret ap_pass

text_sensor:
  - platform: wifi_info
    ip_address:
      name: "${device_name} IP Adres"
    ssid:
      name: "${device_name} SSID"
    bssid:
      name: "${device_name} BSSID"
      
  - platform: version
    name: "${device_name} ESPHome Version"

captive_portal:

# Enable Web Server
web_server:
  port: 80
  auth:
    username: !secret web_user
    password: !secret web_pass

time:
  - platform: homeassistant
    id: homeassistant_time

sensor:
  # NTC Temperature
  - platform: ntc
    sensor: temp_resistance_reading
    name: ${device_name} temperature
    unit_of_measurement: "°C"
    accuracy_decimals: 1
    icon: "mdi:thermometer"
    calibration:
      b_constant: 3350
      reference_resistance: 10kOhm
      reference_temperature: 298.15K
    on_value_range:
      - above: ${max_temp}
        then:
          - logger.log: "Switch turned off because temperature exceeded ${max_temp}°C"
          - homeassistant.service:
              service: persistent_notification.create
              data:
                title: Message from ${device_name}
              data_template:
                message: Switch turned off because temperature exceeded ${max_temp}°C
  - platform: resistance
    id: temp_resistance_reading
    sensor: temp_analog_reading
    configuration: DOWNSTREAM
    resistor: 32kOhm
  - platform: adc
    id: temp_analog_reading
    pin: A0

  - platform: total_daily_energy
    name: "${device_name} Energy"
    power_id: power_dimmer
    filters:
      # Multiplication factor from W to kW is 0.001
      - multiply: 0.001
    unit_of_measurement: kWh

switch:
  - platform: restart
    name: "${device_name} Herstart"
    
light:
  - platform: shelly_dimmer
    name: "${device_name}"
    id: dimmer
    icon: mdi:globe-light-outline
    power:
      name: "${device_name} Power"
      id: power_dimmer
    voltage:
      name: "${device_name} Voltage"
    current:
      name: "${device_name} Current"
    min_brightness: 460
    max_brightness: 1000
    firmware:
      version: "51.6"
      update: true

binary_sensor:
  - platform: gpio
    name: 'Switch 1'
    id: switchID_1
    pin:
      number: GPIO14
      mode: INPUT
    internal: true
    filters:
      - delayed_on_off: 50ms
    on_multi_click:
    
      # single click
      - timing:
        - ON for at most 1s
        - OFF for at least 0.5s
        then:
          - logger.log: "Single-Clicked"
          - light.toggle: dimmer

      # double click
      - timing: 
        - ON for at most 0.5s
        - OFF for at most 0.5s
        - ON for at most 0.5s
        - OFF for at least 0.2s
        then:
          - logger.log: "Double-Clicked"
          - light.turn_on:
              id: dimmer
              brightness: 100%

      # triple click
      - timing: 
        - ON for at most 0.5s
        - OFF for at most 0.5s
        - ON for at most 0.5s
        - OFF for at most 0.5s
        - ON for at most 0.5s
        - OFF for at least 0.2s
        then:
          - logger.log: "Triple-Clicked"
          - light.turn_on:
              id: dimmer
              brightness: 25%

      # Press and Hold button
      - timing: 
        - ON for at most 0.5s
        - OFF for at least 0.1s
        - ON for at least 1s
        then:
          - logger.log: "Press and Hold - Dim DOWN"
          - while:
                condition:
                  - binary_sensor.is_on: switchID_1
                then:
                  - light.dim_relative:
                      id: dimmer
                      relative_brightness: -1%
                      transition_length: 0.01s
                  - delay: 0.01s

      # Hold button
      - timing: 
        - ON for at least 1s
        then:
          - logger.log: "Hold - Dim UP"
          - while:
                condition:
                  - binary_sensor.is_on: switchID_1
                then:
                  - light.dim_relative:
                      id: dimmer
                      relative_brightness: +1%
                      transition_length: 0.01s
                  - delay: 0.01s

```

sanderlv, 2022-11-17:
> 

Going to try your code... can you tell met if I only need to use SW1 and SW1 as inputs? I think I only see SW1?

svh1985, 2022-12-31:
@sanderlv 

Then I think this would work for you, and then just use the single click to +1% and -1% relative_brightness as in the full example.

```
  - platform: gpio
    name: 'Switch 2'
    id: switchID_2
    pin:
      number: GPIO12
      mode: INPUT
    internal: true
    filters:
      - delayed_on_off: 50ms
    on_multi_click:
```

pimw1, 2023-05-07:
High, i've improved upon the solution. The most important one is that **the overheating securtiy contained a bug**: a on_value_range will not trigger when the device is booted while already overheating (it won't transition from outside the range into the range, since it is already in the range). This could happen when a child quickly disconnects and reconnects the dimmer on mains power while the device is already overheating. Not an unlikely scenario.

The solution below contains several features:

- several settings are explicitly mentioned to help you setup the light (gamma_correct, min_brightness, max_brightness, leading_edge, restore_mode
- light remembers brightness value when changing from off to on
- brightness cannot be changed when the light is off and the brightness cannot go below 0 or above 1000. Because of this, the "brigthness remembering" actually works in practice. And also because of this, the light won't shut down when dimming down.
- both physical buttons can be used to turn on/off the light. Long press initiates the dimming (one button for dimming up, the other button for dimming down). Press-release stops the dimming. The settings in the script are tuned for a full dimming in 5 seconds. You can change this to your liking. All processing is done locally on the dimmer itself.
- Additional robuustness is introduced: the dimming will stop automaticlly after 5 seconds, even if the stop-signal is missed (e.g. due to an issue with your physical switch or due to misconfiguration in Home Assistant). You change the duration to your liking.
- 3 virtual buttons (dim up, dim down, dim stop) are introduced to Home Assistant, so you can performa super smooth dimming from the Home Assistant front-end, automations and scripts. All with minimum network traffic and all of the logic performed locally on the dimmer.
```
esphome:
  name: choose_your_light_name
  friendly_name: ${device_name}
  comment: Shelly Dimmer 2

substitutions:
  device_name: "Name of the Light"
  light_name: "Light"
  max_temp: "70" # °C

esp8266:
  board: esp01_1m

# Enable logging
logger:
    baud_rate: 0

# Enable Home Assistant API
api:
  encryption:
    key: "your encryption key"

ota:
  password: "your ota key"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "ssid name"
    password: "password"

captive_portal:

uart:
    tx_pin: 1
    rx_pin: 3
    baud_rate: 115200

light:
    - platform: shelly_dimmer
      name: ${light_name}
      id: dimmer
      leading_edge: false #choose between leading edge and trailing edge (use trailing edge for led dimming)
      min_brightness: 470
      max_brightness: 1000
      restore_mode: ALWAYS_OFF
      default_transition_length: 1s
      gamma_correct: 0 #change this to your liking. Default value is 2.8, but 0 prevents some brightness pops for me
      firmware:
        version: "51.6"
        update: true

sensor: #Important: don't change this sensor-part unless you know what you are doing. These sensors will shut the light down when overheating temperature is reached.
  # NTC Temperature
  - platform: ntc
    sensor: temp_resistance_reading
[... 172 lines cut ...]
              - logger.log: "Physical dim_up release trigger"
              - script.stop: script_dim_up_timer
              - logger.log: "Script_dim_up_timer stopped"

button:
  - platform: template
    name: Dim Down
    on_press:
      then:
        - script.execute: script_dim_down_timer
        - script.execute: script_dim_down
  - platform: template
    name: Dim Up
    on_press:
      then:
        - script.execute: script_dim_up_timer
        - script.execute: script_dim_up
  - platform: template
    name: Dim Stop
    on_press:
      then:
        - logger.log: "Stopping timer script"
        - script.stop: script_dim_down_timer
        - script.stop: script_dim_up_timer
```

sanderlv, 2023-05-10:
Wow, this looks good! I tried to embed some in my code but I cannot easily find the bit and pieces. Would it be possible to add the "notes" in the esphome code? To describe what's for what?

pimw1, 2023-05-10:
Thanks sanderlv! Inline comments are added to the code.
At a very high level, the scripts are introduced are generic pieces of logic that are reutilized by several triggers (physical button presses and virtual button presses).
The script script_dim_down_timer are active for 5 seconds and are used as a condition for the while loop that is used in the script_dim_down and script_dim_up (otherwise, those scripts would run for eternity). 

Is there something specific that you would like to know?

PaperLawyer, 2024-01-20:
> Thanks sanderlv! Inline comments are added to the code. At a very high level, the scripts are generic pieces of logic that are reutilized by several triggers (physical button presses and virtual button presses). The script script_dim_down_timer/script_dim_up_timer are active for 5 seconds and are used as a condition for the while loop that is used in the script_dim_down and script_dim_up (otherwise, those scripts would run for eternity).
> 
> Is there something specific that you would like to know?

Can I jump in to thank you for the code?  It has been invaluable as I have converted a number of Shelly Dimmer 2 modules to Esphome.  I have one question which hopefully @pimw1 can answer - one or two of my momentary switches appear to be a little "sticky" and an initial click produces just a quick flash of light, requiring a second click to switch the light on.   How would I apply a debounce filter (if that is the correct solution) to the code to try to remove this issue?  Thank you.

mariodivece, 2024-02-14:
Wow! Thanks so much for all of this work @pimw1 and @svh1985 
I just ordered a couple of these and have not tested but I have added the following:
 - Overcurrent protection
 - Overpower protection
 - Power sensor
 - Current Sensor
 - Voltage sensor
 - Simplifying scripts

Again, I have not tested yet, but I will be sharing my findings and improvements on this thread.
My code is hosted here: [Shelly Dimmer2 Template](https://github.com/mariodivece/esphometemplates/blob/main/shelly-dimmer2.yaml)

bastiitsab, 2025-01-29:
@pimw1 & @mariodivece Very cool that you provide your examples here!
As I can see, you are using 2 button for the shelly dimmer in your example. 
Can you point me in the right direction what I need to change to use only 1 button to dim up/down?