esphome/issues #3488: Add a warning if trying to compile on a system with not enough RAM/swap

the-programmer, 2022-08-08:
### The problem

If you try to compile ESPHome on a default raspberry pi 3 the compile action runs out of memory and causes home assistant to crash along.
See the following posts
https://community.home-assistant.io/t/esp-home-crashes-ha/335415
https://github.com/esphome/issues/issues/2956

Since ESPHome needs at least 1GB or RAM/SWAP to compile is it an idea to add a warning that the compilation might fail and/or crash home assistant.

### Which version of ESPHome has the issue?

2022.6.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.8.2

### What platform are you using?

ESP8266

### Board

Wemos D1 Mini

### Additional information

I did have the original error on 1 version less. But since it existed a few years ago that should not matter.

Since I increased/added the SWAP space it would compile.
(see instructions from here https://community.home-assistant.io/t/how-to-increase-the-swap-file-size-on-home-assistant-os/272226)

ssieb, 2022-08-22:
esphome doesn't control the compiling directly, that's done by platformio.  There's no good way to detect this.  If you weren't running HA, then there probably would be enough memory.

the-programmer, 2022-08-22:
I know that platformio handles the compile. However since esphome tells platformio to start the compile I was thinking of adding something like this.
```
if(ammountOfRamAndSwap<1Gb){
    if(!showWarning("Less than 1GB of RAM available, Home assistant might crash, do you want to continue?")){
        return 0;
    }
}
PioCompile();
```

So just a warning before the compiler starts doing it's thing. That is all I am suggesting here.

ssieb, 2022-10-27:
A compile option has just been added to dev, so you can limit the number of processes, but probably a warning would be good as well, so you know to look for that option when it crashes.

si458, 2022-11-13:
@ssieb i have this issue with a 1gb pi 4, how do you limit the number of processes and will it help with the build?

ssieb, 2022-11-13:
https://next.esphome.io/components/esphome.html `compile_process_limit`
Set it to 1.  1GB is pretty low, but it might work.
(You need to use dev or beta esphome.)

the-programmer, 2022-11-13:
Maybe try to add some swap space. (See last link in bug report)
Maybe 1GB RAM is not enough.

si458, 2022-11-13:
thanks @ssieb and @the-programmer will give them a try!

si458, 2022-11-16:
@ssieb ok it no longer crashes home assistant by adding the line on beta
BUT it wont successfully build on my Pi4 1GB model sadly
```
Compiling /data/pico-w-home/.pioenvs/pico-w-home/src/esphome/core/log.cpp.o
Compiling /data/pico-w-home/.pioenvs/pico-w-home/src/esphome/core/scheduler.cpp.o
Compiling /data/pico-w-home/.pioenvs/pico-w-home/src/esphome/core/util.cpp.o
Compiling /data/pico-w-home/.pioenvs/pico-w-home/src/main.cpp.o
arm-none-eabi-g++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
*** [/data/pico-w-home/.pioenvs/pico-w-home/src/main.cpp.o] Error 1
========================= [FAILED] Took 181.51 seconds =========================
```

CarlosGS, 2022-12-19:
Hi, if you have 1GB RAM (either RPi3 or RPi4) I'd recommend installing the 32 bit version of Home Assistant.
It uses way less RAM and there is no need to use `compile_process_limit`.
My setup was 64bit Rpi 3 and it was unusable. With 32 bit everything works perfectly.

johnboiles, 2024-06-27:
FWIW even on 2GB of ram when compiling esp-idf+esp-adf things (voice assistants) I get OOM crashes too. It takes down all of Home Assistant and corrupts the ESPHome install such that i have to uninstall and reinstall the addon or I get submodule errors.