esphome/issues #3489: ruuvi_ble: incorrect parsing of invalid values (NA)

ssalonen, 2022-08-09:
### The problem

Ruuvi tag sensor handling seems to handle "invalid" (N/A values) sensor values incorrectly with data format 0x05 (aka RawV2):

1. Invalid values with Signed two byte fields are  signaled with [0x80, 0x00], not [0x7F, 0xFF] (implementation with temperature) nor [0xFF, 0xFF] (implementation with acceleration x/y/z). Apart from temperature and acceleration fields, there are no other fields affected. See https://docs.ruuvi.com/communication/bluetooth-advertisements/data-format-5-rawv2
2. Invalid values with unsigned fields are signaled with "largest presentable number for unsigned values", e.g. 0xFF or [0xFF, 0xFF]. The invalid value handling is missing completely with "movement counter" and "measurement sequence number". According to my tests other fields are handled correctly.
3. There is a calculated field, "acceleration" (magnitude of acceleration vector). The calculated field is set as NAN if any of the components (x, y or z) "equal" to NAN, i.e. == NAN. According to my ad-hoc testing [1] of the code on x86_64 this is not working as expected, as the comparison will be always false.
4. RuuviParseResult is storing fields as optional<float>. However, invalid values are stored as NAN, not as empty optional. Is this intended behavior? It will influence at least [publishing of values](https://github.com/esphome/esphome/blob/df853bf61ed7be05a7b7ff0722846eb3760c6a95/esphome/components/ruuvitag/ruuvitag.h#L25-L46) and [logging to Serial](https://github.com/esphome/esphome/blob/df853bf61ed7be05a7b7ff0722846eb3760c6a95/esphome/components/ruuvi_ble/ruuvi_ble.cpp#L97).

I have currently done these tests "locally" on my PC with x86_64 architecture.

[1]
```c++
// Test code for NAN comparison
            float floatNAN = NAN;
            esphome::optional<float> optNAN = NAN;
            std::cout << "false <=>   result.acceleration_x == NAN  result.acceleration_y == NAN  result.acceleration_z == NAN" << std::endl;

            std::cout << std::endl
                      << std::endl
                      << "NAN tests" << std::endl;
            std::cout << "float floatNAN = NAN;" << std::endl
                      << "esphome::optional<float> optNAN = NAN;" << std::endl;
            std::cout << std::endl;
            std::cout << ((optNAN == NAN) ? "true <=> optNAN == NAN" : "false <=> optNAN == NAN") << std::endl;
            std::cout << ((NAN == NAN) ? "true <=> NAN == NAN" : "false <=> NAN == NAN") << std::endl;
            std::cout << ((floatNAN == floatNAN) ? "true <=> floatNAN == floatNAN" : "false <=> floatNAN == floatNAN") << std::endl;
            std::cout << ((*optNAN == NAN) ? "true <=> *optNAN == NAN" : "false <=> *optNAN == NAN") << std::endl;
            std::cout << ((isnan(*optNAN)) ? "true <=> isnan(*optNAN)" : "false <=> isnan(*optNAN)") << std::endl;
            std::cout << ((isnan(NAN)) ? "true <=> isnan(NAN)" : "false <=> isnan(NAN)") << std::endl;
            std::cout << ((isnan(floatNAN)) ? "true <=> isnan(floatNAN)" : "false <=> isnan(floatNAN)") << std::endl;

// output (x86_64 linux, compilation with g++)

false <=> optNAN == NAN
false <=> NAN == NAN
false <=> floatNAN == floatNAN
false <=> *optNAN == NAN
true <=> isnan(*optNAN)
true <=> isnan(NAN)
true <=> isnan(floatNAN)
```
[2]

```c++

// test code for parsing against official test vectors in 
// https://docs.ruuvi.com/communication/bluetooth-advertisements/data-format-5-rawv2

int main()
{
    std::cout << "Testing Ruuvi parsing";

    {
        auto result = esphome::RuuviParseResult();
        auto data = esphome::esp32_ble_tracker::adv_data_t();
        data.push_back(0x05);
        data.push_back(0x12);
        data.push_back(0xFC);
        data.push_back(0x53);
        data.push_back(0x94);
        data.push_back(0xC3);
        data.push_back(0x7C);
        data.push_back(0x00);
        data.push_back(0x04);
        data.push_back(0xFF);
        data.push_back(0xFC);
        data.push_back(0x04);
        data.push_back(0x0C);
        data.push_back(0xAC);
[... 167 lines cut ...]
        print_field("measurement_sequence_number", result.measurement_sequence_number);

   }
}
```

### Which version of ESPHome has the issue?

dev branch 4d56a975e6fc2521dc77af656f056ef8bd6f119c

### What type of installation are you using?

pip

### What platform are you using?

ESP32

### Component causing the issue

ruuvi_ble

### Additional information

Initial fix here https://github.com/esphome/esphome/compare/dev...ssalonen:esphome:dev This is _not_ changing the optional behaviour and we still would store NANs in optional.

ssalonen, 2022-08-09:
Note: I have tested this mostly locally on linux PC using simulated data. The test vectors in the spec were very useful here

I believe the test vectors were probably not even available when the original support in esphome was introduced, in PR 
https://github.com/esphome/esphome/pull/810

ssalonen, 2022-08-09:
Indicative observation for NAN comparison (point 3): there seems no matches to `== NAN`, `!= NAN`, `NAN ==`, `NAN !=` apart from `ruuvi_ble`. There are several matches for calls to `std::isnan` though.

Alex9779, 2022-08-10:
Yeah well, this is so long ago I wrote this and I think your assumption is correct, some things didn't exists at that time and the Ruuvi docs didn't look like they do now. Can't remember if the details about invalid values were there or not, but I remember that I used some other implementation as a starting point.
So yeah, this should be corrected, feel free to do so...
I have no idea about the details your wrote about the implementation regarding optionals and so, sorry, I am just a hacker, modifying stuff so it works for me but I usually don't have a deep knowledge of what I am actually doing :P

ssalonen, 2022-08-10:
Thanks, appreciate the quick response!

I will probably then push a PR on this one.

Re. optionals, if it were up to me, I would keep the current behavior. It makes sense in a way, we are receiving a value, it just happens to be NAN/NA and as such it should be published in mqtt. However, if there are some other conventions in play, I can change the behavior...

closed 2022-12-24 by github-actions[bot]