esphome/issues #3490: Two vl53lox sensors with single reset line

osresearch, 2022-08-09:
### The problem

[image]

Since the ESP32 is often very pin-limited, it would be nice to be able to use two vl530lox sensors without having to use two separate reset pins.  This can be done by hooking one sensor's reset pin and holding that device in reset while the other one is configured to use a different i2c address.  Then the reset pin can be released, and the second sensor configured onto its address.  This works with the code as is, although a fake reset pin needs to be assigned to the other device.

A problem comes up if there has been a warm reset of the esp, such as a OTA update.  The sensors will still be on their assigned addresses, but the code assumes that it has reset it to the default 0x29 address with the (non existant) reset pin.  The reads from the sensor will be invalid, and the calibration ends up being wrong.

It looks like an easy fix to move the `final_address` configuration to before the rest of the device is configured. This handles both the power-on case (where the sensor has the default ID) as well as the warm reset case, in which the command to reset the ID will be lost, but the sensor is already on the correct address.  My changes against https://github.com/esphome/esphome/blob/dev/esphome/components/vl53l0x/vl53l0x_sensor.cpp#L53

```
  // Save the i2c address we want
  const uint8_t final_address = address_;

  // use the default address only to send the command to switch to the final address.
  // if the device didn't reset (if the pin is not actually connected and this
  // is a warm reset), then the command will be lost, but the sensor is already
  // on the new address.
  this->set_i2c_address(0x29);

  // Set the sensor to the desired final address
  // The following is different for VL53L0X vs VL53L1X
  // I2C_SXXXX_DEVICE_ADDRESS = 0x8A for VL53L0X
  // I2C_SXXXX__DEVICE_ADDRESS = 0x0001 for VL53L1X
  reg(0x8A) = final_address & 0x7F;

  // continue configuration with the new address
  this->set_i2c_address(final_address);
```

### Which version of ESPHome has the issue?

2022.08.08-dev

### What type of installation are you using?

pip

### What platform are you using?

ESP32

### Board

generic esp32

### Component causing the issue

vl53lox

### Example YAML snippet

```yaml
i2c:
  - id: i2c_a
    scl: 15
    sda: 14
    scan: false

sensor:
  # two laser range finders
  - platform: vl53l0x
    name: "distance-left"
    id: distance1
    i2c_id: i2c_a
    address: 0x11
    update_interval: 0.25s
    enable_pin: 12 # not actually used

  - platform: vl53l0x
    name: "distance-right"
    id: distance2
    i2c_id: i2c_a
    address: 0x22
    update_interval: 0.25s
    enable_pin: 13 # actually used
```
```

RoboMagus, 2022-08-14:
I've got a device with 2 vl53x0x sensors and only a single reset line as well. I was trying to implement some changes to the EspHome component to deal with it, but as it's set up I found it easier to add a small snippet of code to the `on_boot` section of the config to handle correctly assigning addresses to all my sensors:

```yaml
on_boot:
  priority: 700
  # Sensors initialize with priority 600, hardware 800, BUS (I2C) 1000...
  # Upon startup, check for the VL53l0x which address it has by reading these:  
  #IDENTIFICATION_MODEL_ID	0xc0	
  #IDENTIFICATION_REVISION_ID	0xc2	
  then:
    - lambda: |-                
        delay(100);
        // Search for devices with alternative address config and restore:
        uint8_t addresses[] = {0x29, 0x30, 0x31};
        for (uint8_t addr : addresses)
        {
          I2CDevice dev;
          dev.set_i2c_address(addr);
          dev.set_i2c_bus(i2c_arduinoi2cbus);

          uint8_t buf[4];
          bool _success = dev.read_bytes(0xC0, buf, 3);
          ESP_LOGI("main", "I2C: 0x%0x, read 0XC0: %x %x %x, success: %d", addr, buf[0], buf[1], buf[2], _success);
          bool found_vl53l0x = (buf[0] == 0xee) &&
                                             (buf[1] == 0xaa) &&
                                             (buf[2] == 0x10);
                              
          if (_success && found_vl53l0x) {
            uint8_t default_addr = 0x29 & 0x7F;
            bool rewrite_success = dev.write_byte(0x8A, default_addr);
            dev.set_i2c_address(0x29);
          }
          delay(100);
        }
```

And then for my sensors, it seems important to keep the one with the actual reset pin configured last, sucht that it is kept reset during initialization of the first.
```yaml
sensor:
  - platform: vl53l0x
    address: 0x31
    # Dummy enable-pin:
    enable_pin: LED
          
  - platform: vl53l0x
    address: 0x30
    # Actual enable pin:
    enable_pin: D7
```

osresearch, 2024-02-22:
This is still a problem for the most recent version of esphome.  While it is possible to work around it with the `on_boot` code provided by @RoboMagus, is there a reason to not do the address re-assignment first in `vl53l0x_sensor.h` so that it works without requiring any custom hacks?