esphome/issues #3491: ESP32 C3 undefined reference for template binary sensor

markusg1234, 2022-08-10:
### The problem

Template binary sensor and analog threshold sensor not compiling for ESP32 C3.

### Which version of ESPHome has the issue?

V2022.7.0-dev

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.8.2

### What platform are you using?

ESP32-IDF

### Board

ESP32-c3-devkitm-1

### Component causing the issue

binary sensor

### Example YAML snippet

```yaml
esphome:
  name: test-3
  platformio_options:
          board_build.flash_mode: dio

esp32:
  board: esp32-c3-devkitm-1  
  variant: ESP32C3
  framework:
   type: esp-idf  

binary_sensor:
   - platform: template
     name: "Open end stop"
     id: open_end_stop
```

### Anything in the logs that might be useful for us?

```txt
HARDWARE: ESP32C3 160MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.40302.0 (4.3.2) 
 - tool-cmake @ 3.16.4 
 - tool-ninja @ 1.7.1 
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch2 
 - toolchain-xtensa-esp32s2 @ 8.4.0+2021r2-patch2
Reading CMake configuration...
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
No dependencies
Linking /data/test-3/.pioenvs/test-3/firmware.elf
/data/cache/platformio/packages/toolchain-riscv32-esp@8.4.0+2021r2-patch2/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: /data/test-3/.pioenvs/test-3/src/main.o: in function `setup()':
/config/esphome/.esphome/build/test-3/src/main.cpp:48: undefined reference to `esphome::binary_sensor::BinarySensor::BinarySensor()'
/data/cache/platformio/packages/toolchain-riscv32-esp@8.4.0+2021r2-patch2/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: /config/esphome/.esphome/build/test-3/src/main.cpp:48: undefined reference to `vtable for esphome::template_::TemplateBinarySensor'
/data/cache/platformio/packages/toolchain-riscv32-esp@8.4.0+2021r2-patch2/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: /data/test-3/.pioenvs/test-3/src/main.o: in function `esphome::Component::Component()':
/config/esphome/.esphome/build/test-3/src/esphome/core/component.h:66: undefined reference to `vtable for esphome::template_::TemplateBinarySensor'
/data/cache/platformio/packages/toolchain-riscv32-esp@8.4.0+2021r2-patch2/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: /config/esphome/.esphome/build/test-3/src/esphome/core/component.h:66: undefined reference to `vtable for esphome::template_::TemplateBinarySensor'
/data/cache/platformio/packages/toolchain-riscv32-esp@8.4.0+2021r2-patch2/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: /config/esphome/.esphome/build/test-3/src/esphome/core/component.h:66: undefined reference to `vtable for esphome::template_::TemplateBinarySensor'
collect2: error: ld returned 1 exit status
*** [/data/test-3/.pioenvs/test-3/firmware.elf] Error 1
```

markusg1234, 2022-08-10:
Tested with 2022.8.0b1 and worked correctly.

lassi-niemisto, 2022-11-23:
I experience similar need of rebuild for the binary_sensor linkage with
```
esphome:
  name: heatroom-pumpdriver1

esp32:
  board: nodemcu-32s
  framework:
    type: esp-idf
    sdkconfig_options:
      CONFIG_FREERTOS_UNICORE: y
    advanced:
      ignore_efuse_mac_crc: true

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "snip"

ota:
  password: "snip"

wifi:
  ssid: snip
  password: snip

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Heatroom-Pumpdriver1"
    password: "snip"

switch:
  - platform: gpio 
    pin: GPIO27
    name: "switch_hr_circ_pump_tank"
    id: switch_hr_circ_pump_tank

  - platform: gpio 
    pin: GPIO14
    name: "switch_hr_circ_pump_house_extra"
    id: switch_hr_circ_pump_house_extra

binary_sensor:
  - platform: gpio 
    pin: GPIO35
    name: "test_io35"
    id: test_io35

  - platform: gpio 
    pin: GPIO33
    name: "test_io33"
    id: test_io33

  - platform: gpio 
    pin: GPIO21
    name: "test_io21"
    id: test_io21

  - platform: gpio 
    pin: GPIO19
    name: "test_io19"
    id: test_io19
```

ssieb, 2022-11-23:
IDF has issues with compile dependency checking.  Clean the build and try again.

closed 2022-11-23 by ssieb

clydebarrow, 2024-08-25:
Fixed by https://github.com/esphome/esphome/pull/7344