esphome/issues #3493: Obsolete PIO Core is used - add option to change `.platformio` dir location

TheLastGimbus, 2022-08-11:
### The problem

Hi there!

I am a happy user of Platformio in all my projects. So i had it installed, the newest version - `6.0.1`

I wanted to update my esphome devices - so i ran `pip install esphome`, and then `esphome run mydevice.yaml`

But it prints (on yellow):
```
esphome run mydevice.yaml
INFO Reading configuration mydevice.yaml...
INFO Generating C++ source...
INFO Compiling app...
**********************************************************************************************
Obsolete PIO Core v5.2.5 is used (previous was 6.0.1)
Please remove multiple PIO Cores from a system:
https://docs.platformio.org/page/faq.html#multiple-platformio-cores-in-a-system
**********************************************************************************************
```
and then big error on red:
```python
Error: Traceback (most recent call last):
  File "/home/matiii/dev/proj/esphome/venv/lib/python3.10/site-packages/platformio/__main__.py", line 121, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "/home/matiii/dev/proj/esphome/venv/lib/python3.10/site-packages/click/core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
  File "/home/matiii/dev/proj/esphome/venv/lib/python3.10/site-packages/click/core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "/home/matiii/dev/proj/esphome/venv/lib/python3.10/site-packages/platformio/commands/__init__.py", line 44, in invoke
    return super(PlatformioCLI, self).invoke(ctx)
  File "/home/matiii/dev/proj/esphome/venv/lib/python3.10/site-packages/click/core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/matiii/dev/proj/esphome/venv/lib/python3.10/site-packages/click/core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/matiii/dev/proj/esphome/venv/lib/python3.10/site-packages/click/core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "/home/matiii/dev/proj/esphome/venv/lib/python3.10/site-packages/click/decorators.py", line 26, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/home/matiii/dev/proj/esphome/venv/lib/python3.10/site-packages/platformio/commands/run/command.py", line 134, in cli
    process_env(
  File "/home/matiii/dev/proj/esphome/venv/lib/python3.10/site-packages/platformio/commands/run/command.py", line 176, in process_env
    result = {"env": name, "duration": time(), "succeeded": ep.process()}
  File "/home/matiii/dev/proj/esphome/venv/lib/python3.10/site-packages/platformio/commands/run/processor.py", line 67, in process
    result = init_platform(self.options["platform"]).run(
  File "/home/matiii/dev/proj/esphome/venv/lib/python3.10/site-packages/platformio/commands/platform.py", line 322, in init_platform
    return PlatformFactory.new(name)
  File "/home/matiii/dev/proj/esphome/venv/lib/python3.10/site-packages/platformio/platform/factory.py", line 52, in new
    pkg = PlatformPackageManager().get_package(pkg_or_spec)
  File "/home/matiii/dev/proj/esphome/venv/lib/python3.10/site-packages/platformio/package/manager/base.py", line 236, in get_package
    for pkg in self.get_installed():
  File "/home/matiii/dev/proj/esphome/venv/lib/python3.10/site-packages/platformio/package/manager/base.py", line 209, in get_installed
    pkg = PackageItem(pkg_dir)
  File "/home/matiii/dev/proj/esphome/venv/lib/python3.10/site-packages/platformio/package/meta.py", line 405, in __init__
    self.metadata = self.load_meta()
  File "/home/matiii/dev/proj/esphome/venv/lib/python3.10/site-packages/platformio/package/meta.py", line 440, in load_meta
    return PackageMetaData.load(manifest_path)
  File "/home/matiii/dev/proj/esphome/venv/lib/python3.10/site-packages/platformio/package/meta.py", line 393, in load
    data["spec"] = PackageSpec(**data["spec"])
TypeError: PackageSpec.__init__() got an unexpected keyword argument 'uri'

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================
```

(i suppose because of said `yellow` section - because it's a downgrade version of platformio)

This is a general problem that i always had - my global platformio version was in conflict with what esphome wanted to use

I tried to solve this with `virtualenv`, but it is still using same `$HOME/.platformio` path for all of it's stuff

I fixed this temporarli for myself by setting env var [PLATFORMIO_CORE_DIR](https://docs.platformio.org/en/latest/envvars.html#envvar-PLATFORMIO_CORE_DIR) to my local project dir. I tried adding [core_dir](https://docs.platformio.org/en/latest/projectconf/section_platformio.html#core-dir) to [platformio_options](https://esphome.io/components/esphome.html#platformio-options) in my yaml file, but:

```
Warning! Ignore unknown configuration option `core_dir` in section [env:myproject]
```
...it is not recognized, because it's the env section

Solution for me now, is to set `PLATFORMIO_CORE_DIR` *every time*... but it would be nice if there was an option to set this permanantly... either in each and every device `.yaml` file, or some new `esphome.yaml` config... 

### Which version of ESPHome has the issue?

2022.6.3

### What type of installation are you using?

pip

### What platform are you using?

ESP8266

### Component causing the issue

platformio / core esphome

r0oland, 2022-09-27:
- any help / ideas / solutions for this? I have the same issue...

TheLastGimbus, 2022-10-02:
> I fixed this temporarli for myself by setting env var [PLATFORMIO_CORE_DIR](https://docs.platformio.org/en/latest/envvars.html#envvar-PLATFORMIO_CORE_DIR)