esphome/issues #3496: Unable to use Switch component with esp-idf resulting with linker error

kamilosxd678, 2022-08-14:
### The problem

I cannot add `switch` component, because it results in an instantenous linker error. Basically provided snippet works well if you comment out `switch:` line. Otherwise I get linker error

### Which version of ESPHome has the issue?

2022.3.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.08.3

### What platform are you using?

ESP32-IDF

### Board

Shelly Plus 1

### Component causing the issue

switch

### Example YAML snippet

```yaml
substitutions:
  device_name: "Bedroom Shelly"
  # Higher value gives lower watt readout
  current_res: "0.001"
  # Lower value gives lower voltage readout
  voltage_div: "1925"

esphome:
  name: bedroom-shelly
  platformio_options:
    board_build.f_cpu: 160000000L

esp32:
  board: esp32doit-devkit-v1
  framework:
    type: esp-idf
    sdkconfig_options:
      CONFIG_FREERTOS_UNICORE: y
      CONFIG_ESP32_DEFAULT_CPU_FREQ_160: y
      CONFIG_ESP32_DEFAULT_CPU_FREQ_MHZ: "160"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Bedroom-Shelly Fallback Hotspot"

logger:

api:

ota:

output:
  - platform: gpio
    id: "relay_output"
    pin: GPIO26

light:
  - platform: binary
    id: "light1"
    name: "Bedroom Main Light"
    output: "relay_output"

binary_sensor:
  - platform: gpio
    name: "${device_name} Switch"
    pin: GPIO4
    on_state:
      then:
        - light.toggle: "light1"
    filters:
      - delayed_on_off: 50ms
  - platform: gpio
    name: "${device_name} Button"
    pin:
      number: GPIO25
      inverted: yes
      mode:
        input: true
        pullup: true
    on_state:
      then:
        - light.toggle: "light1"
    filters:
      - delayed_on_off: 5ms

switch:

sensor:
  - platform: ntc
    sensor: temp_resistance_reading
    name: "${device_name} Temperature"
    unit_of_measurement: "°C"
    accuracy_decimals: 1
    icon: "mdi:thermometer"
    calibration:
      b_constant: 3350
      reference_resistance: 10kOhm
      reference_temperature: 298.15K
    on_value_range:
      - above: "80.0"
        then:
          - light.turn_off: "light1"
  - platform: resistance
    id: temp_resistance_reading
    sensor: temp_analog_reading
    configuration: DOWNSTREAM
    resistor: 6kOhm
  - platform: adc
    id: temp_analog_reading
    pin: GPIO32
    attenuation: 11db

  - platform: adc
    name: "${device_name} Relay Supply Voltage"
    pin: GPIO33
    attenuation: 11db
    filters:
      - multiply: 8

status_led:
  pin:
    number: GPIO0
    inverted: true
```

### Anything in the logs that might be useful for us?

```txt
INFO Reading configuration /config/esphome/bedroom-shelly.yaml...
WARNING GPIO4 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING GPIO0 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Generating C++ source...
INFO Compiling app...
Processing bedroom-shelly (board: esp32doit-devkit-v1; framework: espidf; platform: platformio/espressif32 @ 3.5.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 160MHz, 320KB RAM, 4MB Flash
 - framework-espidf 3.40302.0 (4.3.2) 
 - tool-cmake 3.16.4 
 - tool-ninja 1.7.1 
 - toolchain-riscv32-esp 8.4.0+2021r2-patch2 
 - toolchain-xtensa-esp32 8.4.0+2021r2-patch2 
 - toolchain-xtensa-esp32s2 8.4.0+2021r2-patch2
Reading CMake configuration...
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
No dependencies
Compiling /data/bedroom-shelly/.pioenvs/bedroom-shelly/src/main.o
Linking /data/bedroom-shelly/.pioenvs/bedroom-shelly/firmware.elf
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /data/bedroom-shelly/.pioenvs/bedroom-shelly/src/esphome/components/api/api_connection.o:(.literal._ZN7esphome3api13APIConnection14switch_commandERKNS0_20SwitchCommandRequestE+0x0): undefined reference to `esphome::switch_::Switch::turn_on()'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /data/bedroom-shelly/.pioenvs/bedroom-shelly/src/esphome/components/api/api_connection.o:(.literal._ZN7esphome3api13APIConnection14switch_commandERKNS0_20SwitchCommandRequestE+0x4): undefined reference to `esphome::switch_::Switch::turn_off()'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /data/bedroom-shelly/.pioenvs/bedroom-shelly/src/esphome/components/api/api_connection.o:(.literal._ZN7esphome3api13APIConnection16send_switch_infoEPNS_7switch_6SwitchE+0x4): undefined reference to `esphome::switch_::Switch::get_device_class[abi:cxx11]()'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /data/bedroom-shelly/.pioenvs/bedroom-shelly/src/esphome/components/api/api_connection.o: in function `esphome::api::APIConnection::switch_command(esphome::api::SwitchCommandRequest const&)':
/config/esphome/.esphome/build/bedroom-shelly/src/esphome/components/api/api_connection.cpp:475: undefined reference to `esphome::switch_::Switch::turn_on()'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /config/esphome/.esphome/build/bedroom-shelly/src/esphome/components/api/api_connection.cpp:477: undefined reference to `esphome::switch_::Switch::turn_off()'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /data/bedroom-shelly/.pioenvs/bedroom-shelly/src/esphome/components/api/api_connection.o: in function `esphome::api::APIConnection::send_switch_info(esphome::switch_::Switch*)':
/config/esphome/.esphome/build/bedroom-shelly/src/esphome/components/api/api_connection.cpp:464: undefined reference to `esphome::switch_::Switch::get_device_class[abi:cxx11]()'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /data/bedroom-shelly/.pioenvs/bedroom-shelly/src/esphome/core/controller.o:(.literal._ZN7esphome10Controller16setup_controllerEb+0x20): undefined reference to `esphome::switch_::Switch::add_on_state_callback(std::function<void (bool)>&&)'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /data/bedroom-shelly/.pioenvs/bedroom-shelly/src/esphome/core/controller.o: in function `esphome::Controller::setup_controller(bool)':
/config/esphome/.esphome/build/bedroom-shelly/src/esphome/core/controller.cpp:35: undefined reference to `esphome::switch_::Switch::add_on_state_callback(std::function<void (bool)>&&)'
collect2: error: ld returned 1 exit status
*** [/data/bedroom-shelly/.pioenvs/bedroom-shelly/firmware.elf] Error 1
========================= [FAILED] Took 45.22 seconds =========================
```

### Additional information

YAML taken from https://www.esphome-devices.com/devices/Shelly-Plus-1

jesserockz, 2022-08-14:
Have you tried `cleaning the build files`?
This can happen if you change from arduino to esp-idf after a compile.

jesserockz, 2022-08-14:
I hate to say this, but I took that yaml from esphome-devices and it compiled fine for me first time.
```
Linking .pioenvs/shelly-plus-1/firmware.elf
RAM:   [=         ]   9.6% (used 31592 bytes from 327680 bytes)
Flash: [====      ]  43.1% (used 791005 bytes from 1835008 bytes)
Building .pioenvs/shelly-plus-1/firmware.bin
esp32_create_combined_bin([".pioenvs/shelly-plus-1/firmware.bin"], [".pioenvs/shelly-plus-1/firmware.elf"])
Wrote 0xd1360 bytes to file /home/jesse/workspace/esphome/esphome/config/dev/.esphome/build/shelly-plus-1/.pioenvs/shelly-plus-1/firmware-factory.bin, ready to flash to offset 0x0
==================================================================================================================================================================================================== [SUCCESS] Took 142.25 seconds ====================================================================================================================================================================================================
INFO Successfully compiled program.
```

markusg1234, 2022-08-15:
Hi,
I get this error if the following is done.

1) New compile with a single switch or binary sensor.
2) Remove the switch or binary sensor.
3) Recompile.

Also get a different error, undefined reference to `vtable for esphome::gpio::GPIOSwitch' if I add a single switch after the first compile.

kamilosxd678, 2022-08-15:
Well, the unmodified one works OK, but when I try to add `switch` component for additional logic, it all breaks

jenscski, 2022-08-15:
I have seen something like this before my self. What helped was to clean build files. It's a option in dashboard, and with the cli.

kamilosxd678, 2022-08-15:
@jenscski Thank, cleaning files fixed the issue

closed 2022-08-15 by kamilosxd678

markusg1234, 2022-08-15:
> I have seen something like this before my self. What helped was to clean build files. It's a option in dashboard, and with the cli.

Thanks for the suggestion.
It allowed me to compile again until I made another change and then I need to clean the build files again.

jenscski, 2022-08-15:
> > I have seen something like this before my self. What helped was to clean build files. It's a option in dashboard, and with the cli.
> 
> Thanks for the suggestion. It allowed me to compile again until I made another change and then I need to clean the build files again.

Same here, but I seldom use idf, so I haven't taken the time to create an issue for it.