esphome/issues #3497: Utilising motor with rotary encoder (off the shelf geared motor) to drive chicken run door. Variable = NaN?

Swiftnesses, 2022-08-15:
### The problem

Hello everyone,

I've been fighting with this over the weekend, and thought I'd reach out for some help.

I'm trying to use a geared motor (motor with encoder + gearbox) to drive a chicken coop door. I'm looking to develop the following features:

**Calibrate closed position** (move door to closed position in steps of 1000)
**Save closed position** (save encoder position to a variable for the intended close position)
**Calibrate open position** (move door to open position in steps of 1000)
**Save open position** (save encoder position to a variable for the intended open position)
**Open door** (move to open position saved in variable)
**Close door** (move to closed position saved to variable)

I've started to develop the code (see snippet), but am struggling. Initially I tried scripting the logic, but I wasn't getting updates from the encoder using a while loop, which caused a reboot. I then resorted to writing the logic within the encoder section (using on_value). 

The issue currently is that when I press one of the calibrate buttons, the motor turns and the encoder updates are received, but the global I use to store the initial encoder state (to compare against the live reading), returns NaN. Please see log below.

### Which version of ESPHome has the issue?

2022.6.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

Latest

### What platform are you using?

ESP8266

### Board

NodeMCU

### Component causing the issue

Not sure

### Example YAML snippet

```yaml
esphome:
  name: esphome-chicken-run-door

esp8266:
  board: esp01_1m

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: "x"
ota:
  password: "x"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esphome-Chicken-Run-Door"
    password: "12345678"

captive_portal:

globals:
  - id: opening
    type: bool
    restore_value: no
    initial_value: "false"
  - id: closing
    type: bool
    restore_value: no
    initial_value: "false"
  - id: calibrate_open
    type: bool
    restore_value: no
    initial_value: "false"
  - id: calibrate_closed
    type: bool
    restore_value: no
    initial_value: "false"
  - id: encoder_temp_position
    type: float
    restore_value: no
  - id: encoder_open_position
    type: float
    restore_value: yes
  - id: encoder_closed_position
    type: float
    restore_value: yes

button:
  - platform: template
    name: Calibrate Open Position
    id: calibrate_open_posiiton
    icon: "mdi:menu-up-outline"
    on_press:
      - logger.log: "Moving door up a little"
      - globals.set:
          id: encoder_temp_position
          value: id(motor_encoder).state
      - globals.set:
          id: calibrate_open
          value: "true"
      - output.turn_off: direction
      - output.set_level:
          id: pwm_output
          level: 1%
      
  - platform: template
    name: Calibrate Closed Position
    id: calibrate_closed_position
    icon: "mdi:menu-down-outline"
    on_press:
      - logger.log: "Moving door down a little"
      - globals.set:
          id: encoder_temp_position
          value: id(motor_encoder).state
      - globals.set:
          id: calibrate_closed
          value: "true"
      - output.turn_on: direction
      - output.set_level:
          id: pwm_output
          level: 1%
      
  - platform: template
    name: Save Open Position
    id: save_open_position
    icon: "mdi:content-save"
    on_press:
      - logger.log: "Saved open position"
      - globals.set:
          id: encoder_open_position
          value: id(motor_encoder).state
        
  - platform: template
    name: Save Closed Position
    id: save_closed_position
    icon: "mdi:content-save"
    on_press:
      - logger.log: "Saved close position"
      - sensor.rotary_encoder.set_value:
          id: motor_encoder
          value: 0
      - globals.set:
          id: encoder_closed_position
          value: "0"

output:
- platform: gpio
  pin: GPIO02
  id: direction

- platform: esp8266_pwm
  pin: GPIO04
  frequency: 100Hz
  id: pwm_output
  min_power: 0.25
  max_power: 1.0
  zero_means_zero: true
    
fan:
  - platform: speed
    output: pwm_output
    name: "Run Door"
    direction_output: direction

sensor:
  - platform: rotary_encoder
    id: motor_encoder
    name: "Motor Encoder"
    pin_a: GPIO13
    pin_b: GPIO15
    publish_initial_value: true
    on_value:
      then:
        - if:
            condition:
              - lambda: |-
                  ESP_LOGI("custom", "Start Encoder Position: %f", id(encoder_temp_position));
                  ESP_LOGI("custom", "Current Encoder Position: %f", id(motor_encoder).state);
                  ESP_LOGI("custom", "Difference Encoder Position: %f", id(motor_encoder).state - id(encoder_temp_position));
                
                  if (id(calibrate_open) = true && id(motor_encoder).state - id(encoder_temp_position) >= 1000) {
                    ESP_LOGI("custom", "In the if loop");
                    return true;
                  } else {
                    ESP_LOGI("custom", "Outside the if loop");
                    return false;
                  }
            then:
              - lambda: |-
                  ESP_LOGI("custom", "In the then statement");
                  id(calibrate_open) = false;
                  id(pwm_output).set_level(0);
                  return;
```

### Anything in the logs that might be useful for us?

```txt
[08:59:00][D][button:013]: 'Calibrate Open Position' Pressed.
[08:59:00][D][main:240]: Moving door up a little
[08:59:00][D][sensor:124]: 'Motor Encoder': Sending state 1924.00000 steps with 0 decimals of accuracy
[08:59:00][I][custom:143]: Start Encoder Position: nan
[08:59:00][I][custom:144]: Current Encoder Position: 1924.000000
[08:59:00][I][custom:145]: Difference Encoder Position: -nan
[08:59:00][I][custom:151]: Outside the if loop
[08:59:00][D][sensor:124]: 'Motor Encoder': Sending state 1929.00000 steps with 0 decimals of accuracy
[08:59:00][I][custom:143]: Start Encoder Position: nan
[08:59:00][I][custom:144]: Current Encoder Position: 1929.000000
[08:59:00][I][custom:145]: Difference Encoder Position: -nan
[08:59:00][I][custom:151]: Outside the if loop
[08:59:00][D][sensor:124]: 'Motor Encoder': Sending state 1936.00000 steps with 0 decimals of accuracy
[08:59:00][I][custom:143]: Start Encoder Position: nan
[08:59:00][I][custom:144]: Current Encoder Position: 1936.000000
[08:59:00][I][custom:145]: Difference Encoder Position: -nan
[08:59:00][I][custom:151]: Outside the if loop
[08:59:00][D][sensor:124]: 'Motor Encoder': Sending state 1946.00000 steps with 0 decimals of accuracy
[08:59:00][I][custom:143]: Start Encoder Position: nan
[08:59:00][I][custom:144]: Current Encoder Position: 1946.000000
[08:59:00][I][custom:145]: Difference Encoder Position: -nan
[08:59:00][I][custom:151]: Outside the if loop
[08:59:00][D][sensor:124]: 'Motor Encoder': Sending state 1958.00000 steps with 0 decimals of accuracy
[08:59:00][I][custom:143]: Start Encoder Position: nan
[08:59:00][I][custom:144]: Current Encoder Position: 1958.000000
[08:59:00][I][custom:145]: Difference Encoder Position: -nan
[08:59:00][I][custom:151]: Outside the if loop
```

Swiftnesses, 2022-08-15:
Seems I found a workaround, using Lambda to set the global in the button action seems to work:

```
      # - globals.set:
          # id: encoder_temp_position
          # value: id(motor_encoder).state
      - lambda: |-
          id(encoder_temp_position) = id(motor_encoder).state;
          id(calibrate_open) = true;
```

ssieb, 2022-08-22:
github issues are not for support.  The Discord server is the best option, but there are also the HA forums and various other locations.

closed 2022-08-22 by ssieb