esphome/issues #3500: Sprinkler Controller Example

rb2k, 2022-08-17:
### The problem

https://esphome.io/components/sprinkler.html#additional-tricks lists a number example that looks like this:

```
# Example configuration to set multiplier via number
number:
  - platform: template
    id: sprinkler_ctrlr_multiplier
    name: "Sprinkler Controller Multiplier"
    min_value: 0.1
    max_value: 10.0
    step: 0.1
    initial_value: 1.0
    lambda: "return id(lawn_sprinkler_ctrlr).multiplier();"
    set_action:
      - sprinkler.set_multiplier:
          id: lawn_sprinkler_ctrlr
          multiplier: !lambda 'return x;'
```

Trying to use this in esphome will result in:

```
  initial_value cannot be used with lambda.
```

Looking at https://esphome.io/components/number/template.html , that seems to be correct.

### Which version of ESPHome has the issue?

2022.8.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.8.5

### What platform are you using?

ESP8266

### Board

sonoff 4ch pro

rb2k, 2022-08-17:
/cc @kbx81 as the author of the docs :)

kbx81, 2022-08-17:
Oh no! That's what I get for trying to improve an example at the [last minute](https://github.com/esphome/esphome-docs/pull/1471/commits/78dd863ecb0a2d1348799c3bcd070e18c3e191af). 😆
Thanks for calling this out -- fix incoming...

rb2k, 2022-08-17:
Ha, no problem :)

Do you have any ideas what the most elegant way is to set this number?
Otherwise it'll default to the minimum after startup.

So far used on_boot with a call to ```sprinkler.set_multiplier()```, just wondering if there's a more straight forward way to do things.

kbx81, 2022-08-17:
Internal to the sprinkler controller, the multiplier value defaults to a value of `1` on start-up. You don't need to set it with `on_boot` unless you are looking for some specific behavior.

If you use the state lambda as shown in the example, the `number` component will reflect the value returned by the sprinkler controller's `multiplier()` accessor method, so you should see it default to 1 there, as well. Are you wanting it to assume a default value other than `1`?

kbx81, 2022-08-17:
Now you've got me wondering if we need a `default_multipler_value` and possibly even `multiplier_min` and `multiplier_max` parameters for sanity checking. 😄 Perhaps something for v2...

closed 2022-08-17 by balloob

rb2k, 2022-08-17:
> Internal to the sprinkler controller, the multiplier value defaults to a value of `1` on start-up (this is not the smallest possible value). You don't need to set it with `on_boot` unless you are looking for some specific behavior.
> 
> If you use the state lambda as shown in the example, the `number` component will reflect the value returned by the sprinkler controller's `multiplier()` accessor method, so you should see it default to 1 there, as well. Are you wanting it to assume a default value other than `1`?

Ah yes. I didn't feel smart enough to remember what the default run time is and what the multiplier is set to, so I just did a default run time of 60 seconds and then the 'multiplier' will just tell me how many minutes each zone will run.

Ends up looking like this
[image]

I guess I could have just left the multiplier alone at 1 and somehow tried to set the duration for all the valves separately, but that seemed more complicated :)

mitchelloharawild, 2022-09-22:
Regarding run durations, is there a neat way to restore set duration values on boot? Perhaps an option like `restore_mode` for the `sprinkler` component?
I'd like to set the run duration via a `number` component and then have the sprinkler component restore the set value on boot.

My relevant sprinkler input is:
```yaml
sprinkler:
  - id: irrigation_ctrlr
    main_switch: "Irrigation"
    auto_advance_switch: "Irrigation Auto Advance"
    valve_overlap: 5s
    valves:
      - valve_switch: "Zone 1"
        enable_switch: "Enable zone 1"
        run_duration: 60s
        valve_switch_id: zone1_relay
```

My number input is:

```yaml
number:
  - platform: template
    id: zone1_minutes
    min_value: 0
    step: 1
    max_value: 15
    name: "Zone 1 minutes"
    icon: "mdi:watering-can"
    unit_of_measurement: minutes
    update_interval: never
    entity_category: config
    mode: slider
    lambda: "return id(irrigation_ctrlr).valve_run_duration(0)/60;"
    set_action:
      - sprinkler.set_valve_run_duration:
          id: irrigation_ctrlr
          valve_number: 0
          run_duration: !lambda 'return x*60;'
      - component.update: zone1_minutes
```

kbx81, 2022-09-22:
At present, restoring is not possible and the run_duration value for any given valve reverts to what appears in the configuration at boot. Perhaps this will be addressed in a future release. 😄

As a side-note, are people really resetting/rebooting their ESPs so often that this is an issue? haha
Mine is plugged in, the only way these values would get lost is if we had a power failure. This is pretty rare so it's not an issue that I encounter myself. 😄 Just curious/wondering. 😇

kbx81, 2022-09-22:
That said, if you _really_ need this functionality at this moment, one option you could do is to use a set of `global` variables to store your desired run durations, then set each valve from its corresponding value (for example) each time a cycle or valve is started.

mitchelloharawild, 2022-09-22:
> At present, restoring is not possible and the run_duration value for any given valve reverts to what appears in the configuration at boot. Perhaps this will be addressed in a future release. smile

Thanks for the info, this is what I encountered. Perhaps an alternative to restoring state would be to incorporate the creation of number templates into the sprinkler configuration? Or accept a number template by id for things like run duration?

> As a side-note, are people really resetting/rebooting their ESPs so often that this is an issue? haha Mine is plugged in, the only way these values would get lost is if we had a power failure. This is pretty rare so it's not an issue that I encounter myself. smile Just curious/wondering. innocent

Power failure is more common for me than I'd like, but yes - generally my ESPs are kept on whenever possible :laughing:.

kbx81, 2022-09-22:
Yes, I think in a future update, the use of `number` templates would be a nice addition to handle that bit automagically. 😉