esphome/issues #3501: Lambda math don't work

EvertJob, 2022-08-17:
### The problem

Hi,

I'm trying to "map" the height to kilograms using this formula:
```
  - platform: template
    name: "test"
    lambda: |-
      return (35/46) * (20.7 - 5);
    update_interval: 1s
```

The 20.7 value is coming from another sensor but for now it's not relevant.

What ever I'm trying, the result is always 0.0. The result of `(35/46)` is `0.74468085106` and if we change it so I become above zero the correct result is coming into ESPHome.

When I use this:
```
  - platform: template
    name: "test"
    lambda: |-
      return 0.74468085106 * (20.7 - 5);
    update_interval: 1s
```

It is showing me the correct value: 11.69.

### Which version of ESPHome has the issue?

2022.8.0

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP8266

### Board

NoceMCU

### Component causing the issue

template

nielsnl68, 2022-08-17:
Try the following:
```yaml
- platform: template
    name: "test"
    lambda: |-
      return (35.0/46.0) * (20.7 - 5.0);
    update_interval: 1s
```

It could be that the compile converts the value to an integer automatically because of the integer values you used.

ssieb, 2022-08-17:
That's correct.  lambdas are embedded C++ code, so follow the C++ type rules.  Integer division gives an integer result.

closed 2022-08-17 by ssieb