esphome/issues #3510: Restore function in servo component doesn't work

blackscreener, 2022-08-22:
### The problem

Restore function doesn't work after esp restart. Template number is restore properly, but servo isn't. Servo is sg90.

### Which version of ESPHome has the issue?

2022.8.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.8.6

### What platform are you using?

ESP8266

### Board

NodeMCU v2

### Component causing the issue

servo, template number

### Example YAML snippet

```yaml
servo:
  - id: my_servo
    output: pwm_output
    auto_detach_time: 0s
    transition_length: 0s
    restore: true
 
output:
  - platform: esp8266_pwm
    id: pwm_output
    pin: D1
    frequency: 50 Hz
 
number:
  - platform: template
    name: Servo Control
    min_value: -12
    max_value: 100
    step: 1
    restore_value: true
    set_action:
      then:
        - servo.write:
            id: my_servo
            level: !lambda 'return x / 100.0;'
```

DivanX10, 2022-10-25:
I also ran into this problem, did not find how to solve the problem?  I resorted to a numeric pattern that preserves the position.

Create a number and name it, for example, servo memory, where the last position of the servo will be stored
```
number:
  - platform: template
    name: servo memory
    id: servo_memory
    min_value: -100
    max_value: 100
    mode: box #slider/box
    step: 1
    optimistic: true
    restore_value: true
```

We use number servo memory to restore the position after restarting the board
```
number:
  - platform: template
    name: servo position
    min_value: -100
    max_value: 100
    update_interval: 5s 
    mode: slider #slider/box
    lambda: 'return id(servo_memory).state;' 
    step: 1
    set_action:
      then:
        - servo.write:
           id: servo_control
           level: !lambda 'return x / 100.0;'
        - number.set:
           id: servo_memory
           value: !lambda 'return x;'
```

blackscreener, 2022-11-19:
@DivanX10 
Try that:

```
esphome:
  name: servo
  on_boot:
    priority: 600
    # ...
    then:
      - servo.write:
           id: my_servo
           level: !lambda 'return id(servo_memory).state / 100.0;'
```

ssieb, 2023-06-05:
Use `on_value` on the number instead of `set_action`.

closed 2023-06-05 by ssieb

DivanX10, 2023-06-05:
I forgot about this post, since a lot of time has passed, but I found a way to fix it and post a ready-made code that allows you to save the position of the servo and the readings in the sensor will not go astray even after restarting the esp

```
substitutions:
  board_name: Servo Control
  node_name: servo-control

esphome:
  name: ${node_name}
  comment: ESP32 Servo Control
#Restoring the current position of the servo
  on_boot:
    priority: -100
    then:
       - sensor.template.publish:
           id: servo_sensor
           state: !lambda "return id(saved_position);"

esp32:
  board: esp32dev
  framework:
    type: arduino

#Wi-Fi credentials for connecting the board to the home network
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  fast_connect: on

#If there is no connection with WiFi, then the access point will rise
  ap:
    ssid: ESP32 Servo Control
    password: !secret ap_esp_password

#The captive portal ESPHome component is a backup mechanism in case the connection to the configured Wi-Fi fails.
captive_portal:

#Web server
web_server:
  port: 80

#Logging
logger:
  level: ERROR #DEBUG

#Enable OTA
ota:
  password: "esphome"

#Enable Home Assistant API
api:
  encryption:
    key: !secret api_key
  services:
   - service: control_servo
     variables:
       level: float
     then:
       - servo.write:
           id: servo_control
           level: !lambda 'return level / 100.0;'
       - sensor.template.publish:
           id: servo_sensor
           state: !lambda 'return (float)level;'

#####################################################################################
############################## Output Platform #####################################
output:
  - platform: ledc
    id: pwm_output
    pin: GPIO23
    frequency: 50 Hz

#####################################################################################
########################### Global variables ###################################
globals:
  - id: saved_position
    type: int
    initial_value: '0'
    restore_value: true

#####################################################################################
################################ Configuration #######################################
#Configuration variables: Output (required, ID): ID of the output component to be used for this servo
servo:
  - id: servo_control
    output: pwm_output
    auto_detach_time: 5s
    transition_length: 5s
    restore: true

#####################################################################################
################################## Sensor ###########################################
sensor:
#Servomotor position sensor
  - platform: template
    name: Servo Sensor Position
    id: servo_sensor
    icon: mdi:valve
    on_value:
      then:
        - script.execute: record_servo_position

#####################################################################################
####################################### Script ######################################
script:
#We save the position of the servo to global variables
  - id: record_servo_position
    then:
      - globals.set:
          id: saved_position
          value: !lambda 'return id(servo_sensor).state;'

#####################################################################################
####################################### Number #######################################
#Slider for servo control
number:
  - platform: template
    name: Servo Position
    min_value: -100
    max_value: 100
    update_interval: 1s 
    mode: slider #slider/box
    lambda: 'return id(servo_sensor).state;' 
    step: 1
    set_action:
      then:
        - servo.write:
           id: servo_control
           level: !lambda 'return x / 100.0;'
        - sensor.template.publish:
           id: servo_sensor
           state: !lambda 'return x;'

```