esphome/issues #3515: media_player component: Guru meditation when setting volume on boot

veitw, 2022-08-23:
### The problem

Using the media_player component, I am trying to set a default volume on boot, as the 100% default is too loud.

Unfortunately, using esphome.on_boot to set the volume at boot time crashes the ESP in a boot loop.

### Which version of ESPHome has the issue?

2022.8.0, 2022.6.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.8.4

### What platform are you using?

ESP32

### Board

DOIT ESP32 DevKit V1

### Component causing the issue

media_player

### Example YAML snippet

```yaml
esphome:
...
  on_boot:
    priority: -100
    then:
      - media_player.volume_set:
          id: dac
          volume: 50%
...
media_player:
  - platform: i2s_audio
    id: dac
    name: "${device_name} Lautsprecher"
    dac_type: external
    i2s_dout_pin: GPIO14
    i2s_bclk_pin: GPIO27
    i2s_lrclk_pin: GPIO26
    mode: mono
...
```

### Anything in the logs that might be useful for us?

```txt
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
[I][logger:243]: Log initialized
[C][ota:461]: There have been 7 suspected unsuccessful boot attempts.
[D][esp32.preferences:113]: Saving 1 preferences to flash...
[D][esp32.preferences:142]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[I][app:029]: Running through setup()...
[D][binary_sensor:034]: 'esphome-speaker01 Status': Sending initial state OFF
[D][text_sensor:067]: 'esphome-speaker01 ESPHome Version': Sending state '2022.8.0 Aug 23 2022, 22:26:34'
[C][wifi:037]: Setting up WiFi...
[I][wifi:248]: WiFi Connecting to 'XXX'...
[D][esp32.preferences:113]: Saving 1 preferences to flash...
[D][esp32.preferences:142]: Saving 1 preferences to flash: 1 cached, 0 written, 0 failed
[I][wifi:505]: WiFi Connected!
[C][wifi:353]:   Local MAC: 78:E3:6D:XX:XX:XX
[C][wifi:354]:   SSID: [redacted]
[C][wifi:355]:   IP Address: 192.168.X.X
[C][wifi:357]:   BSSID: [redacted]
[C][wifi:358]:   Hostname: 'esphome-speaker01'
[C][wifi:360]:   Signal strength: -51 dB ▂▄▆█
[C][wifi:364]:   Channel: 1
[C][wifi:365]:   Subnet: 255.255.255.0
[C][wifi:366]:   Gateway: 192.168.X.X
[C][wifi:367]:   DNS1: 192.168.X.X
[C][wifi:368]:   DNS2: 0.0.0.0
[D][wifi:514]: Disabling AP...
[C][ota:085]: Over-The-Air Updates:
[C][ota:086]:   Address: esphome-speaker01.local:3232
[C][ota:089]:   Using Password.
[W][ota:095]: Last Boot was an unhandled reset, will proceed to safe mode in 3 restarts
[C][api:025]: Setting up Home Assistant API server...
[D][media_player:059]: 'esphome-speaker01 Lautsprecher' - Setting
[D][media_player:069]:   Volume: 0.50
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x401a7c41  PS      : 0x00060a30  A0      : 0x800da394  A1      : 0x3ffb1a30  
A2      : 0x00000000  A3      : 0x0000000a  A4      : 0x3f800000  A5      : 0x00000000  
A6      : 0x00000015  A7      : 0x00000000  A8      : 0x00000015  A9      : 0x3ffb1a20  
A10     : 0x41280000  A11     : 0x00800000  A12     : 0x000000fe  A13     : 0x3ffb82fc  
A14     : 0x7f800000  A15     : 0x00000000  SAR     : 0x00000008  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000009e  LBEG    : 0x40002390  LEND    : 0x4000239f  LCOUNT  : 0x00000000  

ELF file SHA256: 0000000000000000

Backtrace: 0x401a7c41:0x3ffb1a30 0x400da391:0x3ffb1a50 0x400da5b8:0x3ffb1a80 0x400deb29:0x3ffb1aa0 0x400eb13f:0x3ffb1ac0 0x401a75ff:0x3ffb1b10 0x401a77eb:0x3ffb1b30 0x400ea31d:0x3ffb1b50 0x401a72dd:0x3ffb1b70 0x401a73a9:0x3ffb1b90 0x400e8471:0x3ffb1bb0 0x400ec6ee:0x3ffb1c00 0x40107ca6:0x3ffb1fb0 0x40089dd2:0x3ffb1fd0

Rebooting...
```

### Additional information

**Workaound:** Adding a delay of 10s before setting the volume, the ESP does **not** crash and the volume is set successfully:

```
  on_boot:
    priority: -100
    then:
      - delay: 10s
      - media_player.volume_set:
          id: dac
          volume: 50%
```

I suppose that the crash only happens if HA has not yet connected to the API and/or WiFi is not connected.

nagyrobi, 2022-09-01:
Out of curiosity, @veitw did you try smaller delays? What's the smallest delay that still works?

veitw, 2022-09-01:
I am running currently with a delay of 5s, but I have not tried any shorter delays.

veitw, 2022-09-01:
I did not try any lower priorities than -100 as the docs said something like "-100 basically means after everything is up".

nagyrobi, 2022-09-01:
I guess it's not an exact definition, it's "more or less". Maybe -100 is right on the edge...

closed 2023-01-19 by github-actions[bot]