esphome/issues #3517: .local domains are not accessible

ArnyminerZ, 2022-08-24:
### The problem

My current ESPHome installation on a Raspberry Pi 4 doesn't manage to find the devices on the network.

Trying to access the device's logs on ESPHome Dashboard clearly shows there's an issue finding the device:
```
INFO Reading configuration config/test.yaml...
INFO Starting log output from test.local using esphome API
WARNING Can't connect to ESPHome API for test.local: Error resolving IP address: [Errno -2] Name or service not known
INFO Trying to reconnect to test.local in the background
```
Concerned about a misconfiguration of the docker container, which is ran over docker-compose, hereby my `docker-compose.yml`:
```yaml
services:
  esphome:
    container_name: esphome
    image: esphome/esphome
    user: 0:0
    environment:
      - HOME="/"
    ports:
      - "192.168.1.22:6052:6052"
      - "192.168.1.22:6123:6123"
    volumes:
      - /opt/esphome/config:/config:rw
      - /etc/localtime:/etc/localtime:ro
    devices:
      - /dev/ttyUSB0:/dev/ttyUSB0
    restart: always
    privileged: true
    command: config/ dashboard
```
I've accessed by container's bash terminal with:
```shell
docker exec -it <container-id> bash
```
And trying to ping my device's hostname has issues with the DNS:
```shell
$ ping test.local
ping: test.local: Name or service not known
```
However, pinging the device's IP address (found with Fing on my Android mobile phone):
```shell
$ ping 192.168.1.55
PING 192.168.1.55 (192.168.1.55) 56(84) bytes of data.
64 bytes from 192.168.1.55: icmp_seq=1 ttl=254 time=83.2 ms
64 bytes from 192.168.1.55: icmp_seq=2 ttl=254 time=103 ms
64 bytes from 192.168.1.55: icmp_seq=3 ttl=254 time=137 ms
^C
--- 192.168.1.55 ping statistics ---
3 packets transmitted, 3 received, 0% packet loss, time 2003ms
rtt min/avg/max/mdev = 83.236/107.808/137.458/22.425 ms
```

### Which version of ESPHome has the issue?

2022.8.0

### What type of installation are you using?

Docker

### What platform are you using?

ESP32

### Board

esp32doit-devkit-v1

ArnyminerZ, 2022-08-24:
I've tried removing the custom ports configuration and used the `network_mode` option as recommended by the installation guide:
```
services:
  esphome:
    container_name: esphome
    image: esphome/esphome
    user: 0:0
    environment:
      - HOME="/"
    network_mode: host
#    ports:
#      - "192.168.1.22:6052:6052"
#      - "192.168.1.22:6123:6123"
    volumes:
      - /opt/esphome/config:/config:rw
      - /etc/localtime:/etc/localtime:ro
    devices:
      - /dev/ttyUSB0:/dev/ttyUSB0
    restart: always
    privileged: true
    command: config/ dashboard
```
But it doesn't fix the issue.

ArnyminerZ, 2022-08-24:
Manually configuring the container's DNS to my router's gateway with:
```yaml
dns:
  - 192.168.1.1
  - 8.8.8.8
  - 4.4.4.4
```
Still doesn't make it out.

ssieb, 2022-08-24:
`.local` is resolved with mdns which is a broadcast protocol.  You need host networking on your docker container.  This isn't an esphome issue.

ski7777, 2022-12-08:
Any news here?

johnp789, 2023-01-02:
For now, I'm having good luck with adding avahi-daemon to the ESPHome Docker image.  This way, it doesn't depend on avahi-daemon running on the host.

avahi-entrypoint.sh: 
```shell
#!/bin/bash
set -e
avahi-daemon --daemonize --no-drop-root
exec "$@"
```

Dockerfile:
```dockerfile
FROM esphome/esphome:latest

# Hat tip: https://gnanesh.me/avahi-docker-non-root.html
RUN set -ex \
 && apt-get update && apt-get install -y --no-install-recommends avahi-daemon libnss-mdns \
 # allow hostnames with more labels to be resolved. so that we can
 # resolve node1.mycluster.local.
 # (https://github.com/lathiat/nss-mdns#etcmdnsallow)
 && echo '*' > /etc/mdns.allow \
 # Configure NSSwitch to use the mdns4 plugin so mdns.allow is respected
 && sed -i "s/hosts:.*/hosts:          files mdns4 dns/g" /etc/nsswitch.conf \
 && printf "[server]\nenable-dbus=no\n" >> /etc/avahi/avahi-daemon.conf \
 && chmod 777 /etc/avahi/avahi-daemon.conf \
 && mkdir -p /var/run/avahi-daemon \
 && chown avahi:avahi /var/run/avahi-daemon \
 && chmod 777 /var/run/avahi-daemon

COPY avahi-entrypoint.sh /avahi-entrypoint.sh

ENTRYPOINT ["bash", "/avahi-entrypoint.sh"]

CMD ["/entrypoint.sh", "dashboard", "/config"]
```
Make sure to run with `--network host`!

gargamelonly, 2023-02-08:
Could you help us, the simple people, with a guide how to resolve this issue?

vxav, 2023-04-03:
FYI, I created a repo which builds ESPhome images with @johnp789 's Dockerfile and pushed to dockerhub based on release tags. For now the release process is manual and I haven't added gchr.
https://github.com/esphome/issues/issues/3517#issuecomment-1369256899

BetaRavener, 2023-05-02:
Confirming the issue. If ESPhome relies on mDNS, the docker image should come with Avahi so this works out of the box without need for additional entrypoint files and messing with overriding the base image.

Note that I'm also using `avahi-daemon.conf`:
```
[server]
enable-dbus=no

[publish]
disable-publishing=yes
```

This is to, in my understanding, to disable reliance on dbus (which requires root permisions - I've created another user in the image so it doesn't use root) and disables mDNS publishing, so the container doesn't try to announce itself as host, just resolves `.local` names of other devices.

oarcher, 2023-06-07:
if the problem is to be able to flash a device by OTA with esphome,  you can add `domain: ""` to the `wifi` section in the `.yaml` file.

domain default is `.local`, and  setting it to empty will allow you to reach your device from a docker container.

closed 2023-11-12 by jesserockz

Thom-x, 2023-11-17:
It's still not working for me with https://github.com/esphome/esphome/pull/5733
And https://github.com/esphome/issues/issues/3517#issuecomment-1369256899 is fixing it for me.

ssieb, 2023-11-18:
@Thom-x that sounds like it's a problem with your docker host then.

Thom-x, 2023-11-18:
How can we check that ?
On host `ping xxx.local` works fine. 
The esphome container is using. `network_mode: host` also.

ssieb, 2023-11-18:
Maybe I misunderstood what the workaround was doing.
But esphome uses the zeroconf library directly, so running avahi shouldn't make any difference and might even interfere.  See if you can do some network sniffing to find out if the packets are getting in and out.

schoubi, 2023-12-02:
> For now, I'm having good luck with adding avahi-daemon to the ESPHome Docker image. This way, it doesn't depend on avahi-daemon running on the host.
>
> avahi-entrypoint.sh:
>
> ```shell
> #!/bin/bash
> set [...]

This should be in the documentation.