esphome/issues #3518: lywsd02 No muestra datos

makgiber1984, 2022-08-24:
### The problem

[image]

[image]

The lywsd02 model does not show the temperature and humidity data in Home assistant. I have tried to have the device only in the ESP32, and it doesn't work either, it always comes out in "unknown" data, it seems that it does not connect to the device. The device is newly purchased from the Xiaomi store, and the firmware version it has is 2.0.1_0060 (Hardware F3_A1), I think this may be the problem, since talking to other users who have the same device, it works correctly for them, but it has firmware version 1.1.2_009

[image]
7

### Which version of ESPHome has the issue?

2022.8.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.8.6

### What platform are you using?

ESP32

### Board

az-delivery-devkit-v4

### Example YAML snippet

```yaml
esphome:
  name: gateway-ble

esp32:
  board: az-delivery-devkit-v4
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "___________________________________"

ota:
  password: "-____----------___________"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Gateway-Ble"
    password: "-----______"

captive_portal:

esp32_ble_tracker:

sensor:

  - platform: xiaomi_hhccjcy01
    mac_address: C4:7C:8D:6D:82:5F
    temperature:
      name: "Planta del Dinero Temperatura"
    moisture:
      name: "Planta del Dinero Humedad"
    illuminance:
      name: "Planta del Dinero Iluminacion"
    conductivity:
      name: "Planta del Dinero Fertilidad"
    battery_level:
      name: "Planta del Dinero Bateria"

  - platform: xiaomi_hhccjcy01
    mac_address: C4:7C:8D:6D:82:0D
    temperature:
      name: "Aloe Vera Temperatura"
    moisture:
      name: "Aloe Vera Humedad"
    illuminance:
      name: "Aloe Vera Iluminacion"
    conductivity:
      name: "Aloe Vera Fertilidad"
    battery_level:
      name: "Aloe Vera Bateria"
      
  - platform: xiaomi_hhccjcy01
    mac_address: C4:7C:8D:6D:81:71
    temperature:
      name: "Lengua de Suegra Temperatura"
    moisture:
      name: "Lengua de Suegra Humedad"
    illuminance:
      name: "Lengua de Suegra Iluminacion"
    conductivity:
      name: "Lengua de Suegra Fertilidad"
    battery_level:
      name: "Lengua de Suegra Bateria"
      
  - platform: xiaomi_hhccjcy01
    mac_address: C4:7C:8D:6D:82:18
    temperature:
      name: "Yuca Temperatura"
    moisture:
      name: "Yuca Humedad"
    illuminance:
      name: "Yuca Iluminacion"
    conductivity:
      name: "Yuca Fertilidad"
    battery_level:
      name: "Yuca Bateria"

  - platform: xiaomi_lywsd02
    mac_address: A4:C1:38:D0:A6:04
    temperature:
      name: "Temperatura Dormitorio"
    humidity:
      name: "Humedad Dormitorio"
```

### Anything in the logs that might be useful for us?

```txt
[21:02:35][D][esp32_ble_tracker:731]: Found device A4:C1:38:D0:A6:04 RSSI=-45
[21:02:35][D][esp32_ble_tracker:752]:   Address Type: PUBLIC
[21:02:38][D][xiaomi_ble:328]: Got Xiaomi HHCCJCY01 (C4:7C:8D:6D:82:5F):
[21:02:38][D][xiaomi_ble:331]:   Temperature: 32.6°C
[21:02:38][D][sensor:127]: 'Planta del Dinero Temperatura': Sending state 32.60000 °C with 1 decimals of accuracy
[21:02:38][D][esp32_ble_tracker:731]: Found device 9C:8C:6E:4A:D8:57 RSSI=-90
[21:02:38][D][esp32_ble_tracker:752]:   Address Type: PUBLIC
[21:02:39][D][esp32_ble_tracker:731]: Found device 70:28:45:62:99:D0 RSSI=-89
[21:02:39][D][esp32_ble_tracker:752]:   Address Type: PUBLIC
```

### Additional information

I have other "Mi Plant" devices in the same ESP32 and they give the information correctly as you can see in the images attached at the beginning.

ssieb, 2022-08-25:
Change the logger level to `VERY_VERBOSE` and see what shows up in the logs when it tries to handle the data.

makgiber1984, 2022-08-26:
[20:39:48][VV][esp32_ble_tracker:505]: Parse Result:
[20:39:48][VV][esp32_ble_tracker:522]:   Address: A4:C1:38:D0:A6:04 (PUBLIC)
[20:39:48][VV][esp32_ble_tracker:524]:   RSSI: -48
[20:39:48][VV][esp32_ble_tracker:525]:   Name: ''
[20:39:48][VV][esp32_ble_tracker:533]:   Ad Flag: 6
[20:39:48][VV][esp32_ble_tracker:539]:   Manufacturer data: 10.58.E4.16.2C.04.A6.D0.38.C1.A4 (11)
[20:39:48][VV][esp32_ble_tracker:550]:   Service data:
[20:39:48][VV][esp32_ble_tracker:551]:     UUID: 0xFE95
[20:39:48][VV][esp32_ble_tracker:552]:     Data: 48.58.E4.16.2C.69.7D.F6.D5.C9.78.64.11.09.00.4C.72.01.2E (19)
[20:39:48][VV][esp32_ble_tracker:555]: Adv data: 02.01.06.16.16.95.FE.48.58.E4.16.2C.69.7D.F6.D5.C9.78.64.11.09.00.4C.72.01.2E.0E.FF.8F.03.10.58.E4.16.2C.04.A6.D0.38.C1.A4 (41)
[20:39:48][VV][xiaomi_hhccjcy01:022]: parse_device(): unknown MAC address.
[20:39:48][VV][xiaomi_hhccjcy01:022]: parse_device(): unknown MAC address.
[20:39:48][VV][xiaomi_hhccjcy01:022]: parse_device(): unknown MAC address.
[20:39:48][VV][xiaomi_hhccjcy01:022]: parse_device(): unknown MAC address.
[20:39:48][VV][xiaomi_lywsd02:023]: parse_device(): MAC address A4:C1:38:D0:A6:04 found.
[20:39:48][VV][xiaomi_ble:227]: parse_xiaomi_header(): unknown device, no magic bytes.
[20:39:48][D][esp32_ble_tracker:731]: Found device A4:C1:38:D0:A6:04 RSSI=-48
[20:39:48][D][esp32_ble_tracker:752]:   Address Type: PUBLIC
[20:39:48][V][component:199]: Component esp32_ble_tracker took a long time for an operation (0.12 s).
[20:39:48][V][component:200]: Components should block for at most 20-30ms.

makgiber1984, 2022-08-27:
I still do not understand why it does not work, I am not able to decrypt that data or where the error is.  Any ideas?

makgiber1984, 2022-09-22:
This model was not supposed to encrypt the data. Is it a new version with encrypted data? Is there a way to show them? Thank you.

ssieb, 2022-09-22:
I don't know anything about the device.  I'm just going by what the esphome component has in the code.  And there's no decryption support there.  I don't know if anyone knows how to do that.

makgiber1984, 2022-10-06:
Lo he descubierto con la escritura de [PiotrMachowski]
(https://github.com/PiotrMachowski)
[Xiaomi-cloud-tokens-extractor](https://github.com/PiotrMachowski/Xiaomi-cloud-tokens-extractor)

Que el dispositivo tiene una clave cifrada, pero no me permite agregarla con él, o al asistente doméstico a través de BLE, o con ESPHOME, incluida la línea de enlace...

Como se indicó anteriormente, el termómetro se compró hace muy poco, puede ser una nueva versión con una clave cifrada, pero no puedo añadirlo de ninguna manera...

¿Alguna idea?

[image]
[image]
[image]

makgiber1984, 2022-12-11:
I still can't connect my LYWSD02MMC either through Xiaomi BLE or ESP32, since the ESP32 integration does not allow me to insert a bindbey for this model. Any news? How can I do it?

variat, 2023-12-27:
A year passed. Any findings ? 
I have the same issue.

bahamut657, 2024-01-04:
The device is working great in HA using the xiaomi_ble integration.
It is still NOT working in ESPHome due to the fact that bindkey is not supported by the `xiaomi_lywsd02` platform