esphome/issues #3519: NSPanel cannot upload TFT file to Nextion display

lordjaxom, 2022-08-24:
### The problem

When trying to upload a TFT file to the Nextion display in my NSPanel, I am always getting "Reading from UART timed out at byte 0!".

When flashing Tasmota and uploading the berry driver, the TFT upload is working, only with ESPHome I can't upload. I have tried with the stock firmware on the display (that one that's installed when it's new) and with the one that''s installed after perfoming these steps with Tasmota:
https://docs.nspanel.pky.eu/prepare_nspanel/#flash-firmware-to-nextion-screen

After the messages seen in the log, the device restarts.

I have tried with and without PR#2956

I have also increased the UART timeout to 250, as proposed in another Issue here, with no success.
(as committed in https://github.com/lordjaxom/esphome branch dev)

I have also tried several ESPHome configurations mentioned in this thread:
https://community.home-assistant.io/t/sonoff-nspanel-uploading-tft-to-nextion-display-fails/418693

I have triggered the upload using the attached (stripped) YAML file by sending PRESSED to the MQTT topic nspanel/button/nspanel-dev_update_tft/command

### Which version of ESPHome has the issue?

2022.8.0

### What type of installation are you using?

pip

### What platform are you using?

ESP32

### Board

NSPanel

### Component causing the issue

nextion

### Example YAML snippet

```yaml
# Set some variables for convenience
substitutions:
  node_name: nspanel
  device_name: nspanel-dev

# Note: this may not be needed if the pull request has been merged.
# Check https://github.com/esphome/esphome/pull/2956 for current status.
external_components:
  - source: github://pr#2956
    components: [nextion]
    refresh: 1h

esphome:
  name: $node_name
  comment: $device_name

esp32:
  board: esp32dev

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

# Enable MQTT
mqtt:
  broker: !secret mqtt_broker
  id: mqtt_client
  client_id: $device_name
  discovery: false
  will_message:

# A reboot button is always useful
button:
  - platform: restart
    name: Restart $device_name

  # A button for the TFT upload
  - platform: template
    name: $device_name Update TFT
    device_class: update
    on_press:
      then:
        - lambda: 'id(disp1)->upload_tft();'

# Configure UART for communicating with the screen
uart:
  id: tf_uart
  tx_pin: 16
  rx_pin: 17
  baud_rate: 115200

# Configure the screen itself
display:
  - platform: nextion
    id: disp1
    uart_id: tf_uart
    tft_url: http://192.168.176.40:8080/hmi.tft
```

### Anything in the logs that might be useful for us?

```txt
[22:22:32][D][button:013]: 'nspanel-dev Update TFT' Pressed.
[22:22:32][D][nextion_upload:169]: Connected
[22:22:32][D][nextion_upload:175]: Requesting URL: http://192.168.176.40:8080/hmi.tft
[22:22:32][D][nextion_upload:209]: Updating Nextion NX4832F035_011C...
[22:22:32][D][nextion_upload:235]: Waiting for upgrade response
[22:22:32][E][uart:015]: Reading from UART timed out at byte 0!
[22:22:32][E][uart:015]: Reading from UART timed out at byte 0!
[22:22:32][E][uart:015]: Reading from UART timed out at byte 0!
[22:22:32][E][uart:015]: Reading from UART timed out at byte 0!
[22:22:33][E][uart:015]: Reading from UART timed out at byte 0!
[22:22:33][E][uart:015]: Reading from UART timed out at byte 0!
[22:22:33][E][uart:015]: Reading from UART timed out at byte 0!
[22:22:33][E][uart:015]: Reading from UART timed out at byte 0!
[22:22:33][E][uart:015]: Reading from UART timed out at byte 0!
[22:22:33][E][uart:015]: Reading from UART timed out at byte 0!
[22:22:33][E][uart:015]: Reading from UART timed out at byte 0!
[22:22:33][E][uart:015]: Reading from UART timed out at byte 0!
[22:22:34][E][uart:015]: Reading from UART timed out at byte 0!
[22:22:34][E][uart:015]: Reading from UART timed out at byte 0!
[22:22:34][E][uart:015]: Reading from UART timed out at byte 0!
[22:22:34][E][uart:015]: Reading from UART timed out at byte 0!
[22:22:34][E][uart:015]: Reading from UART timed out at byte 0!
[22:22:34][D][nextion_upload:239]: Upgrade response is  42
[22:22:34][D][nextion_upload:242]: Available 0 : 0x00
[22:22:34][D][nextion_upload:242]: Available 1 : 0x00
[22:22:34][D][nextion_upload:242]: Available 2 : 0x00
[22:22:34][D][nextion_upload:242]: Available 3 : 0x00
```

Joni1983, 2022-08-26:
Same her  :(

Joni1983, 2022-08-26:
[18:38:03][E][uart:015]: Reading from UART timed out at byte 0!
[18:38:03][D][nextion_upload:239]: Upgrade response is U\xbb 43
[18:38:03][D][nextion_upload:242]: Available 0 : 0x55
[18:38:03][D][nextion_upload:242]: Available 1 : 0xBB
[18:38:03][D][nextion_upload:242]: Available 2 : 0x13
[18:38:03][D][nextion_upload:242]: Available 3 : 0x00
[18:38:03][D][nextion_upload:242]: Available 4 : 0x65
[18:38:03][D][nextion_upload:242]: Available 5 : 0x76
[18:38:03][D][nextion_upload:242]: Available 6 : 0x65
[18:38:03][D][nextion_upload:242]: Available 7 : 0x6E
[18:38:03][D][nextion_upload:242]: Available 8 : 0x74
[18:38:03][D][nextion_upload:242]: Available 9 : 0x2C
[18:38:03][D][nextion_upload:242]: Available 10 : 0x73
[18:38:03][D][nextion_upload:242]: Available 11 : 0x74
[18:38:03][D][nextion_upload:242]: Available 12 : 0x61
[18:38:03][D][nextion_upload:242]: Available 13 : 0x72
[18:38:03][D][nextion_upload:242]: Available 14 : 0x74
[18:38:03][D][nextion_upload:242]: Available 15 : 0x75
[18:38:03][D][nextion_upload:242]: Available 16 : 0x70
[18:38:03][D][nextion_upload:242]: Available 17 : 0x2C
[18:38:03][D][nextion_upload:242]: Available 18 : 0x33
[18:38:03][D][nextion_upload:242]: Available 19 : 0x39
[18:38:03][D][nextion_upload:242]: Available 20 : 0x2C
[18:38:03][D][nextion_upload:242]: Available 21 : 0x65
[18:38:03][D][nextion_upload:242]: Available 22 : 0x75
[18:38:03][D][nextion_upload:242]: Available 23 : 0xDC
[18:38:03][D][nextion_upload:242]: Available 24 : 0x50
[18:38:03][D][nextion_upload:242]: Available 25 : 0x50
[18:38:03][D][nextion_upload:242]: Available 26 : 0x50
[18:38:03][D][nextion_upload:242]: Available 27 : 0x50
[18:38:03][D][nextion_upload:242]: Available 28 : 0x50
[18:38:03][D][nextion_upload:242]: Available 29 : 0x50
[18:38:03][D][nextion_upload:242]: Available 30 : 0x50
[18:38:03][D][nextion_upload:242]: Available 31 : 0x50
[18:38:03][D][nextion_upload:242]: Available 32 : 0x50
[18:38:03][D][nextion_upload:242]: Available 33 : 0x50
[18:38:03][D][nextion_upload:242]: Available 34 : 0x50
[18:38:03][D][nextion_upload:242]: Available 35 : 0x50
[18:38:03][D][nextion_upload:242]: Available 36 : 0x50
[18:38:03][D][nextion_upload:242]: Available 37 : 0x50
[18:38:03][D][nextion_upload:242]: Available 38 : 0x50
[18:38:03][D][nextion_upload:242]: Available 39 : 0x50
[18:38:03][D][nextion_upload:242]: Available 40 : 0x50
[18:38:03][D][nextion_upload:242]: Available 41 : 0x50
[18:38:03][D][nextion_upload:242]: Available 42 : 0x50
[18:38:03][D][nextion_upload:248]: preparation for tft update failed 85 "U\xbb"
[18:38:03][D][nextion_upload:324]: Restarting Nextion
INFO nspanel.local: Error while reading incoming messages: Error while reading data: [Errno 104] Connection reset by peer
INFO Disconnected from ESPHome API for nspanel.local
WARNING Disconnected from API
INFO nspanel.local: Ping Failed: Error while reading data: [Errno 104] Connection reset by peer
INFO Successfully connected to nspanel.local

EzzatQ, 2022-09-27:
+1 I am not able to upload the TFT file to my nspanel
Every time I call the upload_tft service from HA, I get the following in my logs:

```
[21:12:06][D][nextion_upload:169]: Connected
[21:12:06][D][nextion_upload:175]: Requesting URL: http://10.0.1.1:8123/local/nspanel/hmi.tft
[21:12:06][D][nextion_upload:209]: Updating Nextion NX4832F035_011C...
[21:12:06][D][nextion_upload:235]: Waiting for upgrade response
[21:12:06][E][uart:015]: Reading from UART timed out at byte 0!
[21:12:07][E][uart:015]: Reading from UART timed out at byte 0!
[21:12:07][E][uart:015]: Reading from UART timed out at byte 0!
[21:12:07][E][uart:015]: Reading from UART timed out at byte 0!
[21:12:07][E][uart:015]: Reading from UART timed out at byte 0!
[21:12:07][E][uart:015]: Reading from UART timed out at byte 0!
[21:12:07][E][uart:015]: Reading from UART timed out at byte 0!
[21:12:07][E][uart:015]: Reading from UART timed out at byte 0!
[21:12:07][E][uart:015]: Reading from UART timed out at byte 0!
[21:12:07][E][uart:015]: Reading from UART timed out at byte 0!
[21:12:08][E][uart:015]: Reading from UART timed out at byte 0!
[21:12:08][E][uart:015]: Reading from UART timed out at byte 0!
[21:12:08][E][uart:015]: Reading from UART timed out at byte 0!
[21:12:08][E][uart:015]: Reading from UART timed out at byte 0!
[21:12:08][E][uart:015]: Reading from UART timed out at byte 0!
[21:12:08][E][uart:015]: Reading from UART timed out at byte 0!
[21:12:08][E][uart:015]: Reading from UART timed out at byte 0!
[21:12:08][E][uart:015]: Reading from UART timed out at byte 0!
[21:12:08][E][uart:015]: Reading from UART timed out at byte 0!
[21:12:08][D][nextion_upload:239]: Upgrade response is  19
[21:12:08][D][nextion_upload:242]: Available 0 : 0x00
[21:12:08][D][nextion_upload:242]: Available 1 : 0x00
[21:12:08][D][nextion_upload:242]: Available 2 : 0x00
[21:12:08][D][nextion_upload:242]: Available 3 : 0x00
[21:12:08][D][nextion_upload:242]: Available 4 : 0x00
[21:12:08][D][nextion_upload:242]: Available 5 : 0x00
[21:12:08][D][nextion_upload:242]: Available 6 : 0x00
[21:12:08][D][nextion_upload:242]: Available 7 : 0x00
[21:12:08][D][nextion_upload:242]: Available 8 : 0x00
[21:12:08][D][nextion_upload:242]: Available 9 : 0x00
[21:12:08][D][nextion_upload:242]: Available 10 : 0x00
[21:12:08][D][nextion_upload:242]: Available 11 : 0x00
[21:12:08][D][nextion_upload:242]: Available 12 : 0x00
[21:12:08][D][nextion_upload:242]: Available 13 : 0x00
[21:12:08][D][nextion_upload:242]: Available 14 : 0x00
[21:12:08][D][nextion_upload:242]: Available 15 : 0x00
[21:12:09][D][nextion_upload:242]: Available 16 : 0x00
[21:12:09][D][nextion_upload:242]: Available 17 : 0x00
[21:12:09][D][nextion_upload:242]: Available 18 : 0x00
[21:12:09][D][nextion_upload:248]: preparation for tft update failed 0 ""
[21:12:09][D][nextion_upload:324]: Restarting Nextion
[21:12:10][D][nextion_upload:327]: Restarting esphome
```
Please help, I've been on this for hours.
I also tried removing everything that's not essential to the upload from my esphome config and this is what I'm left with:

```
substitutions:
  # Name the device and it's entities
  device_name: master_bedroom_nspanel

# Example config.yaml
esphome:
  name: master_bedroom_nspanel
  comment: $device_name

esp32:
  board: esp32dev

# Wifi settings. Add these to your secrets.yaml. fast_connect must be true for a hidden ssid.
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

# API. 
api:
  
  services:
    # Service to update the HMI file
    - service: upload_tft
      then:
        - lambda: 'id(disp1)->upload_tft();'

        -  # Logger. Disable the temperature sensor etc. to focus on the HMI development
logger:
  baud_rate: 0
  level: DEBUG
  logs:
    sensor: WARN
    resistance: WARN
    text_sensor: WARN
    ntc: WARN

# OTA (Over the air updates) password. Add to your secrets.yaml
ota:
  

# Uart for the Nextion display
uart:
  tx_pin: 16
  rx_pin: 17
  baud_rate: 115200
  id: tf_uart

# Functionality for the Nextion display
external_components:
  - source: github://pr#2956
    components: [nextion]
    refresh: 1h

      
# Configure the screen itself
display:
  - platform: nextion
    id: disp1
    uart_id: tf_uart
    tft_url: !secret tft_upload_url
   ```
   
However, I still get the same logs and nothing happens. This is driving me crazy.
(I've already ruined 2 nspanels trying to solder the wires directly to the nextion display and flashing it from nextion editor, And I really want this 3rd one to work 🥲)

lordjaxom, 2022-09-27:
For the moment I have switched to Tasmota (the upload works flawlessly there as stated in my original post), implementing the necessary commands and responses with BerryScript.

Although I really fell in love with the concepts of ESPHome, which I only discovered a few months ago through the NSPanels (after using Tasmota for decades) this sadly is a showstopper :-(

EzzatQ, 2022-09-27:
Any idea what the root cause is?
The weird thing is that the upload initially worked. It stopped working after I updated the esphome config (doesn't seem to be related, but I guess it is), then after some playing around, it worked once more (no clue what made it work since I only changed what the physical buttons on the NSPanel do), and then stopped again (it's been a week now)

I saw there are a lot of threads about this but no one seems to know what the issue really is.

Oh well.
Could you please share what you did with Tasmota. I never worked with tasmota and would prefer to stay in esphome, but If that's the only way it'll work I guess I gotta do what I gotta do

romainrossi, 2022-10-11:
Hi
I came across this topic searching for another problem.
I have 3 nspanel and the upload of the TFT file is working flawlessly.
May I suggest to try something ?
First of all, I had to add a 'on_setup' magic command to prevent a 30 pixel offset between the touch screen and display :
```
display:
  - platform: nextion
    id: disp1
    uart_id: tf_uart
    tft_url: $nextion_update_url
    touch_sleep_timeout: 10
    auto_wake_on_touch: true
    on_setup:
      then:
        - lambda: id(disp1).send_command_printf("lcd_dev fffb 0002 0000 0020"); # Magic command to avoid 32pixel touchscreen offset on NSPanel (https://github.com/UNUF/nxt-doc/blob/main/Protocols/Full%20Instruction%20Set.md#readme)
````

Another thing : on my firsts attempts to upload TFT files on the NSpanel, I had some issues similar to the one you describe. At that time I thought it was my own incompetency so I didn't paid much attention and keep trying.
It finally worked when I uploaded a very small TFT file before uploading the final one.
If you want I can provide the one I used, but I really think it will work with any _very small_ tft file.

I am sorry that I can't be more specific on you issue, but maybe this advice can help.
Good luck !

EzzatQ, 2022-10-12:
thank you @romainrossi for your reply. For once someone gave concrete steps to take to fix this issue.

I will try to add the lambda to my ESPHome config. 
Could you please also provide the TFT file you used ? I think it would also be useful for other people with this issue.

EzzatQ, 2022-10-12:
@romainrossi I tried adding the 2 lines you mentioned 
```
on_setup:
      then:
        - lambda: id(disp1).send_command_printf("lcd_dev fffb 0002 0000 0020"); # Magic command to avoid 32pixel touchscreen offset on NSPanel (https://github.com/UNUF/nxt-doc/blob/main/Protocols/Full%20Instruction%20Set.md#readme)
        - lambda: id(disp1).send_command_printf("DRAKJHSUYDGBNCJHGJKSHBDN");
```
but the upload still failed.
Trying to search for a small TFT file for now

romainrossi, 2022-10-12:
@EzzatQ  Here it is. It is not from me, I used a tft file from another project on Github (dear original author, forgive me as I don't remember who you are).
[small.zip](https://github.com/esphome/issues/files/9766939/small.zip)
I hope it will help...

cybericebyte, 2022-10-18:
I have a hunch about this issue could you please check the tags on your display and tell me what it says the one on the left is the one I have an issue with as you can see the label tag is slightly different. I have been told that is just a printing error but Im not so sure
[image]

EzzatQ, 2022-10-18:
@romainrossi I tried uploading the small tft file but with no avail.
@cybericebyte I have the EU version of the NSPanel. The sticker reads:
```
E32-MSW-NX
100162fdf2
P/N: WW220010
NSPanel-EU
```

EzzatQ, 2022-10-18:
After spending so much time trying to get the TFT file to upload, I decided to go on and try to flash this project: https://github.com/joBr99/nspanel-lovelace-ui and honestly it is so much better and easier than ESPHome

randybb, 2022-10-18:
I am using https://github.com/sairon/esphome-nspanel-lovelace-ui and no issues with tft upload. Have to mentioned that I never had such problem with our "default" nspanel component.

lordjaxom, 2022-10-18:
> After spending so much time trying to get the TFT file to upload, I decided to go on and try to flash this project: https://github.com/joBr99/nspanel-lovelace-ui and honestly it is so much better and easier than ESPHome

I used their instructions to get anything flashed at all. So flashing in general works definitely. Must be something with ESPHome's serial driver IMHO.

I didn't try actually using the Lovelace UI. Do you need special infrastructure for it? I'm using OpenHAB.

cybericebyte, 2022-10-20:
not sure but I think I might be on to a solution. I flashed a FW backup that I had from another panel. Initially I tried this with the esphome flasher tool and I changed the file name to keep track of my backups. then I tried to flash it with esptool.py via the command line but also failed so I changed the name back to fw-backup-4M.bin and it flashed successfully. but now I can't upload the yaml file but I believe this to be due to I need to reboot my router but can not at the moment because I am in the middle of a different upload that will take some time. However before when I was having the problems I did noticed the auto dim out and screen function was not working on the device either but that has changed. so my suggestion at the moment to those having the same issue in to follow the steps given above. if you need the file I used let me know. also make sure the backup you are flashing is the 4mb i also inadvertently flashed a 1mb backup and flashed it to the panel as well but I will keep you posted on my result whenever the other upload finishes.

cybericebyte, 2022-10-21:
ok guys try changing uart baud rate to 9600. thats what solved the issue for me

platima, 2022-10-31:
Hey I am getting this too. I tried replacing the PR with the on_boot action, not having the logger connected, dropping baud to 9600 instead of 115200, trying the 'small' HMI, to no avail. US version of the panel stamped "NSPanel 120 M Board V1.7 \n 2021.05.27" and the sticker says "E32-MSW-NX \n 10015d05c7 \n P/N: WW210954 \n NSPanel-US". After 4 hours I gave up and put the NSPanel back in the drawer. Would love any help!

randybb, 2022-10-31:
nextion will raise an error if is updated with a tft file for a different type
[image]

cybericebyte, 2022-10-31:
> nextion will raise an error if is updated with a tft file for a different type [image]

believe it or not you can still get that error even if its not. I have 7 of these things and I have only tweaked the ui in small places and have gotten that error. sometimes I have to go back a few files upload one from another then upload the one I wanted.

platima, 2022-10-31:
> 

Oh that is good to know! Cheers

platima, 2022-10-31:
> nextion will raise an error if is updated with a tft file for a different type [image]

Oh awesome to know, cheers!

platima, 2022-10-31:
> > 
> 
> Quick check list check ribbon cable from display to board only change the UART baud rate no others if you do not have currently displaying tft file you may need to upload a new one but you should not see the errors in the log. also in my case I discovered with this one oddball display where I had to change the UART the partition for tft is smaller than the others. at one point I had to upload a different tft the small tft did not work for me

I think you may be right re the ribbon cable - re-seating that is the only thing I did not try. Else I might also try flashing it directly via the TF_RX/TX pins to cut the ESPHome code out as the middleman. Would make UI tweaking a lot easier!

cybericebyte, 2022-11-01:
> Would make UI tweaking a lot easier!

agreed!

asferreira, 2022-11-12:
> ### The problem
> 
> When trying to upload a TFT file to the Nextion display in my NSPanel, I am always getting "Reading from UART timed out at byte 0!".
> 
> When flashing Tasmota and uploading the berry driver, the TFT upload is working, only with ESPHome I can't upload. I have tried with the stock firmware on the display (that one that's installed when it's new) and with the one that''s installed after perfoming these steps with Tasmota: https://docs.nspanel.pky.eu/prepare_nspanel/#flash-firmware-to-nextion-screen
> 
> After the messages seen in the log, the device restarts.
> 
> I have tried with and without PR#2956
> 
> I have also increased the UART timeout to 250, as proposed in another Issue here, with no success. (as committed in https://github.com/lordjaxom/esphome branch dev)
> 
> I have also tried several ESPHome configurations mentioned in this thread: https://community.home-assistant.io/t/sonoff-nspanel-uploading-tft-to-nextion-display-fails/418693
> 
> I have triggered the upload using the attached (stripped) YAML file by sending PRESSED to the MQTT topic nspanel/button/nspanel-dev_update_tft/command
> ### Which version of ESPHome has the issue?
> 
> 2022.8.0
> ### What type of installation are you using?
> 
> pip
> ### Which version of Home Assistant has the issue?
> 
> _No response_
> ### What platform are you using?
> 
> ESP32
> ### Board
> 
> NSPanel
> ### Component causing the issue
> 
> nextion
> ### Example YAML snippet
> 
> ```yaml
> # Set some variables for convenience
> substitutions:
>   node_name: nspanel
>   device_name: nspanel-dev
> 
> # Note: this may not be needed if the pull request has been merged.
> # Check https://github.com/esphome/esphome/pull/2956 for current status.
> external_components:
>   - source: github://pr#2956
>     components: [nextion]
>     refresh: 1h
> 
> esphome:
>   name: $node_name
>   comment: $device_name
> 
> esp32:
>   board: esp32dev
> 
> wifi:
>   ssid: !secret wifi_ssid
>   password: !secret wifi_password
> 
> # Enable MQTT
> mqtt:
>   broker: !secret mqtt_broker
>   id: mqtt_client
>   client_id: $device_name
>   discovery: false
>   will_message:
> 
> # A reboot button is always useful
> button:
>   - platform: restart
>     name: Restart $device_name
> 
>   # A button for the TFT upload
>   - platform: template
>     name: $device_name Update TFT
>     device_class: update
>     on_press:
>       then:
>         - lambda: 'id(disp1)->upload_tft();'
> 
> # Configure UART for communicating with the screen
> uart:
>   id: tf_uart
>   tx_pin: 16
>   rx_pin: 17
>   baud_rate: 115200
> 
> # Configure the screen itself
> display:
>   - platform: nextion
>     id: disp1
>     uart_id: tf_uart
>     tft_url: http://192.168.176.40:8080/hmi.tft
> ```
> 
> ### Anything in the logs that might be useful for us?
> 
> ```
> [22:22:32][D][button:013]: 'nspanel-dev Update TFT' Pressed.
> [22:22:32][D][nextion_upload:169]: Connected
> [22:22:32][D][nextion_upload:175]: Requesting URL: http://192.168.176.40:8080/hmi.tft
> [22:22:32][D][nextion_upload:209]: Updating Nextion NX4832F035_011C...
> [22:22:32][D][nextion_upload:235]: Waiting for upgrade response
> [22:22:32][E][uart:015]: Reading from UART timed out at byte 0!
> [22:22:32][E][uart:015]: Reading from UART timed out at byte 0!
> [22:22:32][E][uart:015]: Reading from UART timed out at byte 0!
> [22:22:32][E][uart:015]: Reading from UART timed out at byte 0!
> [22:22:33][E][uart:015]: Reading from UART timed out at byte 0!
> [22:22:33][E][uart:015]: Reading from UART timed out at byte 0!
> [22:22:33][E][uart:015]: Reading from UART timed out at byte 0!
> [22:22:33][E][uart:015]: Reading from UART timed out at byte 0!
> [22:22:33][E][uart:015]: Reading from UART timed out at byte 0!
> [22:22:33][E][uart:015]: Reading from UART timed out at byte 0!
> [22:22:33][E][uart:015]: Reading from UART timed out at byte 0!
> [22:22:33][E][uart:015]: Reading from UART timed out at byte 0!
> [22:22:34][E][uart:015]: Reading from UART timed out at byte 0!
> [22:22:34][E][uart:015]: Reading from UART timed out at byte 0!
> [22:22:34][E][uart:015]: Reading from UART timed out at byte 0!
> [22:22:34][E][uart:015]: Reading from UART timed out at byte 0!
> [22:22:34][E][uart:015]: Reading from UART timed out at byte 0!
> [22:22:34][D][nextion_upload:239]: Upgrade response is  42
> [22:22:34][D][nextion_upload:242]: Available 0 : 0x00
> [22:22:34][D][nextion_upload:242]: Available 1 : 0x00
> [22:22:34][D][nextion_upload:242]: Available 2 : 0x00
> [22:22:34][D][nextion_upload:242]: Available 3 : 0x00
> ```
> 
> ### Additional information
> 
> _No response_

I came across the same issue on my NS Panel (US Version) and  it took me a long time to understand what really was going on! After many attempts to fix it I realized that every time I tried to trigger 'id(disp1)->upload_tft();' from a button component (the same way you did), the firmware update failed.
In one of the attempts I decide to trigger the upload_tft() from the physical NS Panel's button (a gpio binary sensor component) and for my surprise, it worked! 
I still  don't know the reason why triggering the upload_tft() using the button component fails and using the binary sensor does not, but since I've changed it, the firmware update has been working like a charm!

One more thing, I don't know if it is really necessary, but the first successful firmware upload I did after hit by the issue was using the small.tft file.

Hope it helps!

platima, 2022-11-12:
> Quick check list check ribbon cable from display to board only change the UART baud rate no others if you do not have currently displaying tft file you may need to upload a new one but you should [...]

Not the ribbon cable for me :(

platima, 2022-11-12:
> I came across the same issue on my NS Panel (US Version) and it took me a long time to understand what really was going on! After many attempts to fix it I realized that every time I tried to [...]

Hey that is awesome - I have not tried triggering it using a button instead of a service!

I'll likely try that next week. Can I if you have a copy of the factory TFT / HMI file? Cannot find UART signals to download it.

Cheers

jflecool2, 2022-12-02:
Hello,
After doing 8 consecutide TFT upgrade, it happened to me to. 

- Reset doesnt help, 
- 9600 baud rate creates many error from esphome saying it cannot communicate, 
- the upgrade response is sometime 20 sometime 19. 
- I tried the reparse mode escape on boot, doesnt work
- I tried physical button and service, doesnt work
- my firmware is fine since my 2 other nspanel are on the same nspanel firmware and works well. 
- I think there's a mode that persist accross reboot or something... idk what...

jflecool2, 2022-12-02:
Update working on nspanel3:
`
[22:20:57][D][nextion_upload:175]: Requesting URL: http://192.168.11.254:8123/local/tft/hmi_jf_gpsig.tft
[22:20:57][D][nextion_upload:209]: Updating Nextion NX4832F035_011C...
[22:20:57][D][nextion_upload:235]: Waiting for upgrade response
[22:20:58][E][uart:015]: Reading from UART timed out at byte 0!
[22:20:58][E][uart:015]: Reading from UART timed out at byte 0!
[22:20:58][E][uart:015]: Reading from UART timed out at byte 0!
[22:20:58][D][nextion_upload:239]: Upgrade response is  4
[22:20:58][D][nextion_upload:242]: Available 0 : 0x00
[22:20:58][D][nextion_upload:242]: Available 1 : 0x00
[22:20:58][D][nextion_upload:242]: Available 2 : 0x00
[22:20:58][D][nextion_upload:242]: Available 3 : 0x05
[22:20:58][D][nextion_upload:246]: preparation for tft update done
[22:20:58][D][nextion_upload:283]: Allocating buffer size 65536, Heap size is 223476
[22:20:58][D][nextion_upload:305]: Updating tft from "http://192.168.11.254:8123/local/tft/hmi_jf_gpsig.tft" with a file size of 1072140 using 65536 chunksize, Heap Size 154248
[22:20:59][D][nextion_upload:044]: Requesting range: bytes=0-16383
[22:20:59][E][uart:015]: Reading from UART timed out at byte 0!
[22:20:59][E][uart:015]: Reading from UART timed out at byte 0!
[22:20:59][E][uart:015]: Reading from UART timed out at byte 0!
[22:20:59][E][uart:015]: Reading from UART timed out at byte 0!
[22:20:59][E][uart:015]: Reading from UART timed out at byte 0!
[22:20:59][E][uart:015]: Reading from UART timed out at byte 0!
[22:21:00][E][uart:015]: Reading from UART timed out at byte 0!
[22:21:00][E][uart:015]: Reading from UART timed out at byte 0!
[22:21:00][E][uart:015]: Reading from UART timed out at byte 0!
[22:21:00][E][uart:015]: Reading from UART timed out at byte 0!
[22:21:00][E][uart:015]: Reading from UART timed out at byte 0!
[22:21:00][E][uart:015]: Reading from UART timed out at byte 0!
[22:21:00][E][uart:015]: Reading from UART timed out at byte 0!
[22:21:00][E][uart:015]: Reading from UART timed out at byte 0!
[22:21:00][E][uart:015]: Reading from UART timed out at byte 0!
[22:21:01][E][uart:015]: Reading from UART timed out at byte 0!
[22:21:01][E][uart:015]: Reading from UART timed out at byte 0!
[22:21:01][E][uart:015]: Reading from UART timed out at byte 0!
[22:21:01][E][uart:015]: Reading from UART timed out at byte 0!
[22:21:01][D][nextion_upload:117]: Nextion reported new range 786432
[22:21:01][D][nextion_upload:316]: Heap Size 170464, Bytes left 285708
[22:21:01][D][nextion_upload:044]: Requesting range: bytes=786432-851967
[22:21:07][D][nextion_upload:316]: Heap Size 177500, Bytes left 220172
[22:21:07][D][nextion_upload:044]: Requesting range: bytes=851968-917503
`

Update not working on nspanel2
`
[22:32:17][D][sensor:127]: 'NSPanel2 Temperature': Sending state 20.33959 °C with 1 decimals of accuracy
[22:32:25][D][nextion_upload:169]: Connected
[22:32:25][D][nextion_upload:175]: Requesting URL: http://MY_IP:8123/local/tft/hmi_jf_gpsig.tft
[22:32:25][D][nextion_upload:209]: Updating Nextion NX4832F035_011C...
[22:32:25][D][nextion_upload:235]: Waiting for upgrade response
[22:32:25][E][uart:015]: Reading from UART timed out at byte 0!
[22:32:25][E][uart:015]: Reading from UART timed out at byte 0!
[22:32:26][E][uart:015]: Reading from UART timed out at byte 0!
[22:32:26][E][uart:015]: Reading from UART timed out at byte 0!
[22:32:26][E][uart:015]: Reading from UART timed out at byte 0!
[22:32:26][E][uart:015]: Reading from UART timed out at byte 0!
[22:32:26][E][uart:015]: Reading from UART timed out at byte 0!
[22:32:26][E][uart:015]: Reading from UART timed out at byte 0!
[22:32:26][E][uart:015]: Reading from UART timed out at byte 0!
[22:32:26][E][uart:015]: Reading from UART timed out at byte 0!
[22:32:26][E][uart:015]: Reading from UART timed out at byte 0!
[22:32:27][E][uart:015]: Reading from UART timed out at byte 0!
[22:32:27][E][uart:015]: Reading from UART timed out at byte 0!
[22:32:27][E][uart:015]: Reading from UART timed out at byte 0!
[22:32:27][E][uart:015]: Reading from UART timed out at byte 0!
[22:32:27][E][uart:015]: Reading from UART timed out at byte 0!
[22:32:27][E][uart:015]: Reading from UART timed out at byte 0!
[22:32:27][E][uart:015]: Reading from UART timed out at byte 0!
[22:32:27][E][uart:015]: Reading from UART timed out at byte 0!
[22:32:27][E][uart:015]: Reading from UART timed out at byte 0!
[22:32:27][D][nextion_upload:239]: Upgrade response is  20
[22:32:27][D][nextion_upload:242]: Available 0 : 0x00
[22:32:27][D][nextion_upload:242]: Available 1 : 0x00
[22:32:27][D][nextion_upload:242]: Available 2 : 0x00
[22:32:27][D][nextion_upload:242]: Available 3 : 0x00
[22:32:27][D][nextion_upload:242]: Available 4 : 0x00
[22:32:27][D][nextion_upload:242]: Available 5 : 0x00
[22:32:27][D][nextion_upload:242]: Available 6 : 0x00
[22:32:27][D][nextion_upload:242]: Available 7 : 0x00
[22:32:27][D][nextion_upload:242]: Available 8 : 0x00
[22:32:27][D][nextion_upload:242]: Available 9 : 0x00
[22:32:27][D][nextion_upload:242]: Available 10 : 0x00
[22:32:27][D][nextion_upload:242]: Available 11 : 0x00
[22:32:27][D][nextion_upload:242]: Available 12 : 0x00
[22:32:27][D][nextion_upload:242]: Available 13 : 0x00
[22:32:27][D][nextion_upload:242]: Available 14 : 0x00
[22:32:27][D][nextion_upload:242]: Available 15 : 0x00
[22:32:27][D][nextion_upload:242]: Available 16 : 0x00
[22:32:27][D][nextion_upload:242]: Available 17 : 0x00
[22:32:27][D][nextion_upload:242]: Available 18 : 0x00
[22:32:27][D][nextion_upload:242]: Available 19 : 0x00
[22:32:27][D][nextion_upload:248]: preparation for tft update failed 0 ""
[22:32:27][D][nextion_upload:324]: Restarting Nextion`

jflecool2, 2022-12-02:
An example of update not working with update rtesponse 19 instead (still on nspanel2)
`[01:21:52][D][main:476]: upload tft                                                                        `
`[01:21:52][D][nextion_upload:169]: Connected                                                               `
`[01:21:52][D][nextion_upload:175]: Requesting URL: http://192.168.11.254:8123/local/tft/hmi_jf_gpsig.tft   `
`[01:21:52][D][nextion_upload:209]: Updating Nextion NX4832F035_011C...                                     `
`[01:21:52][D][nextion_upload:235]: Waiting for upgrade response                                            `
`[01:21:52][E][uart:015]: Reading from UART timed out at byte 0!                                            `
`[01:21:52][E][uart:015]: Reading from UART timed out at byte 0!                                            `
`[01:21:52][E][uart:015]: Reading from UART timed out at byte 0!                                            `
`[01:21:52][E][uart:015]: Reading from UART timed out at byte 0!                                            `
`[01:21:53][E][uart:015]: Reading from UART timed out at byte 0!                                            `
`[01:21:53][E][uart:015]: Reading from UART timed out at byte 0!                                            `
`[01:21:53][E][uart:015]: Reading from UART timed out at byte 0!                                            `
`[01:21:53][E][uart:015]: Reading from UART timed out at byte 0!                                            `
`[01:21:53][E][uart:015]: Reading from UART timed out at byte 0!                                            `
`[01:21:53][E][uart:015]: Reading from UART timed out at byte 0!                                            `
`[01:21:53][E][uart:015]: Reading from UART timed out at byte 0!                                            `
`[01:21:53][E][uart:015]: Reading from UART timed out at byte 0!                                            `
`[01:21:53][E][uart:015]: Reading from UART timed out at byte 0!                                            `
`[01:21:53][E][uart:015]: Reading from UART timed out at byte 0!                                            `
`[01:21:54][E][uart:015]: Reading from UART timed out at byte 0!                                            `
`[01:21:54][E][uart:015]: Reading from UART timed out at byte 0!                                            `
`[01:21:54][E][uart:015]: Reading from UART timed out at byte 0!                                            `
`[01:21:54][E][uart:015]: Reading from UART timed out at byte 0!                                            `
`[01:21:54][E][uart:015]: Reading from UART timed out at byte 0!                                            `
`[01:21:54][D][nextion_upload:239]: Upgrade response is  19                                                 `
`[01:21:54][D][nextion_upload:242]: Available 0 : 0x00                                                      `
`[01:21:54][D][nextion_upload:242]: Available 1 : 0x00                                                      `
`[01:21:54][D][nextion_upload:242]: Available 2 : 0x00                                                      `
`[01:21:54][D][nextion_upload:242]: Available 3 : 0x00                                                      `
`[01:21:54][D][nextion_upload:242]: Available 4 : 0x00                                                      `
`[01:21:54][D][nextion_upload:242]: Available 5 : 0x00                                                      `
`[01:21:54][D][nextion_upload:242]: Available 6 : 0x00                                                      `
`[01:21:54][D][nextion_upload:242]: Available 7 : 0x00                                                      `
`[01:21:54][D][nextion_upload:242]: Available 8 : 0x00                                                      `
`[01:21:54][D][nextion_upload:242]: Available 9 : 0x00                                                      `
`[01:21:54][D][nextion_upload:242]: Available 10 : 0x00                                                     `
`[01:21:54][D][nextion_upload:242]: Available 11 : 0x00                                                     `
`[01:21:54][D][nextion_upload:242]: Available 12 : 0x00                                                     `
`[01:21:54][D][nextion_upload:242]: Available 13 : 0x00                                                     `
`[01:21:54][D][nextion_upload:242]: Available 14 : 0x00                                                     `
`[01:21:54][D][nextion_upload:242]: Available 15 : 0x00                                                     `
`[01:21:54][D][nextion_upload:242]: Available 16 : 0x00                                                     `
`[01:21:54][D][nextion_upload:242]: Available 17 : 0x00                                                     `
`[01:21:54][D][nextion_upload:242]: Available 18 : 0x00                                                     `
`[01:21:54][D][nextion_upload:248]: preparation for tft update failed 0 ""                                  `
`[01:21:54][D][nextion_upload:324]: Restarting Nextion                                                      `

jflecool2, 2022-12-05:
I think I found the issue. I cannot update TFT of nspanel3 and nspanel5 with the same problem. At this point I'm pretty sure the cause of my issues is sleep. 
In order to ensure a power-cycle (like in a power outage) wouldnt make my nspanel go full brightness on page 0, I made the first page configure sleep and go to sleep
`thsp=0
thup=1
ussp=0
usup=0
wup=3
sleep=1`
Additionally, maybe im stupid, but after going to sleep that way it got woken up by default. Twice. So I made a counter that would put my nspanel to sleep every time it would wake up up to 3 time. That way, esphome would not be able to wake it up.
TLDR: I made my TFT go to sleep very forcibly.
Now I think something is preventing my nspanel from communicating. maybe is_ready, maybe is is_connected, maybe bkcmd=0 or bkcmd=3 sent by esphome doesnt get received because the nspanel is on sleep, and that is necessary for everything that follows, Im not sure.
However, TFT update AND my nspanel->esphome text component doesnt trigger anymore and I would have to wipe my TFT in order to fix it but Im not sure how to do so.

rbr101, 2022-12-08:
Hi,

I had exact the same problem and adding on_boot statement below solved it for me:
```
esphome:
  on_boot:
    priority: 601
    then:
      - lambda: id(disp1).send_command_printf("DRAKJHSUYDGBNCJHGJKSHBDN");
   ```

brotherseb, 2022-12-29:
Hey !
I had the same problem and just fixed it with a mix of all solutions i found :)
1. baud_rate set to 9600
2. commented the external_components section and added this in the esphome section:
  ```
on_boot:
    priority: 601
    then:
      - lambda: id(disp1).send_command_printf("DRAKJHSUYDGBNCJHGJKSHBDN");
```

3. Set the left button to upload the TFT file
```
binary_sensor:
  # Left button below the display
  - platform: gpio
    name: $device_name Left Button
    pin:
      number: 14
      inverted: true
    on_click:
      - lambda: 'id(disp1)->upload_tft();'
```

4. Comment all the actions in the display section (on_setup) so that nothing happens to the screen on boot

5. And finally configured home assistant to use http instead of httpS for the file:
`nextion_update_url: "http://192.168.1.66:8123/local/hmi.tft"` 
(https worked but was VERY long)

Just unplugged and plugged the NSPanel, waited for 2 seconds and pressed the button to start the upload. Still some errors in the log but there was a progress bar on the screen. 50 minutes later it was done ;)

jn3va, 2023-01-01:
I was having the same problem with seeing these after a successful flash of ESPHome

> [14:55:09][E][uart:015]: Reading from UART timed out at byte 0!
> [14:55:09][W][nextion:078]: Nextion is not connected! 

My display was not on and I could not get the TFT upload to work

I saw a post that mentioned that GPIO4 needs to be low to enable the display
Once I added the switch configuration below the display was enabled

```
switch:
  - platform: gpio
    id: panel_power
    entity_category: config
    pin:
      number: 4
      inverted: true
    restore_mode: ALWAYS_ON
```

To upload the TFT these uart settings worked for me

```
uart:
  tx_pin: 16
  rx_pin: 17
  baud_rate: 115200
  id: tf_uart
```

for the display config I used https and not http (http was being blocked since ssl is enabled)

```
display:
  - platform: nextion
    id: disp1
    uart_id: tf_uart
    tft_url: **https**://192.168.1.111:8123/local/nsp.tft
```

I did not have success using on_boot / priority 601 / printf("DRAKJHSUYDGBNCJHGJKSHBDN"); ...

I reverted back and used

```
external_components:
  - source: github://pr#2956
    components: [nextion]
    refresh: 1h
```

After these change I was able to use the call service on the developer tools to trigger the TFT upload!

I had been trying variations of baud rates, tx/rx pins, and the on-boot and external  component settings.  The part that seemed to make the most difference was adding the switch for GPIO4.    Once that was enabled the nextion was active and the rest started working.

jflecool2, 2023-01-09:
Fixed my issue. 
For everyone that made a TFT that sleep on start, here's how to unstuck yourself.
By enabling verbose log:

You should see in logs something like these:
```
[16:33:33][VV][nextion:304]: print_queue_members_ size 0
[16:33:33][VV][nextion:304]: print_queue_members_ size 0
[16:31:32][VV][nextion:202]: print_queue_members_ (top 10) size 1
[16:31:32][VV][nextion:203]: *******************************************
[16:31:32][VV][nextion:213]: Nextion queue type: 4:TEXT_SENSOR , name: argstr1
[16:31:32][VV][nextion:216]: *******************************************
[16:33:33][VV][nextion:304]: print_queue_members_ size 0
[16:33:33][VV][nextion:304]: print_queue_members_ size 0
[16:33:33][VV][nextion:304]: print_queue_members_ size 0
```

If you see other stuff like
`[16:29:49][VV][scheduler:196]: Running interval 'update' with interval=5000 last_execution=125041 (now=130042)`
but dont see print_queue_members_, you are probably in the bad state.

Here's the steps to unstuck yourself:
1. Open Nextion Editor and remove the sleep=1 you put, generate TFT and upload it to your server
2. add this and upload the resulting esphome firmware to your nspanel
```
    - service: customcmd
      variables:
        cmd_str: string
      then:
      - logger.log:
          format: "Force cmd: %s"
          args: [ cmd_str.c_str() ]
      - lambda: id(disp1).write_str(cmd_str.c_str());
      - lambda: id(disp1).write_array({0xFF, 0xFF, 0xFF});
```
3. wake nspanel from sleep using this new API (developer tools -> services -> ESPHome: nspanel2_customcmd)
`sleep=0`
4. repeat until screen wake up
5. enter in the same API the following
`bkcmd=3`
6. Verify that print_queue_members_ now appears in the log
7. use your upload_tft api as you normally would (ESPHome: nspanel2_upload_tft)

codylo, 2023-01-10:
> binary_sensor:
>   # Left button below the display
>   - platform: gpio
>     name: $device_name Left Button
>     pin:
>       number: 14
>       inverted: true
>     on_click:
>       - lambda: 'id(disp1)->upload_tft();'

I can confirm that after I change the trigger of upload tft to use the physical button it works. You can then change back to trigger the relay after the tft file have uploaded to the display.

```
binary_sensor:
  # Left button below the display
  - platform: gpio
    name: $device_name Left Button
    pin:
      number: 14
      inverted: true
    on_click:
      - lambda: 'id(disp1)->upload_tft();'
```

Goldwing1973, 2023-01-10:
> I can confirm that after I change the trigger of upload tft to use the physical button it works. You can then change back to trigger the relay after the tft file have uploaded to the display.
>
> ``` [...]

I had to fully disable the logger to get it working, changing the left button alone didn't solve it for me.

```
# Logger. Disable the temperature sensor etc. to focus on the HMI development
#logger:
#  baud_rate: 0
  #115200
#  level: DEBUG
#  logs:
#    sensor: WARN
#    resistance: WARN
#    text_sensor: WARN
#    ntc: WARN
```

alexandertonino, 2023-01-31:
I tried pretty much the same stuff. But one change made it work:
```
logger:
  baud_rate: 0
```

also. 9600 baud for the UART did not work for me. I had to go back to 115200baud

HACSSYSTEM, 2023-02-06:
Hello Guys!

it should look something like this? 

`substitutions:

###### CHANGE ME START ######

  device_name: "nsp3" 
  wifi_ssid: "xxxxxxxxx"
  wifi_password: "yyyyyy"

#  nextion_update_url: "http://192.168.1.150:8123/local/nspanel_blank.tft" # URL to local tft File
#  nextion_update_url: "https://raw.githubusercontent.com/Blackymas/NSPanel_HA_Blueprint/main/custom_configuration/nspanel_blank.tft" # URL to Github
  nextion_update_url: "http://raw.githubusercontent.com/joBr99/nspanel-lovelace-ui/main/HMI/nspanel.tft" # URL to Github

##### CHANGE ME END #####

##### UART FOR NEXTION DISPLAY #####
uart:
  tx_pin: 16
  rx_pin: 17
  baud_rate: 115200
  id: tf_uart

##### DO NOT CHANGE ANYTHING! #####

binary_sensor:
  # Left button below the display
  - platform: gpio
    name: $device_name Left Button
    pin:
      number: 14
      inverted: true
    on_click:
      - lambda: 'id(disp1)->upload_tft();'

display:
  - platform: nextion
    id: disp1
    uart_id: tf_uart
    tft_url: "http://raw.githubusercontent.com/joBr99/nspanel-lovelace-ui/main/HMI/nspanel.tft"

packages:
  ##### download esphome code from Github
  remote_package:
    url: https://github.com/Blackymas/NSPanel_HA_Blueprint
    ref: main
    files: [nspanel_esphome.yaml]
    refresh: 300s

##### DO NOT CHANGE ANYTHING! #####`

if someone uploads code.
thank you very much... :P

jflecool2, 2023-02-06:
@HACSSYSTEM

I think you're deeply lost in the woods of HA. 

1. Please review the final format of your comment, and re-do it if it doesnt look right. It seems you're not using "code" and thus it all looks crazy. 
2.  Your 3 nextion_update_url has an url you should validate in your browser, one blank thats correct and one that only works with Tasmota on ESP, not ESPHome thats uncommented
3. Then you add Blackymas's project in the middle of your things... ?!? remote_package infers you need substitutions only, everything else in fonfigured by the remote package... 
4. Then those substitutions are not used by your "display" section
5. etc

At this point of being lost, my best advice is to pick nspanel-lovelace-ui or Blackymas or ESPHome custom. Then head on Youtube and picked a tutorial and follow it to a teeth. No forwarding, fast-forward or skipping. Replicate 1:1 what the youtuber is showing.

HACSSYSTEM, 2023-02-06:
Thank you for reading. 

Actually, the problem is that: 

I try to get back from this:

https://docs.nspanel.pky.eu/precare_nspanel/ 

It is also written down how to but fail! Then I decided to get to this version:
https://github.com/blackymas/nspanel_ha_blueprint/wiki/(en)-common-issues-tft-upload-when-nspanel-lvelace-ui-been-in-installed 

But here too, they describe how to. 
Video is also but not all the way you need

Pulpi86, 2023-02-21:
I had the same error, but the HA system was only available with HTTPS access. So I could not upload the .tft file from internal storage. So I downloaded the following program, made an internal HTTP connection, uploaded the .tft file and uploaded Tasmota.bin. There I uploaded it from the console and it uploaded fine, then I uploaded the ESPhome file back and it works fine since then.
https://www.rejetto.com/hfs/?f=dl

I hope I was able to help those who are having the same problem.

OptimusGREEN, 2023-02-21:
Same issue with a standard nextion screen, can't upload tft file anymore. used to work fine.

tjmonk, 2023-02-26:
I had the same problems.  I have  noted 2 things.

1) When I was trying to program the TFT, I was supplying only 3.3V via the programming header (used during ESP32 programming).  I tried the regular things everyone else here has tried but with no success.  I suspect that either a) the 5V rail is required for flash erase, or the current supplied via the 3.3V source I was using was insufficient.   So, I followed Christopher Masto's approach, and put the NSPanel back together and powered it from 120V.  And the TFT upload worked!!

2) The nextion factory baud rate is 115200, but after programming the nspanel with Christopher Masto's TFT firmware, the baud rate changes to 9600 baud, so you will need to update your yaml to reflect the new baud rate after the initial program load.

Hope this helps some others out there.  Power from 120V before trying to flash the TFT file!!!!!

kchiem, 2023-03-06:
> 2\. The nextion factory baud rate is 115200, but after programming the nspanel with Christopher Masto's TFT firmware, the baud rate changes to 9600 baud, so you will need to update your yaml to reflect the new baud rate after the initial program load.

My experience seems to be the opposite.  Came here after trying to flash Blackymas's NSPanel_HA_Blueprint and the TFT flash (at 115200) left me with "System Data ERROR!" on the screen.  The ESPHome "LOGS" output for the device was filled with:

`Reading from UART timed out at byte 0!`

After adding:

```
uart:
  tx_pin: 16
  rx_pin: 17
  baud_rate: 9600
  id: tf_uart
```

.. to my device yaml file and reinstalling, the errors went away.  Then I had it flash [a small blank tft](https://github.com/Blackymas/NSPanel_HA_Blueprint/raw/main/custom_configuration/nspanel_blank.tft) and that succeeded, but after the reboot, the `Reading from UART timed out` errors came back.  Removing the `uart` block above from my yaml file and reinstalling again got rid of the UART errors, and I was able to retry flashing the full tft file again, which happened at 115200.

So, it seems that when the Nextion is working properly, it's communicating at 115200 baud.  But when it errors out, it reverts to 9600 baud.

NvL-WEAP, 2023-03-10:
I tried everything mentioned in the discussions to get the .tft uploaded to the display.

I got errors like;
```
[21:24:21][E][uart:015]: Reading from UART timed out at byte 0!
[21:24:21][W][nextion:078]: Nextion is not connected! 
```

Putting the .tft in homeasstant didn't help, due the upload to the ESP board went quickly. The serial transfer of the .tft to the display would take aprox 20 hours with the speed it was running (transfering a 8Mb file over serial).

I have the same effect on the EU and US version i'm currently testing.

After providing the blank .tft, everything went fine. After the successfully uploaded blank i altered the config back to the needed .tft. Then the upload went fine (within 10min).

So no need to alter the baudrate etc, just use the blank, and change is back to the needed .tft.

My config (stock);
```
substitutions:
  ###### DIT GEDEELTE AANPASSEN ######
  device_name: nspanelus1
  wifi_ssid: !secret wifi_ssid
  wifi_password: !secret wifi_password
  ##### EIND AANPASBARE SECTIE #####
  # nextion_update_url: "http://192.168.2.11:8123/local/nspanel_blank.tft"
  nextion_update_url: "http://192.168.2.11:8123/local/nspanel_us.tft"
packages:
  remote_package:
    url: https://github.com/Blackymas/NSPanel_HA_Blueprint
    ref: main
    files: [nspanel_esphome.yaml]
    refresh: 300s
```

Hope this helps other people who are experiencing the same issue as i did.

giannoug, 2023-03-12:
@NvL-WEAP Can confirm that just flashing [nspanel_blank.tft](https://github.com/Blackymas/NSPanel_HA_Blueprint/blob/main/custom_configuration/nspanel_blank.tft) fixed my upload issues.

jtberge, 2023-03-16:
Ditto, did it the hard way by installing the nextion editor and creating the small version but yes, small first at 9600 baud worked for me, after that the upload of the big boy goes at speed 115200.
How ever i also removed a lot from nspanel_esphome.yaml first to make sure no other processes could intrude the process of updating nextion...
Think this is still not for the faint harted ;)

giannoug, 2023-03-16:
No need to hack anything. I've upgraded (and downgraded) countless times. When it hangs I just flash [nspanel_blank.tft](https://github.com/Blackymas/NSPanel_HA_Blueprint/blob/main/custom_configuration/nspanel_blank.tft) and it starts magically working again.

anarro, 2023-03-17:
Hi, I hava a same problem, I found this settings in Nextion Editor:
[image]

jtberge, 2023-03-17:
> No need to hack anything. I've upgraded (and downgraded) countless times. When it hangs I just flash [nspanel_blank.tft](https://github.com/Blackymas/NSPanel_HA_Blueprint/blob/main/custom_configuration/nspanel_blank.tft) and it starts magically working again.

You may be right, but also hosting the tft files with nginx seemed to help, and isn’t the tftp specific to your display ? EU and US versions differ right ?(square/rectangular) or is that just optics ?

giannoug, 2023-03-17:
> EU and US versions differ right ?(square/rectangular) or is that just optics ?

I believe it's just the orientation. I found the blank file in this repo, [here](https://github.com/Blackymas/NSPanel_HA_Blueprint/tree/main/custom_configuration), so I believe the same file works for both variants

gaetanlord, 2023-03-30:
I got "model does not match" when flashing my US display

Message report model NX4832F035 011C
So possibly the TFT file do not work on all models

jtberge, 2023-03-30:
Correct, the US TFT is different from the EU TFT, now you have to flash it
in safe mode at 9600baud, use the small empty TFT for that, bare minimum
has it’s settings in program.s and is stripped from graphics and fonts.

gaetanlord, 2023-03-30:
Endup installing the tasmota lovelace Ui, and i got something working.
Thank fit the feedback

strutings, 2023-03-30:
Ended up with flashing tasmota and uploaded .tfi for esphome to nextion and then flashed esphome again. Now its working. I guess it has something to do with the file and buffer size of nextion.

axemann, 2023-04-06:
I had the same issue, and ended up resolving it by changing the hostname to use a dash (-) rather than an underscore (_) and re-flashing.  This was after multiple ESPHome re-flashes using a hostname with an underscore that worked, but failed to be able to upload the .tft file.

Piramidowy, 2023-04-07:
Funny thing, I was struggling with this problem for over 3 hours and the reason for tft not uploading, was underscore symbol in it's name. Changing file name from "nspanel_eu.tft" to "nspaneleu.tft" (without underscore!) fixed everything.

nielsvlaarhoven, 2023-04-21:
I have been struggling with this problem for WEEKS ! 
But finally i have the solution, at least it worked for me since i have tried several things that seemed to work for other people but not for me.
i found the solution after reading the comment of EdwardTFN on this HA thread;
https://community.home-assistant.io/t/sonoff-nspanel-blueprint-configure-your-own-custom-nspanel-easy-via-a-blueprint/500577

I ended up putting in the following code;
---------------------------------------------------------------------------------------------------------------------------------------
substitutions:
  #replace this with your own device name and SSID + Passwd
  device_name: "esptest" 
  wifi_ssid: "wifi"
  wifi_password: "password"
  
 #replace this with your own URL of course
  nextion_update_url: "http://192.168.0.119:8123/local/nspanel.tft" 

  baud_rate: "9600"

external_components:
  - source: github://pr#2956
    components: [nextion]
    refresh: 1h

#notice that the ref; is different **dev** in stead of **main**
packages:
  remote_package:
    url: https://github.com/Blackymas/NSPanel_HA_Blueprint
    ref: dev
    files: [nspanel_esphome.yaml]
    refresh: 300s
---------------------------------------------------------------------------------------------------------------------------------------
Try flashing the TFT after putting in this code, it will probably fail.

After this hash out the baud rate setting, and put in the following code;
---------------------------------------------------------------------------------------------------------------------------------------
substitutions:
  #replace this with your own device name and SSID + Passwd
  device_name: "esptest" 
  wifi_ssid: "wifi"
  wifi_password: "password"
  
 #replace this with your own URL of course
  nextion_update_url: "http://192.168.0.119:8123/local/nspanelblank.tft" 
  #nextion_update_url: "https://raw.githubusercontent.com/Blackymas/NSPanel_HA_Blueprint/main/nspanel_eu.tft"

  #baud_rate: "9600"

external_components:
  - source: github://pr#2956
    components: [nextion]
    refresh: 1h

#notice that the ref; is **main** again and not **dev**
packages:
  remote_package:
    url: https://github.com/Blackymas/NSPanel_HA_Blueprint
    ref: main
    files: [nspanel_esphome.yaml]
    refresh: 300s
---------------------------------------------------------------------------------------------------------------------------------------
After doing this i transfered a small (blank) file before transferring the correct .TFT file.

**The explanation of solving the problem was as following;** 

[EdwardTFN]
You shouldn’t have the same problem again. The original settings from Sonoff puts the display on what is called “[Protocol Reparse mode 1](https://nextion.tech/instruction-set/#:~:text=Protocol%20Reparse%20mode%20is%20an,the%20serial%20buffer%20for%20processing)” and this is what makes difficult to update the display with ESPHome. That instruction I’ve user above is to use an old PR (from more then 2 years ago) which always disables this mode, so it allows the TFT transfer.
The PR was rejected as this Protocol Reparse mode might be desired in some cases, but we can still take advantage of that code for this initial upload.
The nspanel_blank.tft, nspanel_eu.tft and nspanel_us.tft.tft all have Protocol Reparse mode disabled, so once it is in your panel, that old code is not needed anymore and you shouldn’t have this problem to transfer TFT files until you reflash your device with original Sonoff files.

I hope this helps at least some of you guys since it took me quite some time to get my TFT files uploaded since it ended up in the same error many of you guys have;
[image]

rleongcs, 2023-04-25:
Sorry for the long post...Just wanted to share my experience as I just flashed a new NSPanel and hopefully can help others.

Everything went quite well following what I did previously when I setup my 2 NSPanels using v3.2.1. Some issues started when I was updating the nspanel_eu.tft file. It would update to some 20+% and then the log would say disconnected by peer. I tried a few more times of loading the blank and then full and same thing happened. I am always able to upload the nspanel_blank.tft file.

Searching through the issues section, I found https://github.com/Blackymas/NSPanel_HA_Blueprint/issues/706#issuecomment-1517979295. Unfortunately, the `nextion_update_url: "https://weap.nl/ha/nspanel_blank.tft"` gave me more issues than the blank tft file I have. The TFT Update seems to be super fast and it no longer gives a White screen but another error screen. I then tried updating the full tft file and also went super fast and then would have lots of errors on the log.

By this time I was quite frustrated and disappointed thinking i might have bricked my NSPanel. I again searched and landed on this thread. The below post helped me solve the issue! Thanks @Piramidowy. I reverted to my previous settings of using my local blank_tft file and once again the White screen is back. With the name of the full tft renamed without the "_", the update went smoothly. I am so happy and relieved! Not sure why the blank tft works and not the eu tft? Maybe due to file size?

> Funny thing, I was struggling with this problem for over 3 hours and the reason for tft not uploading, was underscore symbol in it's name. Changing file name from "nspanel_eu.tft" to "nspaneleu.tft" (without underscore!) fixed everything.

**Below are the versions I have and the steps I used to setup...** 

ESPHome: v2023.4.1
NSPanel ESPHome: v3.2.2
NSPanel Blueprint: v3.2.2
Blank TFT: http://192.168.xxx.xxx:8123/local/nspanelblank.tft (downloaded mine during v3.2.1)
EU TFT: v3.2.2 http://192.168.xxx.xxx:8123/local/nspaneleu.tft

1. Create a new ESPHome device.
2. Give the device a name for example "room3_nspanel". Use "_" and not "-"
3. Choose ESP32 and then "SKIP".
4. ESPHome will create a "room3_nspanel" device with "room3-nspanel.yaml" as the filename.
5. Edit the device and paste settings from https://github.com/Blackymas/NSPanel_HA_Blueprint/wiki/(EN)-First-Steps---Installation-and-Setup#code.
6. Make changes to the device name, wifi and tft url settings. Example mine is as below...
```
substitutions:

###### CHANGE ME START ######

  device_name: "room3_nspanel" 
  wifi_ssid: !secret wifi_ssid
  wifi_password: !secret wifi_password

  nextion_update_url: "http://192.168.xxx.xxx:8123/local/nspanelblank.tft"
  #nextion_update_url: "http://192.168.xxx.xxx:8123/local/nspaneleu.tft" # URL to local tft File
  #nextion_update_url: "https://raw.githubusercontent.com/Blackymas/NSPanel_HA_Blueprint/main/nspanel_eu.tft" # URL to Github

##### CHANGE ME END #####

##### DO NOT CHANGE ANYTHING! #####

packages:
  ##### download esphome code from Github
  remote_package:
    url: https://github.com/Blackymas/NSPanel_HA_Blueprint
    ref: main
    files: [nspanel_esphome.yaml]
    refresh: 300s

##### DO NOT CHANGE ANYTHING! #####
```
7. Save and install. I flashed mine through the "Connected to this compueter method".
8. Fix back the panel to the backplate and power up. You should see the original Sonoff UI.
9. Go to HA "Devices & Services" and you should see a new ESPHome device "room3_nspanel" to be configured. Go ahead and configure it.
10. Click on the ESPHome device, eg. "room3_nspanel" and scroll till you see the "Update TFT display".
11. Open another HA web instance and go to ESPHome and click the "LOGS" of the "room3_nspanel" device to make sure that the Nextion TFT is detected. If you do not see the below, power cycle the switch and then try again.
[image]
12. Once you see the Nextion TFT is detected, go back to the other HA instance and click on "Update TFT display". You should see the NSPanel screen showing update with %. Once 100%, the screen will initialize and show a White screen as it is a blank tft file.
13. Now go edit the ESPHome device and change the tft url from the blank to the actual version.
```
  #nextion_update_url: "http://192.168.xxx.xxx:8123/local/nspanel_blank.tft"
  nextion_update_url: "http://192.168.xxx.xxx:8123/local/nspaneleu.tft" # URL to local tft File
```
15. Save and install. This time I did it over wireless.
16. Repeat steps 11-12. Do not freak if there are errors. You can always power cycle the NSPanel and update the tft again.
17. The actual TFT file will take maybe 10mins if everything went well. You will see the new UI showing you the version screen and then will stuck at loading screen as we have yet to setup the Blueprint. 
18. Go to Automation and create a new automation using the NSPanel Blueprint. Make sure the Node Name matches the device name you used in ESPHome earlier.
[image]
20. Celebrate!

ahmedbadr3, 2023-05-02:
> I have been struggling with this problem for WEEKS ! But finally i have the solution, at least it worked for me since i have tried several things that seemed to work for other people but not for me. i found the solution after reading the comment of EdwardTFN on this HA thread; https://community.home-assistant.io/t/sonoff-nspanel-blueprint-configure-your-own-custom-nspanel-easy-via-a-blueprint/500577
> 
> ## I ended up putting in the following code;
> substitutions: #replace this with your own device name and SSID + Passwd device_name: "esptest" wifi_ssid: "wifi" wifi_password: "password"
> 
> #replace this with your own URL of course nextion_update_url: "http://192.168.0.119:8123/local/nspanel.tft"
> 
> baud_rate: "9600"
> 
> external_components:
> 
> * source: github://pr#2956
>   components: [nextion]
>   refresh: 1h
> 
> ## packages:
> #notice that the ref; is different dev in stead of main
> remote_package:
> url: https://github.com/Blackymas/NSPanel_HA_Blueprint
> ref: dev
> files: [nspanel_esphome.yaml]
> refresh: 300s
> Try flashing the TFT after putting in this code, it will probably fail.
> 
> After this hash out the baud rate setting and change the ref: to main again, code should look like this;
> 
> substitutions: #replace this with your own device name and SSID + Passwd device_name: "esptest" wifi_ssid: "wifi" wifi_password: "password"
> 
> #replace this with your own URL of course nextion_update_url: "http://192.168.0.119:8123/local/nspanelblank.tft" #nextion_update_url: "https://raw.githubusercontent.com/Blackymas/NSPanel_HA_Blueprint/main/nspanel_eu.tft"
> 
> #baud_rate: "9600"
> 
> external_components:
> 
> * source: github://pr#2956
>   components: [nextion]
>   refresh: 1h
> 
> ## packages:
> #notice that the ref; is main again and not dev
> remote_package:
> url: https://github.com/Blackymas/NSPanel_HA_Blueprint
> ref: main
> files: [nspanel_esphome.yaml]
> refresh: 300s
> After doing this i transfered a small (blank) file before transferring the correct .TFT file.
> 
> **The explanation of solving the problem was as following;**
> 
> _[EdwardTFN] You shouldn’t have the same problem again. The original settings from Sonoff puts the display on what is called “[Protocol Reparse mode 1](https://nextion.tech/instruction-set/#:~:text=Protocol%20Reparse%20mode%20is%20an,the%20serial%20buffer%20for%20processing)” and this is what makes difficult to update the display with ESPHome. That instruction I’ve user above is to use an old PR (from more then 2 years ago) which always disables this mode, so it allows the TFT transfer. The PR was rejected as this Protocol Reparse mode might be desired in some cases, but we can still take advantage of that code for this initial upload. The nspanel_blank.tft, nspanel_eu.tft and nspanel_us.tft.tft all have Protocol Reparse mode disabled, so once it is in your panel, that old code is not needed anymore and you shouldn’t have this problem to transfer TFT files until you reflash your device with original Sonoff files._
> 
> I hope this helps at least some of you guys since it took me quite some time to get my TFT files uploaded since it ended up in the same error many of you guys have; [image]

Thank you! This issue was really confusing me - tried a lot of the proposed solutions, but this was a great fix for me.

[16 more replies]

closed 2023-10-10 by jesserockz