esphome/issues #3520: Error1 when compiling for esp8266 in command line

MitchVL, 2022-08-24:
### The problem

This is my first time working with esphome.
I have followed the guide "Getting Started with the ESPHome Command Line" but I get stuck when compiling.
See the build log below.
Even without making changes in the yaml, I still get this error.

What I tried:
- Started off using python 3.9, but I upgraded to 3.10 with a clean install. Python explicitely shows me version 2022.8.0
- Started off using esphome 2022.2.6 (not sure, since I have to rely on memory here), but now I am on 2022.8.0
- Tried removing everything, and starting over again with a new yaml 
- Tried putting the project in another directory (documents folder, c:\esphome\project, ...) to make sure there was nothing going on with permissions or folder names

Note: The error in the log "Het systeem kan het opgegeven pad niet vinden." translates to "The system can not find the given path"/"Given path does not exist"

### Which version of ESPHome has the issue?

2022.8.0

### What type of installation are you using?

pip

### What platform are you using?

ESP8266

### Board

nodemcu

### Component causing the issue

c++ compiler

### Example YAML snippet

```yaml
esphome:
  name: airqual

esp8266:
  board: nodemcu

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: ""

ota:
  password: ""

wifi:
  ssid: "xx"
  password: "xx"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Airqual Fallback Hotspot"
    password: "yPggWoonC3fb"

captive_portal:
```

### Anything in the logs that might be useful for us?

```txt
PS C:\esphome\airqualitysensor> esphome run airqual.yaml
←[32mINFO Reading configuration airqual.yaml...←[0m
←[32mINFO Generating C++ source...←[0m
←[32mINFO Core config or version changed, cleaning build files...←[0m
←[32mINFO Compiling app...←[0m
Processing airqual (board: nodemcu; framework: arduino; platform: platformio/espressif8266 @ 3.2.0)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Library Manager: Installing ottowinter/ESPAsyncTCP-esphome @ 1.2.3
←[32mINFO Installing ←[36mottowinter/ESPAsyncTCP-esphome @ 1.2.3←[0m←[0m
Unpacking  [####################################]  100%
Library Manager: ESPAsyncTCP-esphome@1.2.3 has been installed!
←[32mINFO ←[32mESPAsyncTCP-esphome@1.2.3 has been installed!←[0m←[0m
Library Manager: Installing esphome/ESPAsyncWebServer-esphome @ 2.1.0
←[32mINFO Installing ←[36mesphome/ESPAsyncWebServer-esphome @ 2.1.0←[0m←[0m
Unpacking  [####################################]  100%
Library Manager: ESPAsyncWebServer-esphome@2.1.0 has been installed!
←[32mINFO ←[32mESPAsyncWebServer-esphome@2.1.0 has been installed!←[0m←[0m
Library Manager: Resolving dependencies...
←[32mINFO Resolving dependencies...←[0m
Library Manager: Installing esphome/AsyncTCP-esphome
←[32mINFO Installing ←[36mesphome/AsyncTCP-esphome←[0m←[0m
Unpacking  [####################################]  100%
Library Manager: AsyncTCP-esphome@2.0.0 has been installed!
←[32mINFO ←[32mAsyncTCP-esphome@2.0.0 has been installed!←[0m←[0m
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/nodemcu.html
PLATFORM: Espressif 8266 (3.2.0) > NodeMCU 0.9 (ESP-12 Module)
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES:
 - framework-arduinoespressif8266 @ 3.30002.0 (3.0.2)
 - tool-esptool @ 1.413.0 (4.13)
 - toolchain-xtensa @ 2.100300.210717 (10.3.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ off, Compatibility ~ soft
Found 37 compatible libraries
Scanning dependencies...
Dependency Graph
|-- ESPAsyncTCP-esphome @ 1.2.3
|-- ESPAsyncWebServer-esphome @ 2.1.0
|   |-- ESPAsyncTCP-esphome @ 1.2.3
|   |-- Hash @ 1.0
|   |-- ESP8266WiFi @ 1.0
|-- DNSServer @ 1.1.1
|-- ESP8266WiFi @ 1.0
|-- ESP8266mDNS @ 1.2
Building in release mode
Compiling .pioenvs\airqual\src\esphome\components\api\api_connection.cpp.o
Compiling .pioenvs\airqual\src\esphome\components\api\api_frame_helper.cpp.o
Het systeem kan het opgegeven pad niet vinden.
Compiling .pioenvs\airqual\src\esphome\components\api\api_pb2.cpp.o
Het systeem kan het opgegeven pad niet vinden.
Compiling .pioenvs\airqual\src\esphome\components\api\api_pb2_service.cpp.o
Het systeem kan het opgegeven pad niet vinden.
Het systeem kan het opgegeven pad niet vinden.
Compiling .pioenvs\airqual\src\esphome\components\api\api_server.cpp.o
Compiling .pioenvs\airqual\src\esphome\components\api\list_entities.cpp.o
Compiling .pioenvs\airqual\src\esphome\components\api\proto.cpp.o
Het systeem kan het opgegeven pad niet vinden.
Compiling .pioenvs\airqual\src\esphome\components\api\subscribe_state.cpp.o
Compiling .pioenvs\airqual\src\esphome\components\api\user_services.cpp.o
Het systeem kan het opgegeven pad niet vinden.
Het systeem kan het opgegeven pad niet vinden.
Het systeem kan het opgegeven pad niet vinden.
Het systeem kan het opgegeven pad niet vinden.
Compiling .pioenvs\airqual\src\esphome\components\captive_portal\captive_portal.cpp.o
Compiling .pioenvs\airqual\src\esphome\components\esp8266\core.cpp.o
Compiling .pioenvs\airqual\src\esphome\components\esp8266\gpio.cpp.o
Het systeem kan het opgegeven pad niet vinden.
Het systeem kan het opgegeven pad niet vinden.
Het systeem kan het opgegeven pad niet vinden.
*** [.pioenvs\airqual\src\esphome\components\api\proto.cpp.o] Error 1
*** [.pioenvs\airqual\src\esphome\components\api\api_pb2_service.cpp.o] Error 1
*** [.pioenvs\airqual\src\esphome\components\api\user_services.cpp.o] Error 1
*** [.pioenvs\airqual\src\esphome\components\api\subscribe_state.cpp.o] Error 1
*** [.pioenvs\airqual\src\esphome\components\esp8266\gpio.cpp.o] Error 1
*** [.pioenvs\airqual\src\esphome\components\api\list_entities.cpp.o] Error 1
*** [.pioenvs\airqual\src\esphome\components\api\api_frame_helper.cpp.o] Error 1
*** [.pioenvs\airqual\src\esphome\components\esp8266\core.cpp.o] Error 1
*** [.pioenvs\airqual\src\esphome\components\api\api_connection.cpp.o] Error 1
*** [.pioenvs\airqual\src\esphome\components\api\api_server.cpp.o] Error 1
*** [.pioenvs\airqual\src\esphome\components\captive_portal\captive_portal.cpp.o] Error 1
*** [.pioenvs\airqual\src\esphome\components\api\api_pb2.cpp.o] Error 1
============================================================================================================== [FAILED] Took 5.94 seconds ==============================================================================================================
```

aqxa1, 2022-08-28:
There might be more issues on your system, but it looks like python 3.10 is broken with `esphome` for me (on two systems). The third system with python 3.9 worked fine. So you might want to focus on trying to get it working with python 3.9 first.

Hopefully somehow one weighs in as to whether `esphome` supports python 3.10 at all yet, or if there is something else going on.

ssieb, 2022-08-28:
I have no problem with Python 3.10.

closed 2023-01-10 by github-actions[bot]