esphome/issues #3522: fatal error: lwip/inet.h: No such file or directory

Mountain-P, 2022-08-25:
### The problem

I can't complie

`INFO Reading configuration livingroom.yaml...
WARNING 'door_av': Using the '_' (underscore) character in the hostname is discouraged as it can cause problems with some DHCP and local name services. For more information, see https://esphome.io/guides/faq.html#why-shouldn-t-i-use-underscores-in-my-device-name
INFO Generating C++ source...
INFO Core config or version changed, cleaning build files...
INFO Compiling app...
Processing door_av (board: nodemcuv2; framework: arduino; platform: platformio/espressif8266 @ 3.2.0)
--------------------------------------------------------------------------------
Library Manager: Installing ottowinter/ESPAsyncTCP-esphome @ 1.2.3
INFO Installing ottowinter/ESPAsyncTCP-esphome @ 1.2.3
Unpacking  [####################################]  100%
Library Manager: ESPAsyncTCP-esphome@1.2.3 has been installed!
INFO ESPAsyncTCP-esphome@1.2.3 has been installed!
Library Manager: Installing esphome/ESPAsyncWebServer-esphome @ 2.1.0
INFO Installing esphome/ESPAsyncWebServer-esphome @ 2.1.0
Unpacking  [####################################]  100%
Library Manager: ESPAsyncWebServer-esphome@2.1.0 has been installed!
INFO ESPAsyncWebServer-esphome@2.1.0 has been installed!
Library Manager: Resolving dependencies...
INFO Resolving dependencies...
Library Manager: Installing esphome/AsyncTCP-esphome
INFO Installing esphome/AsyncTCP-esphome
Unpacking  [####################################]  100%
Library Manager: AsyncTCP-esphome@2.0.0 has been installed!
INFO AsyncTCP-esphome@2.0.0 has been installed!
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- ESPAsyncTCP-esphome @ 1.2.3
|-- ESPAsyncWebServer-esphome @ 2.1.0
|   |-- ESPAsyncTCP-esphome @ 1.2.3
|   |-- Hash @ 1.0
|   |-- ESP8266WiFi @ 1.0
|-- DNSServer @ 1.1.1
|-- ESP8266WiFi @ 1.0
|-- ESP8266mDNS @ 1.2
Compiling .pioenvs/door_av/src/esphome/components/api/api_connection.cpp.o
Compiling .pioenvs/door_av/src/esphome/components/api/api_frame_helper.cpp.o
Compiling .pioenvs/door_av/src/esphome/components/api/api_pb2.cpp.o
Compiling .pioenvs/door_av/src/esphome/components/api/api_pb2_service.cpp.o
Compiling .pioenvs/door_av/src/esphome/components/api/api_server.cpp.o
Compiling .pioenvs/door_av/src/esphome/components/api/list_entities.cpp.o
Compiling .pioenvs/door_av/src/esphome/components/api/proto.cpp.o
Compiling .pioenvs/door_av/src/esphome/components/api/subscribe_state.cpp.o
In file included from src/esphome/components/socket/socket.h:5,
                 from src/esphome/components/api/api_frame_helper.h:13,
                 from src/esphome/components/api/api_frame_helper.cpp:1:
src/esphome/components/socket/headers.h:10:10: fatal error: lwip/inet.h: No such file or directory
   10 | #include "lwip/inet.h"
      |          ^~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/door_av/src/esphome/components/api/api_frame_helper.cpp.o] Error 1
In file included from src/esphome/components/socket/socket.h:5,
                 from src/esphome/components/api/api_server.h:7,
                 from src/esphome/components/api/api_server.cpp:1:
src/esphome/components/socket/headers.h:10:10: fatal error: lwip/inet.h: No such file or directory
   10 | #include "lwip/inet.h"
      |          ^~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/door_av/src/esphome/components/api/api_server.cpp.o] Error 1
In file included from src/esphome/components/socket/socket.h:5,
                 from src/esphome/components/api/api_server.h:7,
                 from src/esphome/components/api/api_connection.h:7,
                 from src/esphome/components/api/list_entities.cpp:5:
src/esphome/components/socket/headers.h:10:10: fatal error: lwip/inet.h: No such file or directory
   10 | #include "lwip/inet.h"
      |          ^~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/door_av/src/esphome/components/api/list_entities.cpp.o] Error 1
In file included from src/esphome/components/socket/socket.h:5,
                 from src/esphome/components/api/api_server.h:7,
                 from src/esphome/components/api/api_connection.h:7,
                 from src/esphome/components/api/api_connection.cpp:1:
src/esphome/components/socket/headers.h:10:10: fatal error: lwip/inet.h: No such file or directory
   10 | #include "lwip/inet.h"
      |          ^~~~~~~~~~~~~
compilation terminated.
In file included from src/esphome/components/socket/socket.h:5,
                 from src/esphome/components/api/api_server.h:7,
                 from src/esphome/components/api/api_connection.h:7,
                 from src/esphome/components/api/subscribe_state.cpp:2:
src/esphome/components/socket/headers.h:10:10: fatal error: lwip/inet.h: No such file or directory
   10 | #include "lwip/inet.h"
      |          ^~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/door_av/src/esphome/components/api/api_connection.cpp.o] Error 1
*** [.pioenvs/door_av/src/esphome/components/api/subscribe_state.cpp.o] Error 1
========================= [FAILED] Took 11.83 seconds =========================`

### Which version of ESPHome has the issue?

Latest

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

no

### What platform are you using?

ESP8266

ssieb, 2022-08-25:
`latest` is not a version.  You also need to include your yaml, preferably a minimal one that reproduces this.

MattWestb, 2022-08-27:
I was having the same problem with  the first warning then i have all my ESP devices with  underscore in the device name.
I was finding the easiest way fixing it was way updating the device name without under score like this (was having underscore before)
```
  devicename: esp32-cam
```
and then updating the firmware thru the web  GUI and was working OK for all device with enough flash but with ESP8255 was having to old firmware / boarder so need GZIP the firmware before the device was accepting it and then online OTA was working OK and also ESPHome integration is finding under the new name without problems and no need struggling with temporary names / IP-address.

closed 2023-01-09 by github-actions[bot]