esphome/issues #3524: esp32cam problem: "[esp32_camera:095]: Setup Failed: ERROR"

andreklug, 2022-08-27:
### The problem

Hi all, 

I try to set up an Esp32cam but always run in the error above. The module I am using is this one (although from another vendor but the pin outs are the same): https://microcontrollerslab.com/esp32-cam-ai-thinker-pinout-gpio-pins-features-how-to-program/

While in general everything else seems to work, I cannot get the camera working. The error is the same with three modules I have, I also tried different power supplies (currently on breadboard with external 5V 1200mA supply), so I believe that's not the issue.

[10:14:00][E][esp32_camera:095]:   Setup Failed: ERROR

I am using the esp-idf framework because I was unable to connect to my WPA3 network with the arduino framework. Maybe that's the reason, but I cannot figure out how to connect otherwise.

Thank you for looking into this!

### Which version of ESPHome has the issue?

2022.8.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.8.7

### What platform are you using?

ESP32-IDF

### Board

esp32cam AI-thinker

### Component causing the issue

camera

### Example YAML snippet

```yaml
esphome:
  name: espc2

esp32:
  board: esp32cam
  framework:
    type: esp-idf
    version: recommended

logger:
  level: VERY_VERBOSE

substitutions:
  # Modify variables based on your settings
  hostname: 'espc2'

wifi:
  ssid: !secret wifi_ssid_btsnv
  password: !secret wifi_password_btsnv

  manual_ip:
    static_ip: 192.168.53.42
    gateway: 192.168.53.1
    subnet: 255.255.255.0
    dns1: 192.168.53.10

  ap:
    ssid: "ec2 Fallback Hotspot"
    password: !secret espcam1_ap_passwd
[... 181 lines cut ...]
Compiling /data/espc2/.pioenvs/espc2/liba52/esp32-camera/sensors/ov5640.o
cc1: warning: command line option '-fno-rtti' is valid for C++/ObjC++ but not for C
Compiling /data/espc2/.pioenvs/espc2/liba52/esp32-camera/sensors/ov7725.o
cc1: warning: command line option '-fno-rtti' is valid for C++/ObjC++ but not for C
Archiving /data/espc2/.pioenvs/espc2/liba52/libesp32-camera.a
Linking /data/espc2/.pioenvs/espc2/firmware.elf
RAM:   [=         ]  10.5% (used 34464 bytes from 327680 bytes)
Flash: [=====     ]  46.0% (used 844777 bytes from 1835008 bytes)
Building /data/espc2/.pioenvs/espc2/firmware.bin
esp32_create_combined_bin(["/data/espc2/.pioenvs/espc2/firmware.bin"], ["/data/espc2/.pioenvs/espc2/firmware.elf"])
Wrote 0xde570 bytes to file /data/espc2/.pioenvs/espc2/firmware-factory.bin, ready to flash to offset 0x0
========================= [SUCCESS] Took 61.92 seconds =========================
INFO Successfully compiled program.
INFO Connecting to 192.168.53.42
INFO Uploading /data/espc2/.pioenvs/espc2/firmware.bin (845168 bytes)
Uploading: [============================================================] 100% Done...

INFO Waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from 192.168.53.42 using esphome API
INFO Successfully connected to 192.168.53.42
[20:42:14][I][app:102]: ESPHome version 2022.8.0 compiled on Aug 26 2022, 20:41:12

```

CarlosGS, 2022-09-06:
Hi! I don't have a solution, but can confirm this happens only under esp-idf. Seems something's missing in the driver :S

hankanman, 2022-10-12:
Can confirm similar issue, if any integration other than just the camera is added then camera setup fails:

```
[C][ledc.output:118]: LEDC Output:
[C][ledc.output:119]:   Pin GPIO4
[C][ledc.output:120]:   LEDC Channel: 2
[C][ledc.output:121]:   Frequency: 1000.0 Hz
[C][esp32_camera:048]: ESP32 Camera:
[C][esp32_camera:049]:   Name: Lounge ESP32CAM
[C][esp32_camera:050]:   Internal: NO
[C][esp32_camera:052]:   Data Pins: D0:5 D1:18 D2:19 D3:21 D4:36 D5:39 D6:34 D7:35
[C][esp32_camera:053]:   VSYNC Pin: 25
[C][esp32_camera:054]:   HREF Pin: 23
[C][esp32_camera:055]:   Pixel Clock Pin: 22
[C][esp32_camera:056]:   External Clock: Pin:0 Frequency:20000000
[C][esp32_camera:057]:   I2C Pins: SDA:26 SCL:27
[C][esp32_camera:058]:   Reset Pin: -1
[C][esp32_camera:076]:   Resolution: 640x480 (VGA)
[E][esp32_camera:095]:   Setup Failed: ERROR
```

If you remove all integrations except the camera it works fine
The big issue here is that it won't work with the camera web server integration, which kind makes it pointless!

aderusha, 2022-11-01:
Seeing the same behavior here as well.  If I use the arduino framework, things kind of work but performance is miserable, enabling the camera component results in 70%+ packet loss and latencies over 2000ms for the packets that do come through.  Switching from arduino to esp-idf results in Setup Failed.

EduardTheThird, 2022-11-08:
Logged the same issue in May 2022, https://github.com/esphome/issues/issues/3300.

Mpercy-Git, 2022-11-24:
Same issue with two different boards I have.

pieterjandebruyne, 2022-12-01:
Same issue with this config:
```
esphome:
  name: "esp-cam"

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esphome-Web-2A9990"
    password: "dslLO2kUXXX"

captive_portal:

esp32_camera:
  name: Cam
  external_clock:
    pin: GPIO0
    frequency: 20MHz
  i2c_pins:
    sda: GPIO26
    scl: GPIO27
  data_pins: [GPIO5, GPIO18, GPIO19, GPIO21, GPIO36, GPIO39, GPIO34, GPIO35]
  vsync_pin: GPIO25
  href_pin: GPIO23
  pixel_clock_pin: GPIO22
  reset_pin: GPIO32
  resolution: 640x480
  jpeg_quality: 10    
  ```
  
  using this board: https://nl.aliexpress.com/item/1005003472117545.html
  
  Additional info from Validating the yaml file:
  ```
  esphome:
    name: espcam-chili
    build_path: .esphome/build/espcam-chili
    platformio_options: {}
    includes: []
    libraries: []
    name_add_mac_suffix: false
    min_version: 2022.11.4
esp32:
    board: esp32dev
    framework:
      version: 1.0.6
      source: ~3.10006.0
      platform_version: platformio/espressif32 @ 3.5.0
      type: arduino
    variant: ESP32
  ```

JacksDR, 2022-12-07:
Same issue with my boards

pieterjandebruyne, 2022-12-07:
Update: boards and cam are working on arduino with the cam webserver template.. so the issue should be within esphome implementation or config

yacinegithub, 2023-01-09:
Same issue using this board: [](https://nl.aliexpress.com/item/1005003472117545.html)  with my code

esp32_camera:
  name: "MADKHAL MOCHTARAk CAM"
  external_clock:
    pin: GPIO3
    frequency: 20MHz
  i2c_pins:
    sda: GPIO26
    scl: GPIO27
  data_pins: [GPIO5, GPIO18, GPIO19, GPIO21, GPIO36, GPIO39, GPIO34, GPIO35]
  vsync_pin: GPIO25
  href_pin: GPIO23
  pixel_clock_pin: GPIO22
  power_down_pin: GPIO32
# Image settings
  max_framerate: 5 fps # default: 10 fps, max 60
  idle_framerate: 0.05 fps # default: 0.1 fps - framerate for 'picture' in HA dashboard
  resolution: 800x600 # default: 640x480 (VGA) - higher res requires more memory
  jpeg_quality: 10 # 10 (best) to 63 (worst)

rradar, 2023-01-19:
Can anyone confirm (or deny) having that particular version/print on the camera flat cable?

> [image]
>
> _TY-OV2 640-V2.0_

I suspect that the issue lies in that particular camera module as they are the only ones that don't work for me but all other work.

takerukoushirou, 2023-01-20:
> Can anyone confirm (or deny) having that particular version/print on the camera flat cable?
> 
> > _TY-OV2 640-V2.0_
> 
> I suspect that the issue lies in that particular camera module as they are the only ones that don't work for me but all other work.

I have three of these in use and two work fine on arduino framework, the third doesn't (probably faulty).

MartinTheBruce, 2023-06-03:
I have identical issue… second I switch from Arduino to idf framework I get the camera setup error; switching back to Arduino, and it’s not performant. I need the idf framework as looking for Bluetooth proxy active connections too! Maybe asking too much for the cam.

berga, 2023-08-07:
Hi all, 
by my side the problem is raising when using the 120 degrees 21mm camera sensor: "DCX-OV2640-v2"
I have other 2 sensors, one with the same label but normal angle and one "DC-2640-v3" and they are working flawlessly
with the custom component from https://github.com/MichaKersloot/esphome_custom_components

my config is:
```
esp32_camera:
  external_clock:
    pin: GPIO0
    frequency: 20MHz
  i2c_pins:
    sda: GPIO26
    scl: GPIO27
  data_pins: [GPIO5, GPIO18, GPIO19, GPIO21, GPIO36, GPIO39, GPIO34, GPIO35]
  vsync_pin: GPIO25
  href_pin: GPIO23
  pixel_clock_pin: GPIO22
  power_down_pin: GPIO32
  idle_framerate: 0.1fps
  resolution: 640x480
  name: ${upper_devicename} 
```

DooMMasteR, 2024-06-17:
I have the same issue I guess.
```yaml
esp32:
  board: esp32-s3-devkitc-1
  flash_size: 16MB
  framework:
    type: esp-idf
    version: recommended
```
with
```yaml
esp32_camera:
  name: My Camera
  external_clock:
    pin: GPIO15
    frequency: 20MHz
  i2c_pins:
    sda: GPIO4
    scl: GPIO5
  data_pins: [GPIO11, GPIO9, GPIO8, GPIO10, GPIO12, GPIO18, GPIO17, GPIO16]
  vsync_pin: GPIO6
  href_pin: GPIO7
  pixel_clock_pin: GPIO13
  resolution: 1600X1200
  jpeg_quality: 10
  aec2: True
  agc_gain_ceiling: 128X
```

flaviobattimo, 2024-10-15:
> Hi all, by my side the problem is raising when using the 120 degrees 21mm camera sensor: "DCX-OV2640-v2" I have other 2 sensors, one with the same label but normal angle and one "DC-2640-v3" and [...]

Hello, did you figured out how to fix the issue with the 120 degrees 21mm camera sensor?