esphome/issues #3525: Compilation problem, default_target_temperature_low mandatory albeit being deprecated

posterberg, 2022-08-27:
### The problem

It's not possible to compile the component without including default_target_temperature_low in the config for a heater device. This is odd since the documentation says that the option is deprecated.
The presets part of the config has default_target_temperature_low, but it doesn't seem to be sufficient.

### Which version of ESPHome has the issue?

2022-08-0

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

n/a

### What platform are you using?

ESP8266

### Board

sonoff basic

### Component causing the issue

climate.thermostat

### Example YAML snippet

```yaml
climate:
  - platform: thermostat
    name: "esp_climate_bedroom_kp_thermostat"
    sensor: temperature
    heat_action:
      - switch.turn_on: relay
    idle_action:
      - switch.turn_off: relay
    min_heating_off_time: 300s
    min_heating_run_time: 10s
    min_idle_time: 59s
    heat_deadband: 0.5
    heat_overrun: 0.5
    preset:
      - name: home
        mode: heat
        default_target_temperature_low: 19
      - name: sleep
        mode: heat
        default_target_temperature_low: 17
      - name: away
        mode: heat
        default_target_temperature_low: 12
      - name: eco
        mode: heat
        default_target_temperature_low: 15
```

### Anything in the logs that might be useful for us?

```txt
Failed config

climate.thermostat: [source climate_radiator_bedroom/climate_radiator_bedroom.yaml:62]

  default_target_temperature_low must be defined in default config when using heat_action.
```

ssieb, 2022-09-15:
@kbx81 is this because of a missing `default_preset:`?

kbx81, 2022-09-15:
`default_preset:` doesn't exist in the version mentioned above -- that'll most likely be in the October release.

ssieb, 2022-09-15:
Then what is causing this error?

kbx81, 2022-09-15:
`default_target_temperature_low must be defined in default config when using heat_action.` is intentional. When only using heat, the lower set point is always required, else it cannot function correctly/at all.

Until we are fully "migrated" to the presets model, the option is still required because presets are (still) optional (for now) and the component cannot operate correctly if this option is not provided and presets are not defined. It would probably be good to make the validation "smarter" (and maybe we can weasel this into https://github.com/esphome/esphome/pull/3643) so that if any presets are defined, this option becomes optional (but is required otherwise).

_Personally_ I don't really see this as an issue because, for now, the message about depreciation is little more than a "heads-up". The option has been flagged as "deprecated" because yes, it will be removed in a future release. For now, it's still required to continue supporting the earlier model, but this will be changed when the "new" presets model is eventually enforced and the previous model is abandoned and ultimately removed.

ssieb, 2022-09-15:
Ok, but "deprecated" normally means that you shouldn't be using it, or at least that you don't have to use it.  You mark something as deprecated once there is a replacement available.

kbx81, 2022-09-15:
That's fair. If it's not already, we'll fix it in https://github.com/esphome/esphome/pull/3643 (which is basically ready to merge, I'll confirm this...probably tonight) or create another PR.

kbx81, 2022-09-15:
So this issue is "fixed" in https://github.com/esphome/esphome/pull/3643 in the sense that these config options (`default_target_temperature_high` and `default_target_temperature_low`) are removed.

Maybe we can squeeze an intermediate PR into this release to fix this specific issue; I was planning to merge https://github.com/esphome/esphome/pull/3643 soon (next week?) so it would go into the October release.

MrMDavidson, 2022-09-24:
I've opened esphome/esphome#2324 to correct the use of deprecated and instead providing a note advising these are going away.