esphome/issues #3526: climate.thermostat crashes when using presets

posterberg, 2022-08-27:
### The problem

The climate.thermostat crasches repeatedly when including the preset option. I have no crashes using the exact same configuration without the preset option.

### Which version of ESPHome has the issue?

2022-08-0

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

n/a

### What platform are you using?

ESP8266

### Board

sonoff basic

### Component causing the issue

climate.thermostat

### Example YAML snippet

```yaml
Causing constant crashes:
climate:
  - platform: thermostat
    name: "esp_climate_bedroom_kp_thermostat"
    sensor: temperature
    heat_action:
      - switch.turn_on: relay
    idle_action:
      - switch.turn_off: relay
    min_heating_off_time: 300s
    min_heating_run_time: 10s
    min_idle_time: 59s
    heat_deadband: 0.5
    heat_overrun: 0.5
    default_target_temperature_low: 19
    preset:
      - name: home
        mode: heat
        default_target_temperature_low: 19
      - name: sleep
        mode: heat
        default_target_temperature_low: 17
      - name: away
        mode: heat
        default_target_temperature_low: 12
      - name: eco
        mode: heat
        default_target_temperature_low: 15

Not causing crashes:
climate:
  - platform: thermostat
    name: "esp_climate_bedroom_kp_thermostat"
    sensor: temperature
    heat_action:
      - switch.turn_on: relay
    idle_action:
      - switch.turn_off: relay
    min_heating_off_time: 300s
[... 114 lines cut ...]
[18:59:08]3fffff60:  3fff0001 00006130 feefef00 feefeffe  
[18:59:08]3fffff70:  3fff0010 3fffff5c 3ffef0bc 3fffff44  
[18:59:08]3fffff80:  00000000 00000000 00000001 3ffef3bc  
[18:59:08]3fffff90:  3fffdad0 00000000 3ffef3a8 40215768  
WARNING Decoded 0x40215768: Stream::timedRead()
[18:59:08]3fffffa0:  3fffdad0 00000000 3ffef3a8 4021df10  
WARNING Decoded 0x4021df10: memmem_P at /workdir/repo/newlib/newlib/libc/sys/xtensa/string_pgmspace.c:217
[18:59:08]3fffffb0:  feefeffe feefeffe 3ffe85d8 40100831  
WARNING Decoded 0x40100831: detachInterrupt
[18:59:08]<<<stack<<<
[18:59:08]
[18:59:08]--------------- CUT HERE FOR EXCEPTION DECODER ---------------
[18:59:08]
[18:59:08] ets Jan  8 2013,rst cause:1, boot mode:(3,0)
[18:59:08]
[18:59:08]load 0x4010f000, len 3460, room 16 
[18:59:08]tail 4
[18:59:08]chksum 0xcc
[18:59:08]load 0x3fff20b8, len 40, room 4 
[18:59:08]tail 4
[18:59:08]chksum 0xc9
[18:59:08]csum 0xc9
[18:59:08]v00063250
[18:59:08]~ld
```

tommy4st, 2022-09-20:
I can confirm this. My workaround right now is to move the thermostats to home assistant, which I like to revert as soon as possible.

oxc, 2022-09-24:
Do you have the same problem if you use just the home and the away preset? 

This might be a red herring, and/or unrelated, but I've found that I can use those two, but as soon as I use any other presets the device does not react anymore (I have not seen any stacktraces, but I'm not sure if they would show up over OTA logs. They do stop at the same point, though).

cubinet-code, 2022-10-03:
I have the same observation as oxc. home and away presets works, configuring any other preset causes the stack trace above. So for example, the below config works. Uncommenting the ECO preset will cause the crash:

```   
preset:
    - name: HOME
      default_target_temperature_low: 22
      mode: HEAT
    # - name: ECO
    #   default_target_temperature_low: 20
    #   mode: HEAT
    - name: AWAY
      default_target_temperature_low: 18
      mode: HEAT
```

kbx81, 2022-10-04:
I expect this should be fixed in either esphome/esphome#3863 or esphome/esphome#3864 -- could someone here please confirm?

oxc, 2022-10-04:
I can confirm that https://github.com/esphome/esphome/pull/3863 fixes this problem for me. 

I can use all presets, they get dumped correctly in the log, and I can use them in the UI as expected.

trunneml, 2022-10-15:
I'm having the same problem and can confirm that https://github.com/esphome/esphome/pull/3863 works.

kbx81, 2022-10-16:
Great -- this fix has been merged into `dev` so I'll close the issue.

closed 2022-10-16 by kbx81