esphome/issues #3530: BME280 errors when WiFi is unavailable

acronce, 2022-08-28:
### The problem

I'm preparing an ESP32 based sensor as a gift. To that end I've configured it with that person's WiFi credentials. But when I boot the device after changing the credentials the BME280 sensor doesn't initialize (i.e. `has_state()` reads as false.

If I go into the device via captive portal and configure it for my network, the BME280 sensor starts working. It's as if lack of available WiFi or the presence of the device's AP is interfering with the I2C bus.

The other sensors (pmsa003 and senseair) work fine when WiFi is unavailable.

I've tried a few things, like increasing the frequency and changing the `setup_priority` of the I2C bus, but have not been able to work around this issue.

I realize that it's unusual to have an ESP device that's not connected to WiFi, but in this case the unit has its own display and can function without a network. It would be nice if the temperature sensor works when WiFi is unavailable.

### Which version of ESPHome has the issue?

Version: 2022.5.1

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

N/A

### What platform are you using?

ESP32

### Board

LILYGO® TTGO T-Display 1.14 Inch LCD ESP32 (16MB CH9102F Chip)

### Component causing the issue

i2c or WiFi

### Example YAML snippet

```yaml
esphome:
  name: $instance_name
  platform: ESP32
  board: featheresp32
    
# Enable logging
logger:
#  level: VERBOSE

# Enable Home Assistant API
# Disabled for the esphome version because this causes periodic watchdog reboots
#api:

# Over the air update credentials
ota:
  password: "$instance_ota_password"

# WiFi credentials
wifi:
  ssid: $instance_wifi_ssid
  password: $instance_wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "$instance_ssid"
    password: "$instance_wifi_ap_password"

captive_portal:

# WiFi information displayed
text_sensor:
  - platform: wifi_info
    ip_address:
      internal: true
      id: wifi_ip_addr
    ssid:
      internal: true
      id: wifi_ssid

# For temp + humidity sensor
i2c:
  sda: 21
  scl: 22
  scan: True
  id: bus_a

sensor:
  - platform: wifi_signal
    name: $instance_friendly_name RSSI
    update_interval: 60s

  - platform: bme280
    address: 0x76
    update_interval: 60s
    temperature:
      id: temp
      name: "${instance_friendly_name} Temperature"
      accuracy_decimals: 2
      filters:
        - offset: $instance_bme280_temperature_offset
    pressure:
      id: pressure
      name: "${instance_friendly_name} Pressure"
    humidity:
      id: humidity
      name: "${instance_friendly_name} Humidity"
```

### Anything in the logs that might be useful for us?

```txt
I don't know how to get the logs when the device is in captive portal mode, and the problem goes away when the device connects to WiFi. I cannot connect via USB due to the port being blocked in my installation.
```

pxe-la, 2022-11-14:
Have same issue. I checked uart logs:
```
[02:11:40][V][bme280.sensor:203]: Sending conversion request...
[02:11:40][C][bme280.sensor:086]: Setting up BME280...
```
That message repeats twice per second until wifi is connected

acronce, 2023-01-03:
Thanks so much for fixing this @pxe-la!

I cloned your fork and verified that it works. When WiFi is inaccessible, the BME280 sensor initializes and works normally in my project.

Again, thanks for your help.

closed 2023-01-03 by acronce