esphome/issues #3532: Convert Tasmota to ESPHome no enough space

array81, 2022-08-30:
### The problem

I have a Teckin SP22 with Tasmota 12.1.1 installed. Now I'd like install ESPHome on it so I have create my binary firmware by ESPHome Dashboard (using this yaml like base because is the same plug with another name https://www.esphome-devices.com/devices/BlitzWolf-BW-SHP2-Power-Monitoring-Plug).

When I try to upload the new firmware by Tasmota I get this error: "no enough space".
The bin file is 553KB but I have compress it by gzip to 375 KB.

From Tasmota I get these data:
ESP Chip Id	15955710 (ESP8285)
Flash Chip Id	0x144051
Flash Size	1024 KB
Program Flash Size	1024 KB
Program Size	624 KB
Free Program Space	376 KB
Free Memory	26.6 KB

so where is the problem?

### Which version of ESPHome has the issue?

2022.8.0

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2022.8

### What platform are you using?

ESP8266

### Board

ESP8285

### Example YAML snippet

```yaml
substitutions:
  devicename: presa-sp22-a
  devicename_underscore: presa_sp22_a
  friendly_name: Teckin SP22 A
  device_description: Teckin SP22 Power Monitoring Plug with button, Red and Blue led.
  current_res: "0.00221" # Random value. Requires power monitoring calibration
  voltage_div: "955" # Random value. Requires power monitoring calibration

esphome:
  name: ${devicename}
  comment: ${device_description}
  platform: ESP8266
  board: esp8285

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "XXX"

ota:
  password: "XXX"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Presa-Sp22-A Fallback Hotspot"
    password: "eaiNgJey7iCr"

captive_portal:

# Web server can be removed after enabling HA API
web_server:
  port: 80

# Enable time component for use by daily power sensor
time:
  - platform: homeassistant
    id: homeassistant_time

sensor:
# Reports how long the device has been powered (in minutes)
  - platform: uptime
    name: ${friendly_name} Uptime
    filters:
      - lambda: return x / 60.0;
    unit_of_measurement: minutes

# Reports the WiFi signal strength
  - platform: wifi_signal
    name: ${friendly_name} Wifi Signal
    update_interval: 60s

# Reports the Current, Voltage, and Power used by the plugged-in device
  - platform: hlw8012
    sel_pin:
      number: GPIO12
      inverted: true
    cf_pin: GPIO05
    cf1_pin: GPIO14
    change_mode_every: 8
    update_interval: 10s
    current:
      name: ${friendly_name} Amperage
      icon: mdi:current-ac
      unit_of_measurement: A
    voltage:
      name: ${friendly_name} Voltage
      icon: mdi:flash-circle
      unit_of_measurement: V
    power:
      id: wattage
      name: ${friendly_name} Wattage
      icon: mdi:flash-outline
      unit_of_measurement: W
    current_resistor: ${current_res}
    voltage_divider: ${voltage_div}

  # Reports the total Power so-far each day, resets at midnight, see https://esphome.io/components/sensor/total_daily_energy.html
  - platform: total_daily_energy
    name: ${friendly_name} Total Daily Energy
    icon: mdi:circle-slice-3
    power_id: wattage
    filters:
      # Multiplication factor from W to kW is 0.001
      - multiply: 0.001
    unit_of_measurement: kWh

binary_sensor:
# Button on the front is pressed and then toggle relay
  - platform: gpio
    device_class: power
    pin:
      number: GPIO13
      mode: INPUT_PULLUP
      inverted: True
    name: ${friendly_name} Button # Name to make button visible in HA
    on_press:
      - switch.toggle: relay

text_sensor:
# Reports the ESPHome Version with compile date
  - platform: version
    name: ${friendly_name} ESPHome Version

# Reports detailed wifi info, can be commented out
  - platform: wifi_info
    ip_address:
      name: ${friendly_name} IP Address
    # ssid: # Some additional wifi info that is not normally needed
    #   name: ${friendly_name} Connected SSID
    # bssid:
    #   name: ${friendly_name} Connected BSSID

switch:
# Relay itself
  - platform: gpio
    name: ${friendly_name}
    pin: GPIO15
    id: relay
    restore_mode: RESTORE_DEFAULT_OFF #Try to restore relay state after reboot/power-loss event.
    #RESTORE_DEFAULT_OFF (Default) - Attempt to restore state and default to OFF if not possible to restore. Uses flash write cycles.
    #RESTORE_DEFAULT_ON - Attempt to restore state and default to ON. Uses flash write cycles.
    #ALWAYS_OFF - Always initialize the pin as OFF on bootup. Does not use flash write cycles.
    #ALWAYS_ON - Always initialize the pin as ON on bootup. Does not use flash write cycles.
    on_turn_on: # Action when relay is turned on
      - script.execute: led_power_on
    on_turn_off: # Action when relay is turned off
      - script.execute: led_relay_off

output:
# Output GPIOs for blue led
- platform: esp8266_pwm # Blue
  id: ${devicename_underscore}_blue_output
  pin:
    number: GPIO00
    inverted: True

status_led:
  pin:
    number: GPIO02
    inverted: True

light:
# RGB light
- platform: monochromatic
  name: ${friendly_name} Light
  output: ${devicename_underscore}_blue_output
  id: ${devicename_underscore}_blue_led

# Blink the blue light if we aren't connected to WiFi. Could use https://esphome.io/components/status_led.html instead but then we couldn't use the blue light for other things as well.
# Only if not connected after 30s
interval:
  - interval: 2000ms
    then:
      - if:
          condition:
            for:
              time: 30s
              condition:
                not:
                  wifi.connected:
          then:
            - light.turn_on:
                id: ${devicename_underscore}_blue_led
                brightness: 100%

script:
  - id: led_relay_off # Normal operation when relay is off
    then:
      - light.turn_off:
          id: ${devicename_underscore}_blue_led
  - id: led_power_on # Normal operation when relay is on
    then:
      - light.turn_on:
          id: ${devicename_underscore}_blue_led
          brightness: 80%
```

github-k8n, 2022-08-31:
Flash tasmota minimal first.
http://ota.tasmota.com/tasmota/release/
"[tasmota-minimal.bin.gz](http://ota.tasmota.com/tasmota/release/tasmota-minimal.bin.gz)"
(download, unzip and upload to tasmota device)
After reboot you can then flash esphome.

nagyrobi, 2022-09-01:
also you can gzip manually the esphome binary too, and upload it like that.

array81, 2022-09-01:
Using GZIP I have the same problem but I solved by Tasmota minimal. Thanks.

closed 2022-09-01 by nagyrobi