esphome/issues #3533: Esphome in docker broke

mabushey, 2022-08-30:
### The problem

I get a brain dead message of:
```
 ESPHome can install chickencoop.yaml on your device via the browser if certain requirements are met: 
 
 ESPHome is visited over HTTPS
 ```
 
 No. No. No. I am not installing a reverse proxy to appease some __bleep__. WTF???? Are you trying to actively run off people from this project?  
 How do I disable this incorrect design requirement (It's MY network, not anyone else's)? Why is this buggy config the default? This >>is<< a serious bug.

### Which version of ESPHome has the issue?

esphome/esphome                        latest                    58b7d9b27aae   13 days ago     264MB

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

esphome, NOT "home assistant"

### What platform are you using?

ESP32

### Board

LILYGO T-Relay

### Component causing the issue

esphome

nielsnl68, 2022-08-30:
This has nottng todo with esphome. This is your browser requere to have to allow using the serial port from within your browser,
When you dont want to use this future you can always open https://web.esphome.io or use an other way to upload your firmware via the serial port.

mabushey, 2022-08-31:
Thanks for the clarification, and thanks for taking the time to let me know. :)

closed 2022-08-31 by mabushey