esphome/issues #3534: Vulnerability report

rushbnt, 2022-08-31:
### The problem

ESPHome can be takeover via leak the api password on LAN network

### Which version of ESPHome has the issue?

all version

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

all version

### What platform are you using?

ESP8266

### Board

nodemcu

### Component causing the issue

Native api

### Additional information

https://studio.youtube.com/video/e_rueSw13ow/edit

ssieb, 2022-08-31:
I don't know what your point is with this.  You flashed a modified firmware that prints the password for you.  This now gives you access to add the device to HA and switch the lights.  There's no "takeover" involved.  The password is probably sent in the clear over the network as well, which would have been much easier to capture.  There is an encryption option for the `api:` that will avoid all of this if you're really concerned.

rushbnt, 2022-08-31:
Oh, sorry, I did not check the doc first. I do this video in last year. And I discovered this bug in 2020. But no response in email. I just found the way and try report again before I write a blog about that.

closed 2022-08-31 by rushbnt