esphome/issues #3536: ESP32s3 neopixel

kj4dxk, 2022-08-31:
### The problem

This is what I have in ESPhome-DEV and it works without the light but with the light I get compiler errors no matter which method I choose if I use - platform: neopixelbus or if I use - platform: fastled_clockless.

Is there a certain configuration that I can use?

### Which version of ESPHome has the issue?

2022.8 dev

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.8

### What platform are you using?

ESP32

### Board

adafruit_qtpy_esp32s3_nopsram

### Component causing the issue

neopixel, fastled_clockless

### Example YAML snippet

```yaml
esp32:
board: adafruit_qtpy_esp32s3_nopsram
variant: esp32s3
framework:
type: arduino
version: 2.0.4
platform_version: 5.1.1

light:

platform: fastled_clockless
chipset: WS2812 #WS2812B
pin: GPIO39 # NEOPIXEL qt py esp32-s3
num_leds: 1
rgb_order: BRG
name: "FastLED WS2811 Light"
```

### Anything in the logs that might be useful for us?

```txt
no
```

### Additional information

no

ssieb, 2022-09-01:
What are the compiler errors?

kj4dxk, 2022-09-01:
When using fastled_clockless I see a dozen of each of these:
#########################################
In file included from /data/esthree/.piolibdeps/esthree/FastLED/platforms/esp/32/clockless_rmt_esp32.h:95,
                 from /data/esthree/.piolibdeps/esthree/FastLED/platforms/esp/32/fastled_esp32.h:8,
                 from /data/esthree/.piolibdeps/esthree/FastLED/platforms.h:36,
                 from /data/esthree/.piolibdeps/esthree/FastLED/FastLED.h:52,
                 from /data/esthree/.piolibdeps/esthree/FastLED/FastLED.cpp:2:
/data/cache/platformio/packages/framework-arduinoespressif32/tools/sdk/esp32s3/include/esp_hw_support/include/esp_intr.h:8:2: warning: #warning esp_intr.h is deprecated, please include esp_intr_alloc.h instead [-Wcpp]
 #warning esp_intr.h is deprecated, please include esp_intr_alloc.h instead

#########################################
In file included from /data/esthree/.piolibdeps/esthree/FastLED/platforms/esp/32/fastled_esp32.h:8,
                 from /data/esthree/.piolibdeps/esthree/FastLED/platforms.h:36,
                 from /data/esthree/.piolibdeps/esthree/FastLED/FastLED.h:52,
                 from /data/esthree/.piolibdeps/esthree/FastLED/FastLED.cpp:2:
/data/esthree/.piolibdeps/esthree/FastLED/platforms/esp/32/clockless_rmt_esp32.h: In member function 'void ClocklessController<DATA_PIN, T1, T2, T3, RGB_ORDER, XTRA0, FLIP, WAIT_TIME>::startOnChannel(int)':
/data/esthree/.piolibdeps/esthree/FastLED/platforms/esp/32/clockless_rmt_esp32.h:399:52: warning: 'esp_err_t rmt_set_pin(rmt_channel_t, rmt_mode_t, gpio_num_t)' is deprecated: use rmt_set_gpio instead [-Wdeprecated-declarations]
         rmt_set_pin(mRMT_channel, RMT_MODE_TX, mPin);
#########################################
In file included from /data/esthree/.piolibdeps/esthree/FastLED/platforms/esp/32/fastled_esp32.h:8,
                 from /data/esthree/.piolibdeps/esthree/FastLED/platforms.h:36,
                 from /data/esthree/.piolibdeps/esthree/FastLED/FastLED.h:52,
                 from /data/esthree/.piolibdeps/esthree/FastLED/power_mgt.cpp:2:
/data/esthree/.piolibdeps/esthree/FastLED/platforms/esp/32/clockless_rmt_esp32.h: In member function 'void ClocklessController<DATA_PIN, T1, T2, T3, RGB_ORDER, XTRA0, FLIP, WAIT_TIME>::startOnChannel(int)':
/data/esthree/.piolibdeps/esthree/FastLED/platforms/esp/32/clockless_rmt_esp32.h:399:52: warning: 'esp_err_t rmt_set_pin(rmt_channel_t, rmt_mode_t, gpio_num_t)' is deprecated: use rmt_set_gpio instead [-Wdeprecated-declarations]
         rmt_set_pin(mRMT_channel, RMT_MODE_TX, mPin);
                                                    ^
In file included from /data/esthree/.piolibdeps/esthree/FastLED/platforms/esp/32/clockless_rmt_esp32.h:97,
                 from /data/esthree/.piolibdeps/esthree/FastLED/platforms/esp/32/fastled_esp32.h:8,
                 from /data/esthree/.piolibdeps/esthree/FastLED/platforms.h:36,
                 from /data/esthree/.piolibdeps/esthree/FastLED/FastLED.h:52,
                 from /data/esthree/.piolibdeps/esthree/FastLED/power_mgt.cpp:2:
/data/cache/platformio/packages/framework-arduinoespressif32/tools/sdk/esp32s3/include/driver/include/driver/rmt.h:930:11: note: declared here
 esp_err_t rmt_set_pin(rmt_channel_t channel, rmt_mode_t mode, gpio_num_t gpio_num)
           ^~~~~~~~~~~

kj4dxk, 2022-09-01:
Regarding light:  - platform: neopixelbus there are these three IDE errors:
bit_bang only supports gpiio0..31 on ESP32
ESP32s3 doe not support the rmt method (type: esp32_rmt)
ESP32s3 does not support the i2s method (type: esp32_i2s)

ssieb, 2022-09-01:
Those compiler warnings are from the library, there's nothing esphome can do about that.
I'm not sure what your point is about the "IDE errors" and what does that mean?  Are those messages incorrect?

kj4dxk, 2022-09-01:
IDE=integrated development environment. That is my way of calling the errors that show before clicking "Install", they show red x on the problem line and a tooltip that displays the error.

You must be right that these are underlying library issues.

kj4dxk, 2022-09-01:
You must be right that these are underlying library issues.

closed 2022-09-01 by kj4dxk