esphome/issues #3538: Allow GPIO8 for ESP32-Pico-V3-02

kj4dxk, 2022-09-01:
### The problem

The  ESP32 QT PY PICO has a neopixel power on GPIO8?  Can we allow GPIO8 on ESP32-Pico-V3-02?  I recognize that ESP32 cannot use it, but "ESP32-Pico-V3-02" can.

See also https://github.com/esphome/issues/issues/3417

### Which version of ESPHome has the issue?

ESPHome 2022.8.0 - 17th August 2022

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

Home Assistant 2022.8.7

### What platform are you using?

ESP32

### Board

adafruit_qtpy_esp32, ESP32-Pico-V3-02

### Component causing the issue

platform: gpio

### Example YAML snippet

```yaml
#esp32:
#  board: adafruit_qtpy_esp32
#  variant: esp32
#  framework:
#    type: arduino

esp32:
  board: esp32dev
  framework:
    type: arduino

output:
  - platform: gpio
    id: "neopixel_power"
    pin: GPIO8 # NEOPIXEL_POWER
    inverted: true
```

arthurkomatsu, 2022-09-09:
Confirming here, you can use GPIO8 in `ESP32-Pico-V3-02`, same for GPIO7. Unfortunatelly, if you currently want to use it, you must comment [this](https://github.com/esphome/esphome/blob/dev/esphome/components/esp32/gpio_esp32.py#L16). There is also an alternative if you use Python:
```Python
def esp32_pico_v3_validate_gpio_pin(self, value):
    if value < 0 or value > 39:
        raise KeyError(f"Invalid pin number: `{value}` (must be 0-39)")
    _ESP_SDIO_PINS = {
        6: "Flash Clock",
        11: "Flash Command",
    }
    if value in _ESP_SDIO_PINS:
        raise KeyError(
            f"This pin cannot be used on ESP32s and is already used by the"
            " flash interface (function: {_ESP_SDIO_PINS[value]})"
        )
    if 9 <= value <= 10:
        self.logger.warning(
            "Pin %s (9-10) might already be used by the "
            "flash interface in QUAD IO flash mode.",
            value,
        )
    _ESP32_STRAPPING_PINS = {0, 2, 4, 15}
    if value in (24, 28, 29, 30, 31):
        # These pins are not exposed in GPIO mux (reason unknown)
        # but they're missing from IO_MUX list in datasheet
        raise KeyError(f"The pin GPIO{value} is not usable on ESP32s.")
    return value
    
# Fix for ESP PICO V3 validator
from esphome.components.esp32 import gpio

gpio._esp32_validations[
    "ESP32"
].pin_validation = self.esp32_pico_v3_validate_gpio_pin
```

kj4dxk, 2022-11-11:
> Confirming here, you can use GPIO8 in `ESP32-Pico-V3-02`, same for GPIO7. Unfortunatelly, if you currently want to use it, you must comment [this](https://github.com/esphome/esphome/blob/dev/esphome/components/esp32/gpio_esp32.py#L16). There is also an alternative if you use Python:
> 
> ```python
> from esphome.components.esp32 import gpio_esp32
> gpio_esp32._ESP_SDIO_PINS = {
>     6: "Flash Clock",
>     11: "Flash Command",
> }
> # Do whatever here...
> ```

Here is where to find this to comment it out.  I hope that this can get fixed in the standard unedited versions soon.

open homeassistant
open studio code server add-in
navigate to the following file path:
config
esphome
.esphome
external_components
esphome
components
esp32
gpio_esp32.py
around line 17, comment out the following line: "8: "Flash Data 1","

kj4dxk, 2022-11-11:
> Here is where to find this to comment it out. I hope that this can get fixed in the standard unedited versions soon.
>
> open homeassistant open studio code server add-in navigate to the following file [...]

This did not actually work.  Help.

decayofmind, 2023-07-13:
@kj4dxk 

In `ESP32-Pico-V3-02` for a board like [Adafruit ESP32 Feather V2](https://learn.adafruit.com/assets/112834) - it's pin number 2, not 8

So in config you should have:

```
output:
  - platform: gpio
    id: "neopixel_power"
    pin: 2
    inverted: true
```