esphome/issues #3543: D1 Mini pro : error when update by OTA

SqualeDev, 2022-09-01:
### The problem

I get this error when trying to flash a Wemos D1 Mini Pro by OTA.
`ERROR Error binary size: Error: ESP has been flashed with wrong flash size. Please choose the correct 'board' option (esp01_1m always works) and then flash over USB.`

### Which version of ESPHome has the issue?

ESPHome v2022.8.0

### What type of installation are you using?

Docker

### What platform are you using?

ESP8266

### Board

d1_mini_pro

### Example YAML snippet

```yaml
# ESPHome configuration file for 
# water-tank-level module

substitutions:
  esphome_name: water-tank-level
  logger_level: DEBUG
  logger_baud_rate: '0'
  static_ip: !secret ip_water_tank_level
  
packages:
    base: !include .config-base.yaml
    wifi: !include .config-wifi.yaml
    ota: !include .config-ota.yaml
    logger: !include .config-logger.yaml
    portal: !include .config-portal.yaml
    api: !include .config-api.yaml
    status: !include .config-status.yaml
    web_server: !include .config-web-server.yaml
  
esp8266:
  board: d1_mini_pro

# Module Definition
i2c:
  sda: D2
  scl: D1
  scan: true
  id: bus_vl5310x

sensor:
  - platform: wifi_signal
    name: "WiFi Signal Sensor"
    update_interval: 60s
    
  - platform: vl53l0x
    name: "Tank Water volume"
    i2c_id: bus_vl5310x
    # address: 0x29
    long_range: true
    update_interval: 15s
    unit_of_measurement: "l"
    accuracy_decimals: 1
    
    # Capteur <=> cuve vide   (HC): 1.30 m => 1.01 m
    # Capteur <=> cuve pleine (HP): 0.22 m => 0.15 m
    # Contenance maxi en litre (TC): 900L
    # Valeur du capteur en metre (x)
    # Formule : ((HC - x) * TC) / (HC - HP)
    
    filters:
      - lambda: return (1.01-x)*900/(1.01-.15);
      - filter_out: nan
```

### Anything in the logs that might be useful for us?

```txt
INFO Reading configuration /config/water-tank-level.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing water-tank-level (board: d1_mini_pro; framework: arduino; platform: platformio/espressif8266 @ 3.2.0)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 16MB Flash
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- ESPAsyncTCP-esphome @ 1.2.3
|-- ESPAsyncWebServer-esphome @ 2.1.0
|   |-- ESPAsyncTCP-esphome @ 1.2.3
|   |-- Hash @ 1.0
|   |-- ESP8266WiFi @ 1.0
|-- DNSServer @ 1.1.1
|-- ESP8266WiFi @ 1.0
|-- ESP8266mDNS @ 1.2
|-- Wire @ 1.0
|-- ArduinoJson @ 6.18.5
RAM:   [====      ]  40.8% (used 33392 bytes from 81920 bytes)
Flash: [====      ]  41.7% (used 435429 bytes from 1044464 bytes)
========================= [SUCCESS] Took 2.41 seconds =========================
INFO Successfully compiled program.
INFO Connecting to 192.168.0.144
INFO Uploading /config/.esphome/build/water-tank-level/.pioenvs/water-tank-level/firmware.bin (439584 bytes)
INFO Compressed to 305049 bytes
ERROR Error binary size: Error: ESP has been flashed with wrong flash size. Please choose the correct 'board' option (esp01_1m always works) and then flash over USB.
```

### Additional information

When I upload this yaml script by "Manual download" and esphomeflasher, all is right but when I try to update by OTA I have the error :
`ERROR Error binary size: Error: ESP has been flashed with wrong flash size. Please choose the correct 'board' option (esp01_1m always works) and then flash over USB.`

SqualeDev, 2022-09-02:
If I flash with [esphomeflasher](https://github.com/esphome/esphome-flasher), I can flash once with ESPHome via OTA but after this flashing, impossible to update via OTA.
I have to go through [esphomeflasher](https://github.com/esphome/esphome-flasher) again...
Which setting is overwritten by the ESPHome OTA update?

nielsnl68, 2022-09-02:
> If I flash with [esphomeflasher](https://github.com/esphome/esphome-flasher), I can flash once with ESPHome via OTA but after this flashing, impossible to update via OTA. I have to go through [...]

to allow OTA you need to enable it by adding `ota:` in your script. I see you have a package for the ota, could you show that as well?

SqualeDev, 2022-09-02:
Here is the code for the config-ota.yaml file :
```yaml
  # OTA Update Component:
  #  https://esphome.io/components/ota.html
  ota:
    password: !secret ota_pass_all
```

`what you could try is setting the board: to esp01_1m and try to reflash it via the esphomeflasher and see of the OTA works again.`
I have already done this and it works well but I need to adapt my GPIOs. I'm not sure if the GPIOs match.

SqualeDev, 2022-09-03:
but the notation D1 & D2 does not exist on esp01_1m

closed 2023-01-16 by github-actions[bot]