esphome/issues #3546: Lolin C3 Mini - Unknown board ID

ozonejunkieau, 2022-09-02:
### The problem

I'm trying to get esphome working with a Lolin C3 Mini board, which I believe is supported from https://github.com/esphome/esphome/pull/3587. I have tried both stable and dev releases and am getting the same issue.

I've confirmed the installed version of platformio is 6.0.2 and the lolin_c3_mini entry does exist in the boards.py within the virtual environment.

Any suggestions?

### Which version of ESPHome has the issue?

2022.9.0-dev

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

N/A

### What platform are you using?

ESP32-IDF

### Board

lolin_c3_mini

### Component causing the issue

core

### Example YAML snippet

```yaml
esp32:
  board: lolin_c3_mini
  framework:
    type: esp-idf
    version: latest
```

### Anything in the logs that might be useful for us?

```txt
INFO Generating C++ source...
INFO Compiling app...
Processing xxxx (board: lolin_c3_mini; framework: espidf; platform: platformio/espressif32 @ 3.5.0)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Error: Unknown board ID 'lolin_c3_mini'
```

AndreasBVB, 2022-09-04:
Had the same problem with 2022.8.1. Very confused.
I used "board: esp32-c3-devkitm-1" instead and it worked. Also needs to add WIFI power option [output_power: "8.5"] to prevent boot loop.

dashdrum, 2022-09-04:
I am on esphome 2022.8.0.  Seeing the same problem.

I have tried many combinations of board name, framework, version, platform, etc. that I have found on other posts, but still can't get this board to actually boot to my code.

@AndreasBVB Did you get your board to boot?

My code:
```
substitutions:
  device_name: wall-clock-test
  upper_device_name: Wall Clock Test

esphome:
  name: $device_name

esp32:
  board: esp32-c3-devkitm-1
  # board: lolin-c3-mini
  # variant: esp32c3
  framework:
    type: esp-idf
    version: recommended
    # type: arduino
    # version: 2.0.3
    # platform_version: 4.3.0

wifi: 
  ssid: !secret wifi_iot_ssid
  password: !secret wifi_iot_password
  output_power: "8.5"
  
  ap:
    ssid: $upper_device_name
    password: !secret hotspot_password

logger:
  level: DEBUG

api:

ota:
    
sensor:
  - platform: wifi_signal
    name: $upper_device_name WiFi Signal
    update_interval: 30min
    icon: mdi:wifi

button:
  - platform: restart
    name: $upper_device_name Restart

```

AndreasBVB, 2022-09-04:
Yes doing quite a few things with it now. At the first look I think you are missing this piece:
esphome:
  name: ${device_name}
  **platformio_options:
    board_build.flash_mode: dio**

dashdrum, 2022-09-05:
Thanks for your input.  Unfortunately it still does not boot, and I'm beginning to think the board may be bad.  Using this to flash the chip, and it runs without error:

    esptool.py --chip esp32c3 --port /dev/cu.usbmodem14201 write_flash -z 0x0 ~/Downloads/wall-clock-test.bin

AndreasBVB, 2022-09-05:
Try getring a USB wired log out of it. Lolin boards are usually robust unless fried by high voltage.

closed 2023-01-18 by github-actions[bot]