esphome/issues #3547: Xiaomi Body Composition Scale 2 integration doesn't see impedance

FredFreegroup, 2022-09-03:
### The problem

It's reads only weight, no impedance. And sometimes it's doesn't read anything at all....

### Which version of ESPHome has the issue?

2022.8.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.8.6

### What platform are you using?

ESP32

### Board

esp32-wrover-e

### Component causing the issue

miScale

### Example YAML snippet

```yaml
- platform: xiaomi_miscale
    mac_address: 'C7:DD:15:XX:XX:XX'
    weight:
      name: "Xiaomi Mi Scale Weight"
    impedance:
      name: "Xiaomi Mi Scale Impedance"
```

### Anything in the logs that might be useful for us?

```txt
here log read only weight:
[13:35:33][D][esp32_ble_tracker:731]: Found device C7:DD:15:xxxxxx RSSI=-82
[13:35:33][D][esp32_ble_tracker:752]:   Address Type: RANDOM
[13:35:33][D][esp32_ble_tracker:754]:   Name: 'MIBCS'
[13:35:35][D][esp32_ble_tracker:731]: Found device F4:D9:A9:xxxxxx RSSI=-78
[13:35:35][D][esp32_ble_tracker:752]:   Address Type: RANDOM
[13:35:35][D][esp32_ble_tracker:754]:   Name: 'fenix 5 APAC'
[13:35:42][D][xiaomi_miscale:154]: Got Xiaomi Miscale v2 (C7:DD:15:xxxxxx):
[13:35:42][D][xiaomi_miscale:157]:   Weight: 81.00kg
[13:35:42][D][sensor:127]: 'Xiaomi Mi Scale Weight': Sending state 81.00000 kg with 2 decimals of accuracy
[13:35:42][D][esp32_ble_tracker:731]: Found device 49:B9:37xxxxx RSSI=-94

here is read nothing:
[14:08:56][D][esp32_ble_tracker:731]: Found device C7:DD:15:xxxxxx RSSI=-93
[14:08:56][D][esp32_ble_tracker:752]:   Address Type: RANDOM
[14:08:56][D][esp32_ble_tracker:754]:   Name: 'MIBCS'
[14:09:28][D][dht:048]: Got Temperature=23.0°C Humidity=49.0%
[14:09:28][D][sensor:127]: 'Room1 Temperature': Sending state 23.00000 °C with 1 decimals of accuracy
[14:09:28][D][sensor:127]: 'Room1 Humidity': Sending state 49.00000 % with 0 decimals of accuracy
[14:09:44][D][dallas.sensor:143]: 'Fishtank Temperature': Got Temperature=25.1°C
[14:09:44][D][sensor:127]: 'Fishtank Temperature': Sending state 25.06250 °C with 2 decimals of accuracy
```

### Additional information

This is log of the same ESP32 in same time, programed by 

github.com/RobertWojtowicz/miscale2garmin

BLE device found with address: c7:dd:15:xxxxxx <= target device
* Reading BLE data complete, finished BLE scan
* Connecting to WiFi: already connected
connected
  IP address: 192.168.31.252
* Connecting to MQTT: connected
* Publishing MQTT data: 81.00;326;1662197752;to_import;2022-9-3 13:35:52;;
* Waiting for next scan, going to sleep

Impedance - 326

FredFreegroup, 2022-09-03:
When I compiled my ESP32 first time, i`ve got error:

/data/cache/platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-spi.c: In function 'spiTransferBytesNL':
/data/cache/platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-spi.c:922:39: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
                 uint8_t * last_out8 = &result[c_longs-1];
                                       ^
/data/cache/platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-spi.c:923:40: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
                 uint8_t * last_data8 = &last_data;
                                        ^
Compiling /data/esphome-web-5e43a4/.pioenvs/esphome-web-5e43a4/FrameworkArduino/esp32-hal-time.c.o

Can this error affect on this sensor?

FredFreegroup, 2022-09-03:
some times it work well

15:02:17 | [D] | [esp32_ble_tracker:731] | Found device C7:DD:15:xxxxxx RSSI=-74
15:02:17 | [D] | [esp32_ble_tracker:752] | Address Type: RANDOM
15:02:17 | [D] | [esp32_ble_tracker:754] | Name: 'MIBCS'
15:02:43 | [D] | [dht:048] | Got Temperature=23.1°C Humidity=49.0%
15:02:43 | [D] | [sensor:127] | 'Room1 Temperature': Sending state 23.10000 °C with 1 decimals of accuracy
15:02:43 | [D] | [sensor:127] | 'Room1 Humidity': Sending state 49.00000 % with 0 decimals of accuracy
15:02:58 | [D] | [dallas.sensor:143] | 'Fishtank Temperature': Got Temperature=25.0°C
15:02:58 | [D] | [sensor:127] | 'Fishtank Temperature': Sending state 25.00000 °C with 2 decimals of accuracy
15:03:43 | [D] | [dht:048] | Got Temperature=23.4°C Humidity=49.0%
15:03:43 | [D] | [sensor:127] | 'Room1 Temperature': Sending state 23.40000 °C with 1 decimals of accuracy
15:03:43 | [D] | [sensor:127] | 'Room1 Humidity': Sending state 49.00000 % with 0 decimals of accuracy
15:03:44 | [D] | [xiaomi_miscale:154] | Got Xiaomi Miscale v2 (C7:DD:15:xxxxxx):
15:03:44 | [D] | [xiaomi_miscale:157] | Weight: 81.50kg
15:03:44 | [D] | [xiaomi_miscale:160] | Impedance: 330ohm
15:03:44 | [D] | [sensor:127] | 'Xiaomi Mi Scale Weight': Sending state 81.50000 kg with 2 decimals of accuracy
15:03:44 | [D] | [sensor:127] | 'Xiaomi Mi Scale Impedance': Sending state 330.00000 Ω with 0 decimals of accuracy
15:03:58 | [D] | [dallas.sensor:143] | 'Fishtank Temperature': Got Temperature=24.9°C
15:03:58 | [D] | [sensor:127] | 'Fishtank Temperature': Sending state 24.93750 °C with 2 decimals of accuracy

<br class="Apple-interchange-newline">
</body>
</html>

vladmovchan, 2022-09-13:
I found there are two possible workarounds which seem to improve reliability.
One is to step off the scales a few seconds later than I used to do.
And another is to reduce interval between BLE scan windows and/or increase duration during which ESP is actively listening for packets:
```
esp32_ble_tracker:
  scan_parameters:
    # default is 320ms:
    interval: 160ms
    # default is 30ms:
    window: 60ms
```

fliespl, 2022-10-01:
I am noticing same behavior. Not all Wright events are being seen.

jeroenterheerdt, 2023-01-02:
have this as well - the weight works fine, but the impedance is never received. the app works fine. I will try the suggestions by @vladmovchan above - any other solutions?