esphome/issues #3551: Problem with Limits.h

Manhem, 2022-09-04:
### The problem

Hi.
I have created a new Home Assistant that also included ESPhome.
Now with the new machine I should update a new ESPhome device with that same code I have used before, I get this:
/data/cache/platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/limits.h:132:26: error: no include path in which to search for limits.h

This is the first time I program the device so this is how I do it:
I choose Manual download and Legacy format.
I want the file on an other computer.

What I have found if I looked into the Home Assistant computer (it's an ordinary PC machine) hassio/addons/data/5c53de3b_esphome (I don't know if this is the right place) I can find /cache/ but if I looked at the path from the error text it should be /data/cache/.

I can also say that I'm a beginner in both Linux, Home Assistant and ESP home.
Played around a bit and have one system that are working but.....

What am I doing wrong??

### Which version of ESPHome has the issue?

ESPhome 2022.8.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

Home Assistant 2022.8.7

### What platform are you using?

ESP8266

### Board

Wemos D1, D1 Mini

### Component causing the issue

Compiling and sending the file to an other computer

### Example YAML snippet

```yaml
esphome:
  name: powermeter-1

esp8266:
  board: d1_mini

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "RdyrRFw4qCyPbO0GDKNZDwebhTzmlMP81VAoMFshdl4="

ota:
  password: "55577be0535c849e7e49165514c5bdc0"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Powermeter-1 Fallback Hotspot"
    password: "jIpb4yVz3Urp"

captive_portal:
```

### Anything in the logs that might be useful for us?

```txt
INFO Reading configuration /config/esphome/powermeter-1.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing powermeter-1 (board: d1_mini; framework: arduino; platform: platformio/espressif8266 @ 3.2.0)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- ESPAsyncTCP-esphome @ 1.2.3
|-- ESPAsyncWebServer-esphome @ 2.1.0
|   |-- ESPAsyncTCP-esphome @ 1.2.3
|   |-- Hash @ 1.0
|   |-- ESP8266WiFi @ 1.0
|-- DNSServer @ 1.1.1
|-- ESP8266WiFi @ 1.0
|-- ESP8266mDNS @ 1.2
|-- noise-c @ 0.1.4
|   |-- libsodium @ 1.10018.1
Compiling /data/powermeter-1/.pioenvs/powermeter-1/src/esphome/components/api/api_connection.cpp.o
Compiling /data/powermeter-1/.pioenvs/powermeter-1/src/esphome/components/api/api_frame_helper.cpp.o
Compiling /data/powermeter-1/.pioenvs/powermeter-1/src/esphome/components/api/api_server.cpp.o
Compiling /data/powermeter-1/.pioenvs/powermeter-1/src/esphome/components/api/list_entities.cpp.o
In file included from /data/cache/platformio/packages/framework-arduinoespressif8266/tools/sdk/lwip2/include/lwip/arch.h:185,
                 from /data/cache/platformio/packages/framework-arduinoespressif8266/tools/sdk/lwip2/include/lwip/debug.h:40,
                 from /data/cache/platformio/packages/framework-arduinoespressif8266/tools/sdk/lwip2/include/lwipopts.h:3632,
                 from /data/cache/platformio/packages/framework-arduinoespressif8266/tools/sdk/lwip2/include/lwip/opt.h:51,
                 from /data/cache/platformio/packages/framework-arduinoespressif8266/tools/sdk/lwip2/include/lwip/inet.h:43,
                 from src/esphome/components/socket/headers.h:10,
                 from src/esphome/components/socket/socket.h:5,
                 from src/esphome/components/api/api_frame_helper.h:13,
                 from src/esphome/components/api/api_frame_helper.cpp:1:
/data/cache/platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/limits.h:132:26: error: no include path in which to search for limits.h
  132 | # include_next <limits.h>
      |                          ^
In file included from /data/cache/platformio/packages/framework-arduinoespressif8266/tools/sdk/lwip2/include/lwip/arch.h:185,
                 from /data/cache/platformio/packages/framework-arduinoespressif8266/tools/sdk/lwip2/include/lwip/debug.h:40,
                 from /data/cache/platformio/packages/framework-arduinoespressif8266/tools/sdk/lwip2/include/lwipopts.h:3632,
                 from /data/cache/platformio/packages/framework-arduinoespressif8266/tools/sdk/lwip2/include/lwip/opt.h:51,
                 from /data/cache/platformio/packages/framework-arduinoespressif8266/tools/sdk/lwip2/include/lwip/inet.h:43,
                 from src/esphome/components/socket/headers.h:10,
                 from src/esphome/components/socket/socket.h:5,
                 from src/esphome/components/api/api_server.h:7,
                 from src/esphome/components/api/api_connection.h:7,
                 from src/esphome/components/api/api_connection.cpp:1:
/data/cache/platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/limits.h:132:26: error: no include path in which to search for limits.h
  132 | # include_next <limits.h>
      |                          ^
In file included from /data/cache/platformio/packages/framework-arduinoespressif8266/tools/sdk/lwip2/include/lwip/arch.h:185,
                 from /data/cache/platformio/packages/framework-arduinoespressif8266/tools/sdk/lwip2/include/lwip/debug.h:40,
                 from /data/cache/platformio/packages/framework-arduinoespressif8266/tools/sdk/lwip2/include/lwipopts.h:3632,
                 from /data/cache/platformio/packages/framework-arduinoespressif8266/tools/sdk/lwip2/include/lwip/opt.h:51,
                 from /data/cache/platformio/packages/framework-arduinoespressif8266/tools/sdk/lwip2/include/lwip/inet.h:43,
                 from src/esphome/components/socket/headers.h:10,
                 from src/esphome/components/socket/socket.h:5,
                 from src/esphome/components/api/api_server.h:7,
                 from src/esphome/components/api/api_connection.h:7,
                 from src/esphome/components/api/list_entities.cpp:5:
/data/cache/platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/limits.h:132:26: error: no include path in which to search for limits.h
  132 | # include_next <limits.h>
      |                          ^
In file included from /data/cache/platformio/packages/framework-arduinoespressif8266/tools/sdk/lwip2/include/lwip/arch.h:185,
                 from /data/cache/platformio/packages/framework-arduinoespressif8266/tools/sdk/lwip2/include/lwip/debug.h:40,
                 from /data/cache/platformio/packages/framework-arduinoespressif8266/tools/sdk/lwip2/include/lwipopts.h:3632,
                 from /data/cache/platformio/packages/framework-arduinoespressif8266/tools/sdk/lwip2/include/lwip/opt.h:51,
                 from /data/cache/platformio/packages/framework-arduinoespressif8266/tools/sdk/lwip2/include/lwip/inet.h:43,
                 from src/esphome/components/socket/headers.h:10,
                 from src/esphome/components/socket/socket.h:5,
                 from src/esphome/components/api/api_server.h:7,
                 from src/esphome/components/api/api_server.cpp:1:
/data/cache/platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/limits.h:132:26: error: no include path in which to search for limits.h
  132 | # include_next <limits.h>
      |                          ^
*** [/data/powermeter-1/.pioenvs/powermeter-1/src/esphome/components/api/api_frame_helper.cpp.o] Error 1
*** [/data/powermeter-1/.pioenvs/powermeter-1/src/esphome/components/api/list_entities.cpp.o] Error 1
*** [/data/powermeter-1/.pioenvs/powermeter-1/src/esphome/components/api/api_connection.cpp.o] Error 1
*** [/data/powermeter-1/.pioenvs/powermeter-1/src/esphome/components/api/api_server.cpp.o] Error 1
========================= [FAILED] Took 18.38 seconds =========================
```

ssieb, 2022-09-16:
Try removing the esphome addon and install it again.

Manhem, 2022-10-11:
Hi.
That doesn't help, so I reinstall the operationsystem, Home Assistant and ESPhome.
And after that everthing is working.

//Bosse

closed 2022-10-12 by ssieb