esphome/issues #3552: Esphome uses IPv6to4

lerra, 2022-09-04:
### The problem

Eesphome shows a mapped ipv4 address in the logs when there is no ipv6 setup in the network or on the device
[12:24:32][D][api:102]: Accepted ::FFFF:A01:104

### Which version of ESPHome has the issue?

2022.8.1

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2022.8.7

### What platform are you using?

ESP32

### Board

esp32-wroom-32

### Anything in the logs that might be useful for us?

```txt
The network the Esphome device resides in does not have any ipv6 setup, home assistant have dual stack,

I tried to disable the ipv6 support in the esphome yaml under network but got the following,
Failed config

network: [source X.yaml:17]
  
  This feature is only available with frameworks ['esp-idf'].
  enable_ipv6: False
```

lerra, 2022-09-04:
I also noticed that esphome is sending arp request and asking for it self?
14:25:51.044873 ARP, Request who-has 10.x.x.13 tell 10.x.x.13, length 42

lerra, 2022-09-28:
Same issue exists in version 2022.9.1

ssieb, 2022-09-29:
> I also noticed that esphome is sending arp request and asking for it self? 14:25:51.044873 ARP, Request who-has 10.x.x.13 tell 10.x.x.13, length 42

This is typically used to detect IP address conflicts.  "Does anyone else have my IP address?"

HeMan, 2023-11-01:
Was this not solved by https://github.com/esphome/esphome/pull/4382?

HeMan, 2023-11-21:
Closing. Please re-open if the problem still occurs.

closed 2023-11-21 by HeMan