esphome/issues #3554: Lolin C3 Mini MQTT log not turning off

AndreasBVB, 2022-09-04:
### The problem

Marked both in code:

logger:
  level: NONE

mqtt:
  log_topic:
    topic: ${device_name}/log
    level: NONE

MQTT still spinning out logs like crazy.

### Which version of ESPHome has the issue?

2022.8.1

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2022.3.5

### What platform are you using?

ESP32-IDF

### Board

Lolin C3 Mini

### Component causing the issue

mqtt

AndreasBVB, 2022-09-04:
Log example:

Sep 04 19:58:14 lc3m/log [][esp-idf:000]: I (8951) gpio: GPIO[4]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
Sep 04 19:58:14 
Sep 04 19:58:14 lc3m/log [][esp-idf:000]: I (8959) gpio: GPIO[4]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
Sep 04 19:58:14 
Sep 04 19:58:14 lc3m/log [][esp-idf:000]: I (9047) gpio: GPIO[4]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
Sep 04 19:58:14 
Sep 04 19:58:14 lc3m/log [][esp-idf:000]: I (9049) gpio: GPIO[4]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
Sep 04 19:58:14 
Sep 04 19:58:14 lc3m/log [][esp-idf:000]: I (9052) gpio: GPIO[4]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
Sep 04 19:58:14 
Sep 04 19:58:14 lc3m/log [][esp-idf:000]: I (9059) gpio: GPIO[4]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0

AndreasBVB, 2022-09-10:
Originally posted as MQTT log not turning off. Recently realized it is the ADC log: even through API the log is always on for the ADC pins not just MQTT.

ssieb, 2022-09-15:
Just a note that that's not esphome logs, it's coming from the idf, but esphome is passing them on.

AndreasBVB, 2022-09-15:
Is this idf log unique to C3? I don't see this kind of crazy log spinning on other boards when logs are off.

closed 2023-01-29 by github-actions[bot]