esphome/issues #3555: Lolin S2 Mini reboots at wake-up

AndreasBVB, 2022-09-04:
### The problem

Every wake-up, by timer or by GPIO, leads to a reboot, confirmed by both wake up cause and uptime reports. Happens when on both USB and battery power.

### Which version of ESPHome has the issue?

2022.8.1

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2022.3.5

### What platform are you using?

ESP32-IDF

### Board

Lolin S2 Mini

### Component causing the issue

deep_sleep

### Example YAML snippet

```yaml
substitutions:
  device_upper: LOLIN S2 MINI
  device_lower: ls2m
  device_name: ls2m

esphome:
  name: ls2m #${device_name}
  on_boot:
    priority: 800
    then:
      lambda: |-
        id(wake_up_reason) = esp_sleep_get_wakeup_cause();

esp32:
  board: esp32-s2-saola-1
  framework:
    type: esp-idf

# Enable logging
logger:
  level: NONE

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  power_save_mode: HIGH
  fast_connect: true

deep_sleep:
  run_duration: 5s
  sleep_duration: 3595s
  touch_wakeup: false
  esp32_ext1_wakeup:
    pins:
      - 9
#      - 7
#      - 11
    mode: ANY_HIGH
  id: sleep0

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO9
      mode:
        input: true
        pullup: false
        pulldown: false #externally attached
    filters:
      - delayed_off: 1s
    name: ${device_upper} GPIO9
    id: ${device_lower}_gpio9
  - platform: template
    lambda: |-
      if ( !(id(${device_lower}_gpio9).state)) {
        return false;
      } else {
        return true;
      }
    name: ${device_upper}
    id: ${device_lower}
    on_press:
      then:
        - deep_sleep.prevent: sleep0
    on_release:
      then:
        - deep_sleep.allow: sleep0

globals:
  - id: wake_up_reason
    type: int
    restore_value: no
    initial_value: '0'
    
sensor:
  - platform: uptime
    update_interval: 1s
    name: ${device_upper} Uptime
    id: ${device_lower}_uptime
  - platform: template
    name: "Wake-up Reason"
    accuracy_decimals: 0
    lambda: |-
      return (id(wake_up_reason));
  - platform: adc
    pin: GPIO1
    attenuation: auto
    name: ${device_upper} ADC
    id: ${device_lower}_adc
    update_interval: 0.1s
    filters:
      - multiply: 2.36363636363636
      - sliding_window_moving_average:
          window_size: 40
          send_every: 40
          send_first_at: 40

mqtt:
  broker: 10.*.*.*
  username: ******
  password: ******
  discovery: false
  discovery_retain: false
  log_topic: ${device_name}/log
  birth_message:
   topic: /status
   payload: "Online"
   retain: false
  will_message:
   topic: /status
   payload: "Offline"
   retain: false
```

AndreasBVB, 2022-09-05:
Further testing shows

AndreasBVB, 2022-09-05:
Further testing shows I can retrieve the correct wake cause by removing the global variable and calling esp_sleep_get_wakeup_cause() directly. It gets reset unexpectedly somewhere. The up-time sensor counterintuitively falls back to zero on wake-ups, but the device did not reboot as previously suspected.

closed 2022-09-05 by AndreasBVB

ssieb, 2022-09-05:
On an ESP, deep sleep wakeup *is* a reboot.  But as you found, you can get the cause to determine what the reboot was for.