esphome/issues #3556: ESP32-C3-M1

Bradleydof, 2022-09-05:
### The problem

I can not get past the initial flash on the device. It is "Unable to compile the configuration".
I have tried specifying differing platform versions and that got it to compile, however the ESP failed to boot.
I'll include the log from the first issue with compiling as well as the second not booting issue.

I have not used any of the C3 boards before but have two 32s's and about 18 8266's running just fine.

### Which version of ESPHome has the issue?

2022.81

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.08.5

### What platform are you using?

ESP32-IDF

### Board

ESP32-C3-M1 Ai Thinker

### Component causing the issue

Initial flash

### Example YAML snippet

```yaml
This is the 'NO COMPILE' YAML

esphome:
  name: c3-test-2

esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: esp-idf

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "REDACTED"

ota:
  password: "REDACTED"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot in case wifi connection fails
  ap:
    ssid: "REDACTED"
    password: "REDACTED"

*************THIS IS THE ONLY CHANGE TO GET IT TO COMPILE BUT NOT BOOT

esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: esp-idf
    platform_version: 3.4.0
    sdkconfig_options:
      CONFIG_COMPILER_OPTIMIZATION_SIZE: y
```

### Anything in the logs that might be useful for us?

```txt
****BELOW THIS IS THE FIRST PART WHERE IT WILL NOT COMPILE

INFO Reading configuration /config/esphome/c3-test-2.yaml...
INFO Generating C++ source...
INFO Core config or version changed, cleaning build files...
INFO Deleting /data/c3-test-2/.pioenvs
INFO Deleting /data/c3-test-2/.piolibdeps
INFO Compiling app...
Processing c3-test-2 (board: esp32-c3-devkitm-1; framework: espidf; platform: platformio/espressif32 @ 3.5.0)
--------------------------------------------------------------------------------
Library Manager: Installing esphome/noise-c @ 0.1.4
INFO Installing esphome/noise-c @ 0.1.4
Unpacking  [####################################]  100%
Library Manager: noise-c@0.1.4 has been installed!
INFO noise-c@0.1.4 has been installed!
Library Manager: Resolving dependencies...
INFO Resolving dependencies...
Library Manager: Installing esphome/libsodium @ 1.10018.1
INFO Installing esphome/libsodium @ 1.10018.1
Unpacking  [####################################]  100%
Library Manager: libsodium@1.10018.1 has been installed!
INFO libsodium@1.10018.1 has been installed!
HARDWARE: ESP32C3 160MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.40302.0 (4.3.2) 
 - tool-cmake @ 3.16.9 
 - tool-ninja @ 1.10.2 
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch2 
 - toolchain-xtensa-esp32s2 @ 8.4.0+2021r2-patch2
Reading CMake configuration...
-- Found Git: /usr/bin/git (found version "2.30.2") 
-- The C compiler identification is unknown
-- The CXX compiler identification is unknown
-- The ASM compiler identification is unknown
-- Found assembler: /data/cache/platformio/packages/toolchain-riscv32-esp/bin/riscv32-esp-elf-gcc
-- Check for working C compiler: /data/cache/platformio/packages/toolchain-riscv32-esp/bin/riscv32-esp-elf-gcc
-- Check for working C compiler: /data/cache/platformio/packages/toolchain-riscv32-esp/bin/riscv32-esp-elf-gcc -- broken
-- Configuring incomplete, errors occurred!
See also "/data/c3-test-2/.pioenvs/c3-test-2/CMakeFiles/CMakeOutput.log".
See also "/data/c3-test-2/.pioenvs/c3-test-2/CMakeFiles/CMakeError.log".

fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
CMake Error at /data/cache/platformio/packages/tool-cmake/share/cmake-3.16/Modules/CMakeTestCCompiler.cmake:60 (message):
  The C compiler

    "/data/cache/platformio/packages/toolchain-riscv32-esp/bin/riscv32-esp-elf-gcc"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: /data/c3-test-2/.pioenvs/c3-test-2/CMakeFiles/CMakeTmp
    
    Run Build Command(s):/data/cache/platformio/packages/tool-ninja/ninja cmTC_9a07d && [1/2] Building C object CMakeFiles/cmTC_9a07d.dir/testCCompiler.c.obj
    FAILED: CMakeFiles/cmTC_9a07d.dir/testCCompiler.c.obj 
    /data/cache/platformio/packages/toolchain-riscv32-esp/bin/riscv32-esp-elf-gcc   -march=rv32imc -o CMakeFiles/cmTC_9a07d.dir/testCCompiler.c.obj   -c testCCompiler.c
    /bin/sh: 1: /data/cache/platformio/packages/toolchain-riscv32-esp/bin/riscv32-esp-elf-gcc: not found
    ninja: build stopped: subcommand failed.
    
    

  

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  /data/cache/platformio/packages/framework-espidf/tools/cmake/project.cmake:311 (__project)
  CMakeLists.txt:3 (project)

========================= [FAILED] Took 17.29 seconds =========================

****BELOW THIS IS THE SECOND PART WHERE IT WILL NOT BOOT

�ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x40049a42
SPIWP:0xee
mode:QIO, clock div:1
load:0x3fcd6100,len:0x19a0
ets_loader.c 78 
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x40049a42
SPIWP:0xee
mode:QIO, clock div:1
load:0x3fcd6100,len:0x19a0
ets_loader.c 78 
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x10 (RTCWDT_RTC_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:QIO, clock div:1
load:0x3fcd6100,len:0x19a0
ets_loader.c 78 
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x40049a42
SPIWP:0xee
mode:QIO, clock div:1
```

AndreasBVB, 2022-09-05:
Try:
esphome:
name: ${device_name}
platformio_options:
board_build.flash_mode: dio

Bradleydof, 2022-09-06:
I have tried as you asked and on the first attempt that wouldn't compile, well it still failed to compile with the exact same log (as I am sure you expected).
The second one, however, worked like a charm. I would have never thought to try DIO. Sure enough that did it! Thank you SO MUCH AndreasBVB!! I owe you a debt of gratitude!

closed 2022-09-06 by Bradleydof

AndreasBVB, 2022-09-07:
Glad I helped. I recently opened up quite a few Lolin C3 Mini and S2 Mini related issues. Kindly share you insights when possible :)