esphome/issues #3557: "rmt_write_items failed: ESP_FAIL" while sending PROTO ir remote command

mr2c12, 2022-09-05:
### The problem

I've setup a esp32 with an IR receiver and an IR blaster.

I successfully "copied" some commands for a simple led strip. Commands are successfully sent via PROTO, NEC or RAW format.

With the very same setup, when trying to send the below commands:

```
  - platform: template
    name: "Samsung Pronto 1"
    on_press:
      - remote_transmitter.transmit_pronto:
          data: "0000 006D 00AE 0000 0072 015E 0011 0019 0013 003B 0011 0018 0013 0015 0011 0018 0013 0015 000F 001A 000E 0019 0014 0015 0010 003F 0013 0016 0010 0019 0015 003A 0011 0018 0011 0018 0013 003B 0013 003D 0010 003F 0015 003A 0011 0040 0013 0015 000E 001A 0015 0015 0010 0019 0015 0013 0011 0016 0013 0017 0012 0016 000E 001A 0014 0015 000E 001B 0010 0019 0013 0015 0012 0017 0014 0013 0010 0019 0013 0017 000E 0019 0012 0016 0011 0018 0012 0017 0013 0015 000E 001"

  - platform: template
    name: "Samsung Pronto 2"
    on_press:
      - remote_transmitter.transmit_pronto:
          data: "0000 006D 00AE 0000 0072 015E 0012 0017 0014 003A 0012 0017 0011 0017 0014 0015 0010 0018 0013 0016 0012 0016 0013 0016 0013 003B 000E 001B 0014 0016 000F 003F 0013 0016 0011 0016 0013 003B 0010 003F 0015 003B 000F 003F 0013 003C 0012 0016 0011 0018 0014 0015 000E 001B 0014 0013 000E 001A 0015 0017 000F 0017 0011 0018 0013 0016 0010 0019 0012 0016 0013 0016 0012 0016 0011 0016 0012 0018 0013 0016 000E 001A 0013 0015 000F 001A 0012 0018 0011 0016 0011 001"

  - platform: template
    name: "Samsung Pronto 3"
    on_press:
      - remote_transmitter.transmit_pronto:
          data: "0000 006D 00AE 0000 0074 015C 000E 001B 0015 003A 000E 001B 0012 0016 000F 001A 0012 0016 0013 0016 000D 001B 0014 0015 0011 003E 0013 0016 000E 001B 0015 003A 000E 001B 0012 0016 0011 003D 0014 003C 000F 0040 0015 003A 000E 0042 0012 0016 0012 0016 0013 0016 0013 0016 0012 0017 0013 0015 0012 0018 0011 0017 0011 0016 0013 0016 0012 0017 000D 001B 0012 0016 0010 0019 0015 0013 0010 0019 0012 0018 0013 0015 000E 001A 0014 0016 000D 001B 0013 0015 0011 001"
```

I always get a log like this:

> [10:02:31][D][button:013]: ‘Samsung Pronto 1’ Pressed.
> [10:02:31][D][remote.pronto:101]: Send Pronto: frequency=38kHz
> [10:02:31][D][remote.pronto:106]: Send Pronto: intros=348
> [10:02:31][D][remote.pronto:107]: Send Pronto: repeats=0
> [10:02:31][V][esp-idf:000]: E (24262) rmt: rmt_write_items(804): RMT ADDRESS ERR
> [10:02:31][W][remote_transmitter:119]: rmt_write_items failed: ESP_FAIL
> [10:02:31][V][component:199]: Component web_server took a long time for an operation (0.05 s).
> [10:02:31][V][component:200]: Components should block for at most 20-30ms.

### Which version of ESPHome has the issue?

2022.4.0

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

HA indipendent

### What platform are you using?

ESP32

### Board

lolin_d32

### Component causing the issue

remote_transmitter

### Example YAML snippet

```yaml
substitutions:
  devicename: rsirda
  upper_devicename: RSIrda

[... 209 lines cut ...]
15:19:00	[V]	[component:200]	
Components should block for at most 20-30ms.
15:19:00	[D]	[remote.pronto:229]	
Received Pronto: data=0000 006D 0025 0000 0016 0019 0039 0017 0053 0018 0039 00C9 0011 001C 0011 001B 0010 001D 0012 001A 0012 001B 0012 001A 0012 001B 0010 001B 0011 0043 0013 0043 0012 0046 0011 0044 0012 001C 0011 0047 0012 0045 0010 0047 0012 001B 0012 001A 0012 0045 0012 001B 0012 001A 0012 001B 0012 001A 0013 001A 0012 0044 0013 0044 0013 001A 0012 0043 0013 0043 0014 0042 0013 0044 0013 0044 0012 06C3
15:19:03	[D]	[button:013]	
'Samsung Pronto 1' Pressed.
15:19:03	[D]	[remote.pronto:101]	
Send Pronto: frequency=38kHz
15:19:03	[D]	[remote.pronto:106]	
Send Pronto: intros=348
15:19:03	[D]	[remote.pronto:107]	
Send Pronto: repeats=0
15:19:03	[V]	[esp-idf:000]	
E (67662) rmt: rmt_write_items(805): RMT PARAM LEN ERROR
15:19:03	[W]	[remote_transmitter:119]	
rmt_write_items failed: ESP_ERR_INVALID_ARG
```

### Additional information

This is the command as sniffed by the esp32:

> [15:20:38][D][remote.pronto:229]: Received Pronto: data=0000 006D 0074 0000 0073 015F 0010 003C 0010 0016 0010 0016 0010 0016 0010 0016 000F 0017 000F 0017 0010 0016 0010 0016 0010 003C 0010 0016 0010 0016 0010 003C 0010 0017 000F 003C 000F 003D 0010 003C 0010 003C 0010 003C 0010 003C 0010 0016 0010 0016 000F 0017 0010 0016 0010 0016 0010 0016 0010 0016 0010 0016 0010 0016 0010 0016 0010 0016 0010 0016 0010 0016 0010 0016 0010 0016 0010 0016 0010 0016 000F 0017 0010 0016 0010 0016 0010 0016 0010 0016 000F 001

The error seems to be happening only with the samsung climate, while the led strip is working just fine.

ssieb, 2022-09-15:
That's an old esphome version, but that doesn't matter in this case.  The problem is that the pronto code is getting cut off, so it's not valid.  The next release will log a message about that instead of getting the RMT error.  It might also have a fix for the pronto code printing.

closed 2022-09-15 by ssieb

mr2c12, 2022-10-22:
After updating to esphome 2022.10.0, now I got this error (as expected):

> [15:02:13][D][button:013]: 'Samsung Pronto 1' Pressed.
> [15:02:13][D][remote.pronto:101]: Send Pronto: frequency=38kHz
> [15:02:13][D][remote.pronto:106]: Send Pronto: intros=348
> [15:02:13][D][remote.pronto:107]: Send Pronto: repeats=0
> [15:02:14][E][remote.pronto:109]: Inconsistent data, not sending
> [15:02:14][E][remote_transmitter:117]: Empty data

@ssieb why is this issue considered "completed"? Should esphomebe able to send this PRONTO message?

ssieb, 2022-10-23:
No, it's not a valid pronto message.  It's not complete.  I also made another PR that will let you receive the full code, but you'll have to receive it again.  The one you have won't work.

mr2c12, 2023-01-28:
With Esphome:

> Version: 2022.12.8

The error I get now is different:

> 21:39:19	[E]	[remote.pronto:109]	
> Inconsistent data, not sending
> 21:39:19	[E]	[remote_transmitter:117]	
> Empty data

This is the message sniffed:

> [21:23:19][D][remote.pronto:238]: Received Pronto: data=0000 006D 00AE 0000 006E 0161 0012 0017 0012 003C 0012 0017 0011 0016 0012 0018 0013 0014 000F 001A 000F 0019 0014 0015 0012 003C 0014 0016 000E 001B 0014 003A 0013 0017 0012 0016 0011 003D 0014 003B 0015 003C 0012 003C 0012 003E
> [21:23:19][D][remote.pronto:240]: 0013 0014 0010 0018 0013 0016 000D 001C 0015 0013 000E 001A 0013 0017 0011 0017 0010 0018 0015 0016 000D 001B 0010 0019 000F 0019 0014 0015 0011 0016 0010 001A 0011 0018 000E 001A 0012 0016 000F 001A 0011 0019 0012 0016 000F 0018 0015 0014 0012 0017 0011 0018 000E 001A 0013 0015 000E 001B 0013 0015 0010 0019 0014 0015 0010 003F 0013 003C 0012 003D 0012 003C 0011 004D 0072 015B 0016 003D 0010 0018 0015 0013 000F 001B 0011 0017 0011 0016 0013 0016 0014 0013 0011 0018 0012 003E 

This is the relevant yaml:

>   - platform: template
>     name: "Samsung HVAC - cool_26"
>     on_press:
>       - remote_transmitter.transmit_pronto:
>           data: "0000 006D 00AE 0000 006E 0161 0012 0017 0012 003C 0012 0017 0011 0016 0012 0018 0013 0014 000F 001A 000F 0019 0014 0015 0012 003C 0014 0016 000E 001B 0014 003A 0013 0017 0012 0016 0011 003D 0014 003B 0015 003C 0012 003C 0012 003E 0013 0014 0010 0018 0013 0016 000D 001C 0015 0013 000E 001A 0013 0017 0011 0017 0010 0018 0015 0016 000D 001B 0010 0019 000F 0019 0014 0015 0011 0016 0010 001A 0011 0018 000E 001A 0012 0016 000F 001A 0011 0019 0012 0016 000F 0018 0015 0014 0012 0017 0011 0018 000E 001A 0013 0015 000E 001B 0013 0015 0010 0019 0014 0015 0010 003F 0013 003C 0012 003D 0012 003C 0011 004D 0072 015B 0016 003D 0010 0018 0015 0013 000F 001B 0011 0017 0011 0016 0013 0016 0014 0013 0011 0018 0012 003E"

And this is the final result:

 
> 21:42:20 | [D] | [button:013] | 'Samsung HVAC - cool_26' Pressed.
> -- | -- | -- | --
> 21:42:20 | [D] | [remote.pronto:101] | Send Pronto: frequency=38kHz
> 21:42:20 | [D] | [remote.pronto:101] | Send Pronto: frequency=38kHz
> 21:42:20 | [D] | [remote.pronto:106] | Send Pronto: intros=348
> 21:42:20 | [D] | [remote.pronto:107] | Send Pronto: repeats=0
> 21:42:20 | [E] | [remote.pronto:109] | Inconsistent data, not sending
> 21:42:20 | [E] | [remote.pronto:109] | Inconsistent data, not sending
> 21:42:20 | [E] | [remote_transmitter:117] | Empty data
> 21:42:20 | [E] | [remote_transmitter:117] | Empty data

@ssieb is your PR already merged in 2022.12.8 ? Or maybe this is a different problem?

ssieb, 2023-01-31:
It is merged.  That's why you get the `Inconsistent data` instead of a more obscure error message.  Also, the second line of pronto data.  However, it appears that's still not long enough to show the entire message, so it needs to be expanded to more lines.