esphome/issues #3559: Lolin C3 Mini UART#2 not useable

AndreasBVB, 2022-09-06:
### The problem

Able to define two UART ports: UART0 TX-21/RX-20, UART1 TX-0/RX-1.
As long as a sensor is associated to UART1, the device will not boot. Log turned off.

### Which version of ESPHome has the issue?

2022.8.1

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2022.3.5

### What platform are you using?

ESP32-IDF

### Board

Lolin C3 Mini

### Component causing the issue

uart

### Example YAML snippet

```yaml
substitutions:
  device_upper: LOLIN C3 MINI
  device_lower: lc3m
  device_name: lc3m

esphome:
  name: ${device_name}
  platformio_options:
    board_build.flash_mode: dio
  
esp32:
  board: esp32-c3-devkitm-1
  variant: esp32c3
  framework:
    type: esp-idf

# Enable logging
logger:
  level: NONE

# Enable Home Assistant API
api:

ota:

wifi:
  output_power: "8.5"
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  #power_save_mode: HIGH
  fast_connect: true
  # Enable fallback hotspot (captive portal) in case wifi connection fails

text_sensor:
  - platform: template
    name: ${device_upper} BRAND
    lambda: return {"Lolin"};
    update_interval: 7d
    entity_category: diagnostic
  - platform: template
    name: ${device_upper} HARDWARE
    lambda: return {"C3 Mini"};
    update_interval: 7d
    entity_category: diagnostic
  - platform: template
    name: ${device_upper} PERIPHERALS
    lambda: return {""};
    update_interval: 7d
    entity_category: diagnostic
  - platform: version
    name: ${device_upper} SOFTWARE
    entity_category: diagnostic
  - platform: wifi_info
    ip_address:
      name: ${device_upper} IP
    mac_address:
      name: ${device_upper} MAC

sensor:
  - platform: wifi_signal
    name: ${device_upper} WiFi Signal
    update_interval: 30s
  - platform: uptime
    update_interval: 30s
    name: ${device_upper} Uptime
  - platform: uptime
    update_interval: 1s
    id: ${device_lower}_uptime_internal
#  - platform: senseair
#    co2:
#      name: "UART0 CO2"
#    update_interval: 10s
#    uart_id: uart_0
  - platform: pmsx003
    type: PMSX003
    pm_1_0:
      name: "PM1"
    pm_2_5:
      name: "PM25"
    pm_10_0:
      name: "PM10"
    uart_id: uart_1
    update_interval: 60000ms

button:
  - platform: restart
    name: ${device_upper} RESTART

uart:
  - id: uart_0
    tx_pin: 21
    rx_pin: 20
    baud_rate: 9600
  - id: uart_1
    tx_pin: 0
    rx_pin: 1
    baud_rate: 9600
```

AndreasBVB, 2022-09-06:
From Espressif I learned C3 has 2x UARTs and from this chart of an unknown C3 device I found this pin combination.
[image]

AndreasBVB, 2022-09-13:
Wired log:

0m

I (850) cpu_start: Pro cpu start user code
I (850) cpu_start: cpu freq: 160000000
I (850) cpu_start: Application information:
I (853) cpu_start: Project name:     lc3m
I (857) cpu_start: App version:      2022.8.1
I (862) cpu_start: Compile time:     Sep 13 2022 16:13:00
I (869) cpu_start: ELF file SHA256:  d55ce725856fc5fe...
I (875) cpu_start: ESP-IDF:          4.3.2
I (879) heap_init: Initializing. RAM available for dynamic allocation:
I (887) heap_init: At 3FC95FD0 len 0002A030 (168 KiB): DRAM
I (893) heap_init: At 3FCC0000 len 0001F060 (124 KiB): STACK/DRAM
I (900) heap_init: At 50000010 len 00001FF0 (7 KiB): RTCRAM
ESP-ROM:esp32c3-api1-20210207
I (850) cpu_start: Pro cpu start user code
I (850) cpu_start: cpu freq: 160000000
I (850) cpu_start: Application information:
I (853) cpu_start: Project name:     lc3m
I (857) cpu_start: App version:      2022.8.1
I (862) cpu_start: Compile time:     Sep 13 2022 16:13:00
I (869) cpu_start: ELF file SHA256:  d55ce725856fc5fe...
I (875) cpu_start: ESP-IDF:          4.3.2
I (879) heap_init: Initializing. RAM available for dynamic allocation:
I (887) heap_init: At 3FC95FD0 len 0002A030 (168 KiB): DRAM
I (893) heap_init: At 3FCC0000 len 0001F060 (124 KiB): STACK/DRAM
I (900) heap_init: At 50000010 len 00001FF0 (7 KiB): RTCRAM
I (850) cpu_start: Pro cpu start user code
I (850) cpu_start: cpu freq: 160000000
I (850) cpu_start: Application information:
I (853) cpu_start: Project name:     lc3m
I (857) cpu_start: App version:      2022.8.1
I (862) cpu_start: Compile time:     Sep 13 2022 16:13:00
I (869) cpu_start: ELF file SHA256:  d55ce725856fc5fe...
I (875) cpu_start: ESP-IDF:          4.3.2
I (879) heap_init: Initializing. RAM available for dynamic allocation:
I (887) heap_init: At 3FC95FD0 len 0002A030 (168 KiB): DRAM
I (893) heap_init: At 3FCC0000 len 0001F060 (124 KiB): STACK/DRAM
I (900) heap_init: At 50000010 len 00001FF0 (7 KiB): RTCRAM
ESP-ROM:esp32c3-api1-20210207
I (850) cpu_start: Pro cpu start user code
I (850) cpu_start: cpu freq: 160000000
I (850) cpu_start: Application information:
I (853) cpu_start: Project name:     lc3m
I (857) cpu_start: App version:      2022.8.1
I (862) cpu_start: Compile time:     Sep 13 2022 16:13:00
I (869) cpu_start: ELF file SHA256:  d55ce725856fc5fe...
I (875) cpu_start: ESP-IDF:          4.3.2
I (879) heap_init: Initializing. RAM available for dynamic allocation:
I (887) heap_init: At 3FC95FD0 len 0002A030 (168 KiB): DRAM
I (893) heap_init: At 3FCC0000 len 0001F060 (124 KiB): STACK/DRAM
I (900) heap_init: At 50000010 len 00001FF0 (7 KiB): RTCRAM
I (850) cpu_start: Pro cpu start user code
I (850) cpu_start: cpu freq: 160000000
I (850) cpu_start: Application information:
I (853) cpu_start: Project name:     lc3m
I (857) cpu_start: App version:      2022.8.1
I (862) cpu_start: Compile time:     Sep 13 2022 16:13:00
I (869) cpu_start: ELF file SHA256:  d55ce725856fc5fe...
I (875) cpu_start: ESP-IDF:          4.3.2
I (879) heap_init: Initializing. RAM available for dynamic allocation:
I (887) heap_init: At 3FC95FD0 len 0002A030 (168 KiB): DRAM
I (893) heap_init: At 3FCC0000 len 0001F060 (124 KiB): STACK/DRAM
I (900) heap_init: At 50000010 len 00001FF0 (7 KiB): RTCRAM
ESP-ROM:esp32c3-api1-20210207
I (850) cpu_start: Pro cpu start user code
I (850) cpu_start: cpu freq: 160000000
I (850) cpu_start: Application information:
I (853) cpu_start: Project name:     lc3m
I (857) cpu_start: App version:      2022.8.1
I (862) cpu_start: Compile time:     Sep 13 2022 16:13:00
I (869) cpu_start: ELF file SHA256:  d55ce725856fc5fe...
I (875) cpu_start: ESP-IDF:          4.3.2
I (879) heap_init: Initializing. RAM available for dynamic allocation:
I (887) heap_init: At 3FC95FD0 len 0002A030 (168 KiB): DRAM
I (893) heap_init: At 3FCC0000 len 0001F060 (124 KiB): STACK/DRAM
I (900) heap_init: At 50000010 len 00001FF0 (7 KiB): RTCRAM
I (850) cpu_start: Pro cpu start user code
I (850) cpu_start: cpu freq: 160000000
I (850) cpu_start: Application information:
I (853) cpu_start: Project name:     lc3m
I (857) cpu_start: App version:      2022.8.1
I (862) cpu_start: Compile time:     Sep 13 2022 16:13:00
I (869) cpu_start: ELF file SHA256:  d55ce725856fc5fe...
I (875) cpu_start: ESP-IDF:          4.3.2
I (879) heap_init: Initializing. RAM available for dynamic allocation:
I (887) heap_init: At 3FC95FD0 len 0002A030 (168 KiB): DRAM
I (893) heap_init: At 3FCC0000 len 0001F060 (124 KiB): STACK/DRAM
I (900) heap_init: At 50000010 len 00001FF0 (7 KiB): RTCRAM

AndreasBVB, 2022-09-24:
Further info:
Switched to Espressif-Arduino, no such a problem, so focus is on esp-idf.

platform_packages:
  - framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#2.0.3
platform:
  - https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream

Schluggi, 2022-11-29:
@AndreasBVB could you share your yaml file? I struggle with the same issue. (ESPHome newbie)