esphome/issues #3564: api services remote_transmitter compile error esp32c3 #2

paccerdk, 2022-09-08:
### The problem

Getting undefined reference when using lambda

Seems that stale bot is closing a **lot** of issues in this repo that should not be closed - I've come across this a lot in this repo., this time, came here from google, with same problem.

original issue: https://github.com/esphome/issues/issues/3201

one thing i'm noticing is that i use esp32-c3 too, in this case, with esp-idf however.

### Which version of ESPHome has the issue?

ESPHome version 2022.8.3 compiled on Sep  7 2022, 22:10:13

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.8.7

### What platform are you using?

ESP32-IDF

### Board

ESP-C3-12F custom hardware

### Component causing the issue

api service lambda

### Example YAML snippet

```yaml
esphome:
  name: espinfra1
  platformio_options:
    board_build.flash_mode: dio
    
esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: esp-idf

# Enable logging
logger:

# Enable Home Assistant API
api:
  services:
    - service: send_ir_raw
      variables:
        keycode: int[]
      then:
      - remote_transmitter.transmit_raw:
          transmitter_id: IR_TX
          code: !lambda 'return keycode;'
          carrier_frequency: 38kHz

ota:
  password: "xxxxxxxxxxx"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
          
remote_transmitter:
  - id: IR_TX
    pin: GPIO3
    carrier_duty_percent: 50%
  - id: dummy #workaround for issue: https://github.com/esphome/issues/issues/2934 
    pin: GPIO0
    carrier_duty_percent: 50%

  
remote_receiver:
  pin:
    number: GPIO4
    inverted: true
  dump: all
```

### Anything in the logs that might be useful for us?

```txt
INFO Reading configuration /config/esphome/espinfra1.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing espinfra1 (board: esp32-c3-devkitm-1; framework: espidf; platform: platformio/espressif32 @ 3.5.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32C3 160MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.40302.0 (4.3.2) 
 - tool-cmake @ 3.16.4 
 - tool-ninja @ 1.7.1 
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch2 
 - toolchain-xtensa-esp32s2 @ 8.4.0+2021r2-patch2
Reading CMake configuration...
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
No dependencies
Compiling /data/espinfra1/.pioenvs/espinfra1/src/main.o
Linking /data/espinfra1/.pioenvs/espinfra1/firmware.elf
/data/cache/platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: /data/espinfra1/.pioenvs/espinfra1/src/main.o: in function `esphome::api::ListEntitiesServicesArgument::ListEntitiesServicesArgument()':
/config/esphome/.esphome/build/espinfra1/src/esphome/components/api/api_pb2.h:786: undefined reference to `esphome::api::enums::ServiceArgType esphome::api::to_service_arg_type<std::vector<long, std::allocator<long> > >()'
collect2: error: ld returned 1 exit status
*** [/data/espinfra1/.pioenvs/espinfra1/firmware.elf] Error 1
========================= [FAILED] Took 21.36 seconds =========================
```

waicool20, 2022-09-18:
Similar issue but I got a bit more in the log
```
HARDWARE: ESP32C3 160MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.40302.0 (4.3.2) 
 - tool-cmake @ 3.16.4 
 - tool-ninja @ 1.7.1 
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch2 
 - toolchain-xtensa-esp32s2 @ 8.4.0+2021r2-patch2
Reading CMake configuration...
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- noise-c @ 0.1.4
|   |-- libsodium @ 1.10018.1
Compiling /data/test/.pioenvs/test/src/main.o
Linking /data/test/.pioenvs/test/firmware.elf
/data/cache/platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: /data/test/.pioenvs/test/src/main.o: in function `esphome::api::ExecuteServiceArgument* std::__uninitialized_copy<false>::__uninit_copy<__gnu_cxx::__normal_iterator<esphome::api::ExecuteServiceArgument const*, std::vector<esphome::api::ExecuteServiceArgument, std::allocator<esphome::api::ExecuteServiceArgument> > >, esphome::api::ExecuteServiceArgument*>(__gnu_cxx::__normal_iterator<esphome::api::ExecuteServiceArgument const*, std::vector<esphome::api::ExecuteServiceArgument, std::allocator<esphome::api::ExecuteServiceArgument> > >, __gnu_cxx::__normal_iterator<esphome::api::ExecuteServiceArgument const*, std::vector<esphome::api::ExecuteServiceArgument, std::allocator<esphome::api::ExecuteServiceArgument> > >, esphome::api::ExecuteServiceArgument*)':
/data/cache/platformio/packages/toolchain-riscv32-esp/riscv32-esp-elf/include/c++/8.4.0/bits/stl_uninitialized.h:82: undefined reference to `long esphome::api::get_execute_arg_value<long>(esphome::api::ExecuteServiceArgument const&)'
/data/cache/platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: /data/test/.pioenvs/test/src/main.o: in function `esphome::api::ListEntitiesServicesArgument::ListEntitiesServicesArgument()':
/data/cache/platformio/packages/toolchain-riscv32-esp/riscv32-esp-elf/include/c++/8.4.0/bits/stl_vector.h:1085: undefined reference to `esphome::api::enums::ServiceArgType esphome::api::to_service_arg_type<long>()'
```

paccerdk, 2023-01-18:
not stale

CarlosGS, 2023-03-15:
I confirm this is still an issue with ESP32C3, in this case using Arduino idf (haven't tried with esp-idf).
```
/data/cache/platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: /data/co2portable/.pioenvs/co2portable/src/main.cpp.o: in function `esphome::api::ExecuteServiceArgument* std::__uninitialized_copy<false>::__uninit_copy<__gnu_cxx::__normal_iterator<esphome::api::ExecuteServiceArgument const*, std::vector<esphome::api::ExecuteServiceArgument, std::allocator<esphome::api::ExecuteServiceArgument> > >, esphome::api::ExecuteServiceArgument*>(__gnu_cxx::__normal_iterator<esphome::api::ExecuteServiceArgument const*, std::vector<esphome::api::ExecuteServiceArgument, std::allocator<esphome::api::ExecuteServiceArgument> > >, __gnu_cxx::__normal_iterator<esphome::api::ExecuteServiceArgument const*, std::vector<esphome::api::ExecuteServiceArgument, std::allocator<esphome::api::ExecuteServiceArgument> > >, esphome::api::ExecuteServiceArgument*)':
/data/cache/platformio/packages/toolchain-riscv32-esp/riscv32-esp-elf/include/c++/8.4.0/bits/stl_uninitialized.h:82: undefined reference to `long esphome::api::get_execute_arg_value<long>(esphome::api::ExecuteServiceArgument const&)'
/data/cache/platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: /data/co2portable/.pioenvs/co2portable/src/main.cpp.o: in function `esphome::api::ListEntitiesServicesArgument::ListEntitiesServicesArgument()':
/config/esphome/.esphome/build/co2portable/src/esphome/components/adc/adc_sensor.h:17: undefined reference to `esphome::api::enums::ServiceArgType esphome::api::to_service_arg_type<long>()'
collect2: error: ld returned 1 exit status
```
It only happens when enabling services:, removing that part allows to compile correctly.

agners, 2023-03-28:
Can't you just define it using `int`, e.g.

```
template<> std::vector<int> get_execute_arg_value<std::vector<int>>(const ExecuteServiceArgument &arg) {
```

Honestly, I am also not sure as to why RISC-V defines uint32_t differently.

agners, 2023-03-28:
Hm, that would essentially revert my change :thinking: Yeah, not sure :cold_sweat:

CarlosGS, 2023-03-28:
Thanks! No worries, I was just hoping it'd be something more evident :smile:
Unfortunately Espressif has adopted this oddity for Xtensa too: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/migration-guides/release-5.x/5.0/gcc.html#int32-t-and-uint32-t-for-xtensa-compiler & https://github.com/espressif/esp-idf/issues/9511#issuecomment-1354539638.

Another issue caused by this: https://github.com/esphome/issues/issues/3970

For the api services, I've tried just swapping all `int` to `int32_t` https://github.com/CarlosGS/esphome/commit/a07ad7f7074e68a03aa89c8ef463bac0585955a9, in line with your change, and it compiles without errors/warnings. But the API stops working even for the normal HASS connection.
The compilation is clean & USB logging doesn't show anything relevant with VERY_VERBOSE (sensors & OTA work OK, just no API). And on the HASS side, logs just show `Connection error occurred: [Errno 104] Connection reset by peer`.

Thanks for having a look. Can't do much more at the moment either, but it's good to keep an eye on that annoying change.

agners, 2023-03-29:
> Unfortunately Espressif has adopted this oddity for Xtensa too: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/migration-guides/release-5.x/5.0/gcc.html#int32-t-and-uint32-t-for-xtensa-compiler & [espressif/esp-idf#9511 (comment)](https://github.com/espressif/esp-idf/issues/9511#issuecomment-1354539638).

Well, I guess it is a good thing in the end, as it matches now and it seems to be what upstream GCC is doing. But yeah, the change is annoying at first :sweat_smile: 

> But the API stops working even for the normal HASS connection.

I mean it didn't compile before, so "stops working" sounds weird to me?

CarlosGS, 2023-03-29:
Good point! Well the linker error mentioned in this thread only appears when adding `services:` to the api section.

I meant when changing `user_services.cpp` to turn all int to int32_t, there are no linker errors anymore, but the api connection now doesn't work (with or without `services:` enabled in YAML).

twosky2000, 2023-04-24:
here's my fix:
`
esphome:
  name: fingerprint
  platformio_options:
    board_build.flash_mode: dio
    board_build.extra_flags:
      #- "-DARDUINO_USB_MODE=0"
      - "-DARDUINO_USB_CDC_ON_BOOT=0"
      - "-include \"fix.h\""
 `

.esphome/build/fingerprint/fix.h
`
#ifdef __riscv

#undef __INT32_TYPE__
#define __INT32_TYPE__      int

#undef __UINT32_TYPE__
#define __UINT32_TYPE__     unsigned int

#endif // __riscv
`

I got the idea from
https://github.com/espressif/esp-idf/issues/6906
https://github.com/arendst/Tasmota/pull/11808/commits

seams esp-idf assumes int is int long for RISC-V/esp32-3c. Maybe PlatformIO can use this fix too.

I try to make a pr when I figure out how to make this automatic.

mkaiser, 2023-08-11:
`

> here's my fix:
> 
> ```
> 
> esphome:
>   name: fingerprint
>   platformio_options:
>     board_build.flash_mode: dio
>     board_build.extra_flags:
>       #- "-DARDUINO_USB_MODE=0"
>       - "-DARDUINO_USB_CDC_ON_BOOT=0"
>       - "-include \"fix.h\""
> ```
> 
> file at .esphome/build/fingerprint/fix.h
> 
> ```
> 
> #ifdef __riscv
> 
> #undef __INT32_TYPE__
> #define __INT32_TYPE__      int
> 
> #undef __UINT32_TYPE__
> #define __UINT32_TYPE__     unsigned int
> 
> #endif // __riscv
> ```
> 
> I got the idea from [espressif/esp-idf#6906](https://github.com/espressif/esp-idf/issues/6906) https://github.com/arendst/Tasmota/pull/11808/commits
> 
> seams esp-idf assumes int is int long for RISC-V/esp32-3c. Maybe PlatformIO can use this fix too.
> 
> I try to make a pr when I figure out how to make this automatic.

thank you very much for this hotfix, got my stuff working now, after several headaches ;)

@ESP Core developers: Is there any chance you can "really fix" this datatype interpretation issue?

clomads, 2023-08-29:
I can also confirm the hotfix works

CarlosGS, 2023-11-21:
Here's the updated patch, after the .esphome build folder was moved in https://github.com/esphome/esphome/pull/5374
- Create the file `fix.h` mentioned above, but now place it in the main esphome folder (next to the yaml files).
- In the yaml code you need to add `src/` and additional `includes:` statement:
```yaml
esphome:
  name: theName
  platformio_options:
    board_build.extra_flags:
      - "-include \"src/fix.h\""
  includes:
    - fix.h
```
This way the fix.h file is automatically copied into the src/ folder within the platformio build directory.

babgvant, 2024-01-23:
> Here's the updated patch, after the .esphome build folder was moved in [esphome/esphome#5374](https://github.com/esphome/esphome/pull/5374)
> 
>     * Create the file `fix.h` mentioned above, but now place it in the main esphome folder (next to the yaml files).
> 
>     * In the yaml code you need to add `src/` and additional `includes:` statement:
> 
> 
> ```yaml
> esphome:
>   name: theName
>   platformio_options:
>     board_build.extra_flags:
>       - "-include \"src/fix.h\""
>   includes:
>     - fix.h
> ```
> 
> This way the fix.h file is automatically copied into the src/ folder within the platformio build directory.

Thank you! This worked for me as well with a ESP-C3-Zero board.

CarlosGS, 2024-04-05:
Not sure when or how, but this seems to be already fixed :tada:
Anyone else can confirm it is possible to compile without the patch?

mircsicz, 2024-05-23:
For me it just didn

> Not sure when or how, but this seems to be already fixed 🎉 Anyone else can confirm it is possible to compile without the patch?

I'm on a Lolin C3 Pico and for me it didn't without the fix...